espressif/esp-webrtc-solution #3: Seeed studio ReSpeaker support

patrickkeenan, 2025-01-29:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

It would be great to have support for the ReSpeaker Lit with XIAO esp32 chip. Maybe that’s just adding an entry to the boards config file.

### Use cases

Use a popular audio board

TempoTian, 2025-02-05:
Sorry for the later repy for the long holiday.  

I have check the Respeaker, esp32s3 connect to DSP XU316 through I2S and I2C directly. So I think can add adapter for it use `esp_codec_dev`.  Then can use the `esp_codec_dev` handle to do playback and record as the example code.

For XU316 already have already support AEC and other audio processing, you can use `esp_capture_new_audio_codec_src` as the capture source directly. Then all other code can be reused. 

 I see https://github.com/respeaker/ReSpeaker_Lite/tree/master/xiao_esp32s3_arduino_examples show how to do control and read registers for SU316. As I don’t have the device on hand, you may need to manually add a wrapper based on https://github.com/espressif/esp-adf/blob/master/components/esp_codec_dev/device/es8311/es8311.c.

diegoasua, 2025-07-29:
hey @patrickkeenan did you ever figure it out?

patrickkeenan, 2025-08-21:
Nope haven't tried

spikelab, 2025-09-05:
fwiw I'm trying to figure out which hw to buy and the respeaker is on the list so I'd love to see this working.

@TempoTian any chance you'd be willing to work on it if I donated a board?

happy to ship you a unit if that helps the community. thanks

TempoTian, 2025-09-08:
@spikelab If you have a board you can following https://github.com/espressif/esp-webrtc-solution/issues/89#issuecomment-3264400903 steps and check whether it works or not. If meet any problem, you can add your logs and description in the notes, I will help you to check the possible issues. I think it is more convinient maybe.

TempoTian, 2025-09-18:
I have add a very basic version for respeaker_lite.
You can replace it to codec_board in components folder.
[codec_board_respeaker_lite.zip](https://github.com/user-attachments/files/22404033/codec_board_respeaker_lite.zip)

Then change settings.h to change board name to
```
#define TEST_BOARD_NAME "RESPEAKER_LITE"
```
Change media_sys.c to use codec_src for codec internally support AEC
```
   esp_capture_audio_dec_src_cfg_t codec_cfg = {
        .record_handle = get_record_handle(),
    };
    capture_sys.aud_src = esp_capture_new_audio_dev_src(&codec_cfg);
```
If any issue, please feedback I will help to look into.

diegoasua, 2025-09-18:
build stops during PSRAM init

```
I (24) boot: ESP-IDF v5.5 2nd stage bootloader
I (24) boot: compile time Sep 18 2025 14:30:49
I (25) boot: Multicore bootloader
I (25) boot: chip revision: v0.2
I (27) boot: efuse block revision: v1.3
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (35) boot.esp32s3: SPI Mode       : DIO
I (39) boot.esp32s3: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (62) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (69) boot:  2 factory          factory app      00 00 00010000 00300000
I (75) boot: End of partition table
I (79) esp_image: segment 0: paddr=00010020 vaddr=3c1a0020 size=8d188h (577928) map
I (188) esp_image: segment 1: paddr=0009d1b0 vaddr=3fca4200 size=02e68h ( 11880) load
I (191) esp_image: segment 2: paddr=000a0020 vaddr=42000020 size=19f358h (1700696) map
I (492) esp_image: segment 3: paddr=0023f380 vaddr=3fca7068 size=02d68h ( 11624) load
I (494) esp_image: segment 4: paddr=002420f0 vaddr=40378000 size=1c1d8h (115160) load
I (520) esp_image: segment 5: paddr=0025e2d0 vaddr=600fe000 size=00020h (    32) load
I (532) boot: Loaded app from partition at offset 0x10000
I (532) boot: Disabling RNG early entropy source...
W (542) flash HPM: HPM mode is optional feature that depends on flash model. Read Docs First!
W (542) flash HPM: HPM mode with DC adjustment is disabled. Some flash models may not be supported. Read Docs First!
W (549) flash HPM: High performance mode of this flash model hasn't been supported.
I (557) MSPI Timing: Flash timing tuning index: 2
I (561) octal_psram: vendor id    : 0x0d (AP)
I (565) octal_psram: dev id       : 0x02 (generation 3)
I (570) octal_psram: density      : 0x03 (64 Mbit)
I (575) octal_psram: good-die     : 0x01 (Pass)
I (579) octal_psram: Latency      : 0x01 (Fixed)
I (583) octal_psram: VCC          : 0x01 (3V)
I (587) octal_psram: SRF          : 0x01 (Fast Refresh)
I (592) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (597) octal_psram: BurstLen     : 0x01 (32 Byte)
I (602) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (607) octal_psram: DriveStrength: 0x00 (1/1)
```

did I forget some setting?

TempoTian, 2025-09-19:
Does the board carried with octal_psram or quad_psram? Can use menuconfig to select quad version and have a try.

diegoasua, 2025-09-19:
quad crashes. must be octal

TempoTian, 2025-09-22:
PSRAM config to 40M will make program run slow, the log also show that watchdog triggered for CPU too busy on opus encoding. Is it possible to config flash and psram both to 80M and have a try? From the log I see that codec init success, if cpu is enough I think there may be no other issues.

diegoasua, 2025-09-22:
Was able to run at 80MHz. Doesn't crash anymore. I don't hear anything tho on the speaker. When I talk it does trigger a response that I can see in the terminal but it seems it is crazy fast.

```
Get samplerate 16000 chanel 1
I (8836) AV_RENDER: Get need resample 1 in:16000 out:16000
I (8836) I2S_RENDER: open channel:2 sample_rate:16000 bits:16
I (8836) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:3
I (8836) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:3
I (8836) Adev_Codec: Open codec device OK
I (8846) AV_RENDER: ARender Auto pause audio first frame at 42199100
I (8956) SCTP: 0 Receive chunk 10 SCTP_COOKIE_ECHO
I (8956) SCTP: Send ECHO_ACK chunk
I (8956) SCTP: 0 state 5
I (8956) webrtc: PeerConnectionState: 8
====================Event 4======================
I (8966) SCTP: Send DCEP esp_channel event:3 type:80 si:1
I (9066) webrtc: PeerConnectionState: 9
====================Event 6======================
Begin to send json:{
        "type": "response.create",
        "response":     {
                "modalities":   ["text", "audio"],
                "instructions": "You are helpful and have some tools installed. In the tools you have the ability to control a light bulb and change speaker volume. Say 'How can I help?"
        }
}
Transcript: 
I (9496) webrtc: Send A:720 [37:1776] Recv A:42201080 [34:736]
I (9496) PEER_DEF: Send 59:6386 d_out:1598 receive 90:8699 din:6424
I (9496) SCTP: Send: 1238 Ack: 1237 Cached: 0 Receive:0
I (9496) SCTP: Recv: 4264756648 ACK:4264756648 Cached: 0 SackCount:1

Transcript: How can I help?
Transcript: How can I help?
Transcript: How can I help?
Transcript: How can I help?
I (11496) webrtc: Send A:2740 [101:4848] Recv A:42207020 [99:5513]
I (11496) PEER_DEF: Send 119:8240 d_out:0 receive 206:10454 din:2328
I (11496) SCTP: Send: 1238 Ack: 1237 Cached: 0 Receive:0
I (11496) SCTP: Recv: 4264756655 ACK:4264756655 Cached: 0 SackCount:9

I (13496) webrtc: Send A:4720 [99:4752] Recv A:42213080 [100:300]
I (13496) PEER_DEF: Send 111:7710 d_out:0 receive 201:2676 din:0
I (13496) SCTP: Send: 1238 Ack: 1237 Cached: 0 Receive:0
I (13496) SCTP: Recv: 4264756655 ACK:4264756655 Cached: 0 SackCount:21

Transcript: 
Transcript: ¡Hola! ¿Cómo estás?
Transcript: ¡Hola! ¿Cómo estás?
Transcript: ¡Hola! ¿Cómo estás?
Transcript: ¡Hola! ¿Cómo estás?
I (15496) webrtc: Send A:6720 [99:4752] Recv A:42219080 [100:5195]
I (15496) PEER_DEF: Send 109:7597 d_out:0 receive 236:16966 din:8220
I (15496) SCTP: Send: 1238 Ack: 1237 Cached: 0 Receive:0
I (15496) SCTP: Recv: 4264756680 ACK:4264756680 Cached: 0 SackCount:1

Transcript: 
Transcript: I'm here to assist you! What would you like to know or do?
Transcript: I'm here to assist you! What would you like to know or do?
Transcript: I'm here to assist you! What would you like to know or do?
Transcript: I'm here to assist you! What would you like to know or do?
I (17496) webrtc: Send A:8720 [100:4800] Recv A:42225080 [100:12738]
I (17496) PEER_DEF: Send 137:9405 d_out:0 receive 223:24428 din:8500
I (17496) SCTP: Send: 1238 Ack: 1237 Cached: 0 Receive:0
I (17496) SCTP: Recv: 4264756709 ACK:4264756709 Cached: 0 SackCount:6
```

TempoTian, 2025-09-22:
You can set speaker volume to max and check whether have sound or not. Also there is a debug function ` test_capture_to_player`, can do echo playback, you can use it to debug. As open AI answered so I think the mic record path is all right now.
```c
#include “esp_codec_dev.h”
static int build_player_system()
{
    i2s_render_cfg_t i2s_cfg = {
        .play_handle = get_playback_handle(),
    };
    esp_codec_dev_set_out_vol(get_playback_handle(), 100);
```

diegoasua, 2025-10-02:
Any updates on this? How can I help?