espressif/esp-webrtc-solution #6: openai_demo no sound with ESP32-S3 Korvo 2 3.1

prothan, 2025-02-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Expect open ai reply to be emitted from speaker. 

### Actual behavior (suspected bug)

Console seems to show that everything is functioning correctly, but no sound emitted. 

### Error logs or terminal output

```shell
From korvo startup. 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4 2nd stage bootloader
I (27) boot: compile time Feb  4 2025 07:45:24
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (30) boot: efuse block revision: v1.2       
I (33) boot.esp32s3: Boot SPI Speed : 80MHz   
I (37) boot.esp32s3: SPI Mode       : DIO     
I (41) boot.esp32s3: SPI Flash Size : 4MB        
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length  
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00300000
I (78) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c190020 size=81a30h (530992) map
I (182) esp_image: segment 1: paddr=00091a58 vaddr=3fca5a00 size=0669ch ( 26268) load
I (188) esp_image: segment 2: paddr=000980fc vaddr=40378000 size=07f1ch ( 32540) load
I (195) esp_image: segment 3: paddr=000a0020 vaddr=42000020 size=18ed14h (1633556) map
I (483) esp_image: segment 4: paddr=0022ed3c vaddr=4037ff1c size=15a54h ( 88660) load
I (503) esp_image: segment 5: paddr=00244798 vaddr=600fe100 size=0001ch (    28) load
I (514) boot: Loaded app from partition at offset 0x10000
I (515) boot: Disabling RNG early entropy source...
W (525) flash HPM: HPM mode is optional feature that depends on flash model. Read Docs First!
W (525) flash HPM: HPM mode with DC adjustment is disabled. Some flash models may not be supported. Read Docs First!
W (532) flash HPM: High performance mode of this flash model hasn't been supported.
I (540) MSPI Timing: Flash timing tuning index: 2
I (544) octal_psram: vendor id    : 0x0d (AP)
I (548) octal_psram: dev id       : 0x02 (generation 3)
I (553) octal_psram: density      : 0x03 (64 Mbit)
I (557) octal_psram: good-die     : 0x01 (Pass)
I (561) octal_psram: Latency      : 0x01 (Fixed)
I (566) octal_psram: VCC          : 0x01 (3V)
I (570) octal_psram: SRF          : 0x01 (Fast Refresh)
I (575) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (580) octal_psram: BurstLen     : 0x01 (32 Byte)
I (584) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
[... 307 lines cut ...]
### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

CMD

### Additional context

Thanks for creating this, but I cannot get the demo to work. I am using a new ESP32-S3-Korvo 2 v3.1 board connected to a windows 10 pc via usb(comm uart). I am using ESP-IDF v5.4.0. Code compiles fine and will upload to board. Application appears to be running properly. However, when I make connection to openai(with api key )and type start I can only see the text message being sent(per console). No audio is played through speaker. I have a similar setup as was shown in demo video, 4 Ohm 3 Watt Mini Speaker. I am using putty as console app. I tried typing help in console, which shows available commands. I tried typing start. The esp32-s3 continues to log as if it is working, but no audio.

I tried rec2play and it works well. I can talk into microphone and hear audio repeated from board. So I am unsure what issue is. 

Also, as others have mentioned, it would be nice to have a demo that could be used with I2S only (if possible). I understand for the demo sake you want to demonstrate optimum capabilities, but it makes demo overly complicated. Particularly, for people like me that are not as familiar with webrtc. There is an overwhelming amount to learn in one demo. 

Thanks again, I think this demo will be great once I get it working.

TempoTian, 2025-02-05:
I have checked the log, if rec2play is OK, then it means the whole record and playback path works.  
From the log I see that received data is empty packets,(received 100 packets total size is 2600,  each size is 26 means an empty packet): 
```
I (10045) webrtc: Send A:2520 [100:22500] Recv A:34030915 [100:2600]
```
You need talk to openAI, if it replies, the packet size will be changed, you will heared replied sound. Can you take more logs to confirm.
For I2S only devices, codec_board have a dummy board configuration, https://github.com/espressif/esp-webrtc-solution/blob/main/components/codec_board/board_cfg.txt#L20  
You can change the i2s setting to your board settings, if it supports both input and output change its type to be `in_out`
```
Board: DUMMY_CODEC_BOARD
i2s: {mclk: 5, bclk: 6, ws: 16, din: -1, dout: 8}
out: {codec: DUMMY, pa: 15, i2c_port: -1}
```

prothan, 2025-02-06:
Thanks for the quick feedback. I tried running the demo again. I noticed the line you specified continues to repeat, even after speaking into device. 

I (10045) webrtc: Send A:2520 [100:22500] Recv A:34030915 [100:2600]

I tried verifying c_cpp_properties.json paths were correct. I think that they look good. Paths were defined as follows:

 "includePath": [
                "${workspaceFolder}/**",
                "C:/esp/v5.4/esp-idf/components/**",
                "${workspaceFolder}/main/**",
                "${workspaceFolder}../../common/**",
                "${workspaceFolder}../../../components/**",
                "${workspaceFolder}/managed_components/**"
            ],

I git cloned https://github.com/espressif/esp-webrtc-solution.git again. Tried to re-run on windows, it did not work. I tried with wsl ubuntu in vscode remote. In both instances I get same output, it is slightly different than before. I am not sure why. All these lines were highlighted in yellow, except for CODEC_INIT which was highlighted in red.

---
W (525) flash HPM: HPM mode is optional feature that depends on flash model. Read Docs First!
W (525) flash HPM: HPM mode with DC adjustment is disabled. Some flash models may not be supported. Read Docs First!
W (532) flash HPM: High performance mode of this flash model hasn't been supported.

W (921) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.

E (1027) CODEC_INIT: in:1 out:1 port: 1
W (1047) i2s_tdm: the current mclk multiple is too small, adjust the mclk multiple to 384
---

Full output

--- esp-idf-monitor 1.5.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (27) boot: compile time Feb  5 2025 23:42:23
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (30) boot: efuse block revision: v1.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00300000
I (78) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c190020 size=81a50h (531024) map
I (182) esp_image: segment 1: paddr=00091a78 vaddr=3fca5a00 size=0669ch ( 26268) load
I (188) esp_image: segment 2: paddr=0009811c vaddr=40378000 size=07efch ( 32508) load
I (195) esp_image: segment 3: paddr=000a0020 vaddr=42000020 size=18ed14h (1633556) map
I (484) esp_image: segment 4: paddr=0022ed3c vaddr=4037fefc size=15a74h ( 88692) load
I (503) esp_image: segment 5: paddr=002447b8 vaddr=600fe100 size=0001ch (    28) load
I (515) boot: Loaded app from partition at offset 0x10000
I (515) boot: Disabling RNG early entropy source...
W (525) flash HPM: HPM mode is optional feature that depends on flash model. Read Docs First!
W (525) flash HPM: HPM mode with DC adjustment is disabled. Some flash models may not be supported. Read Docs First!
W (532) flash HPM: High performance mode of this flash model hasn't been supported.
I (540) MSPI Timing: Flash timing tuning index: 2
I (544) octal_psram: vendor id    : 0x0d (AP)
[... 267 lines cut ...]
a=setup:active
a=mid:1
a=sendrecv
a=sctp-port:5000
a=ice-ufrag:AtNXNaYkgyMuadyW
a=ice-pwd:nxuHEkQzlAqUDgrgcQWqloivsnKlFbfQ
Յ���`�
I (26277) HTTPS_CLIENT: HTTP POST Status = 201, content_length = 1145
I (26297) PEER_DEF: Get peer role 0
I (26297) PEER_DEF: A SRC: 1816055715
I (26297) PEER_DEF: A SRC: 1816055715
I (26307) PEER_DEF: A SRC: 1816055715
I (26307) PEER_DEF: A SRC: 1816055715
I (26307) PEER_DEF: Get peer role 0
I (26307) AGENT: 0 Add remote type:1 40.127.136.72:3478
0 Sorted pair 0 type: 1 local:192.168.137.195:0 Remote:40.127.136.72:3478
I (26317) webrtc: PeerConnectionState: 3
I (26327) webrtc: PeerConnectionState: 3
I (26337) AGENT: 0 Send binding request (cand:0) local:192.168.137.195:0 remote:40.127.136.72:3478 id:7043bfa4398150e937521657
I (26857) AGENT: 0 Send binding request (cand:0) local:192.168.137.195:0 remote:40.127.136.72:3478 id:4d9a2fdb76c3625e55075020
I (27377) AGENT: 0 Send binding request (cand:0) local:192.168.137.195:0 remote:40.127.136.72:3478 id:329b54fb381fc0385378d5a8
I (27897) AGENT: 0 Send binding request (cand:0) local:192.168.137.195:0 remote:40.127.136.72:3478 id:76e5004e797d87a62756c714
I (28417) AGENT: 0 Send binding request (cand:0) local:192.168.137.195:0 remote:40.127.136.72:3478 id:4c598d93643bd7252b449456
I (28937) webrtc: PeerConnectionState: 7
====================Event 2======================

TempoTian, 2025-02-06:
The log is not complete, you can attach the whole log, and let me check.

prothan, 2025-02-06:
On the second download, build and flash what I posted is where the console log halted. It did not continue to print like first time. I will investigate more today and tell you my findings.

filzek, 2025-02-10:
not a thread hijack, but does it work on esp32?

prothan, 2025-02-11:
I resolved my issues. Main one was located in opeai_signaling.c line 70 there is only 128 bytes allocated for the Auth Bearer line. I updated like so. 

int len = strlen((char *)sig->cfg.extra_cfg)+24; '1 extra byte
char auth[len];

Also, in main.c thread_scheduler line 170, increased stack size to 40 * 1024.

Works great, thanks!

closed 2025-02-11 by prothan

tyoh-love, 2025-02-11:
Dear @prothan, 

I'm experiencing the same issue, but I couldn't fix it by user guide above!!

Could you give us more info/complete code to fix this issue? 

Thank you!!!

TempoTian, 2025-02-12:
@prothan Thanks for your info, I will merge the change in next commit.