espressif/esp-webrtc-solution #7: Will same work on esp32-P4-NANO development board

deepakbinwal97, 2025-02-05:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I want to know if same code of doorbell demo works for ESP32-P4-Nano Board from Waveshare - https://www.waveshare.com/esp32-p4-nano.htm?sku=29027 ?

TempoTian, 2025-02-06:
Yes, I have checked the board it carried a ES8311 the same configuration with ESP32P4-Function-Ev board. You can add a section in:
[board_cfg.txt](https://github.com/espressif/esp-webrtc-solution/blob/main/components/codec_board/board_cfg.txt) like: (for you are not use the display you can remove it directly):
```
Board: ESP32_P4_NANO
i2c: {sda: 7, scl: 8}
i2s: {mclk: 13, bclk: 12, ws: 10, dout: 9, din: 11}
in_out: {codec: ES8311, pa: 53, use_mclk: 1, pa_gain:6}
camera: {type: mipi}。
```

And change the board name in [settings.h ](https://github.com/espressif/esp-webrtc-solution/blob/main/solutions/doorbell_demo/main/settings.h) to your new section name like `ESP32_P4_NANO`.

If you are using SC2336 camera no need further change, if you are using OV5647, you should use menuconfig to select the camera type: component config --> Espressif Camera Sensors Configurations --> Camera Sensor Configuration, and change the resolution to match the selected resolution in [settings.h ](https://github.com/espressif/esp-webrtc-solution/blob/main/solutions/doorbell_demo/main/settings.h)

aramix, 2025-05-24:
@TempoTian @deepakbinwal97 for NANO dout should be 11 and din 9

@TempoTian I hear the sound but in doorbell local and cloud demos I don't see the video from ESP32-P4-NANO.

I have the camera set to OV5647 and updated settings and sdkconfig to match 1920x1080 30FPS

TempoTian, 2025-05-26:
I see from log webrtc connection build success and send received OK.
```
I (67474) AV_RENDER: ARender Auto pause audio first frame at 13384214
I (68699) webrtc: Send A:1260 [64:10240] V:1193 [18:219240] Recv A:13385614 [65:10400] Recv V:[0:0]
```
I also notice there is a crash in log, you can check https://github.com/espressif/esp-webrtc-solution/issues/8 whether the IDF already have the patch or not. If the crash fixed, you can try call multiple times to check whether it still have issue or not.