espressif/esp-webrtc-solution #9: Connection failed to Openai api

tyoh-love, 2025-02-08:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

should be able to talk with openai realtime api with voice. 

### Actual behavior (suspected bug)

it seems like the connection is not successful with unknown reason. 

voice chat is not started. 

### Error logs or terminal output

```shell
======================================================================= 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4-398-g473771bc14 2nd stage bootloader
I (27) boot: compile time Feb  8 2025 21:54:15
I (27) boot: Multicore bootloader
I (28) boot: chip revision: v0.1
I (31) boot: efuse block revision: v1.2
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00300000
I (79) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3c190020 size=81bech (531436) map
I (183) esp_image: segment 1: paddr=00091c14 vaddr=3fca5c00 size=066ach ( 26284) load
I (189) esp_image: segment 2: paddr=000982c8 vaddr=40378000 size=07d50h ( 32080) load
I (196) esp_image: segment 3: paddr=000a0020 vaddr=42000020 size=18f2f0h (1635056) map
I (485) esp_image: segment 4: paddr=0022f318 vaddr=4037fd50 size=15dd0h ( 89552) load
I (504) esp_image: segment 5: paddr=002450f0 vaddr=600fe100 size=0001ch (    28) load
I (516) boot: Loaded app from partition at offset 0x10000
I (516) boot: Disabling RNG early entropy source...
W (526) flash HPM: HPM mode is optional feature that depends on flash model. Read Docs First!
W (527) flash HPM: HPM mode with DC adjustment is disabled. Some flash models may not be supported. Read Docs First!
W (534) flash HPM: High performance mode of this flash model hasn't been supported.
I (541) MSPI Timing: Flash timing tuning index: 2
I (546) octal_psram: vendor id    : 0x0d (AP)
I (549) octal_psram: dev id       : 0x02 (generation 3)
I (554) octal_psram: density      : 0x03 (64 Mbit)
I (559) octal_psram: good-die     : 0x01 (Pass)
I (563) octal_psram: Latency      : 0x01 (Fixed)
I (568) octal_psram: VCC          : 0x01 (3V)
I (572) octal_psram: SRF          : 0x01 (Fast Refresh)
I (577) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
[... 219 lines cut ...]
* no lcd 
* attached external speaker 

* i have follow the build/flash/monitor instruction. 
* rec2play works.  

### Project release version

v5.4

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

15.3(24D60)

### Shell

ZSH

TempoTian, 2025-02-10:
Seems network issue, binding request not responsed by server.
If you can fetch network traffic to check whether UDP request is replied or not. for it is STUN request you can filter with `stun` keywords. Correct answer like belowing:

[image]

If you do not have wireshark you can patch using following script:
https://[dl.espressif.com/AE/at_net_debug.py](https://dl.espressif.com/AE/at_net_debug.py
```shell
python <path to at_net_debug.py> --udp-rx-port 3478
```

TempoTian, 2025-02-10:
PS: Some crowded routes may limit UDP data transfer. You can try using a less crowded route and check if the issue still occurs.

tyoh-love, 2025-02-12:
fixed by #6  solution!!!  

Thank you!!

closed 2025-02-12 by tyoh-love