espressif/esp-webrtc-solution #11: Support for ESP32 S3 Box 3

alikian, 2025-02-10:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Thanks for adding Support for ESP32 S3 Box quickly, please also add
Support for **ESP32 S3 Box 3**
Some pins are different 

### Use cases

Testing Support for ESP32 S3 Box 3

TempoTian, 2025-02-10:
Comparing with esp32s3 box only ws field need change, take following differ as refer:

[image]

TempoTian, 2025-02-10:
By the way, through change `board_cfg.txt` as database, and change boardname in `settings.h`, is it convinient for you Comparing with use `esp-bsp`?  The demo code use a simple board realization for fast verification of  media related solution, for complex project still recommend to use `esp-bsp` for it supports all peripherals that connected to the supported board.

TempoTian, 2025-02-10:
@lmartinez51 Great! Thanks for your demo.

closed 2025-02-10 by wujiangang