espressif/esp-webrtc-solution #13: DTLS Role is always server

amr-kasem, 2025-02-12:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I'm building a system of 2 esp32 peers that should connect over webrtc 
I've built a custom implementation signaling using mqtt
intiator peer should be actpass or passive while the other must be active 

### Actual behavior (suspected bug)

instead the caller (intiator peer) sends offer sdp with role passive 
and the callee responds with answer with passive also which must be active
therefore handshake fails due to timeout waiting.

### Error logs or terminal output

```shell
Caller logs
I (29167) MQTT_SIGNALING: Received Data: {"peer_id": "ba310f14-f23c-43ca-a635-8920b2c81403", "type": "request_offer", "sdp": "v=0\r\no=- 3948308635 3948308635 IN IP4 0.0.0.0\r\ns=-\r\nt=0 0\r\na=group:BUNDLE 0\r\na=msid-semantic:WMS *\r\nm=application 38474 DTLS/SCTP 5000\r\nc=IN IP4 192.168.1.17\r\na=mid:0\r\na=sctpmap:5000 webrtc-datachannel 65535\r\na=max-message-size:65536\r\na=candidate:97335fc4d7cfcbb401af0e4689511738 1 udp 2130706431 192.168.1.17 38474 typ host\r\na=end-of-candidates\r\na=ice-ufrag:kwnB\r\na=ice-pwd:w3qV4gt7VHGYv4Snsc41pD\r\na=fingerprint:sha-256 82:59:B6:BA:02:C1:89:32:DE:D6:40:8C:EB:E7:C0:8C:4A:CF:BD:52:35:82:57:2E:8B:4B:92:56:D2:17:91:E7\r\na=setup:active"}
I (29227) MQTT_SIGNALING: Updated peer topic: /f47ac10b-58cc-4372-a567-0e02b2c3d479/ba310f14-f23c-43ca-a635-8920b2c81403
I (29237) MQTT_SIGNALING: Received request_offer, will send offer to /f47ac10b-58cc-4372-a567-0e02b2c3d479/ba310f14-f23c-43ca-a635-8920b2c81403
I (29247) MQTT_SIGNALING: Signaling ice info handler
I (29247) MQTT_SIGNALING: processing SDP
I (29257) AGENT: 0 Add remote type:1 192.168.1.17:38474
I (29277) PEER_DEF: Start DTLS role as 1
I (29277) AGENT: Send STUN binding request
I (29277) AGENT: Send allocate now
Got error code 401
I (29287) AGENT: Send allocate now
I (29287) AGENT: 0 Get candidate success user:myuser psw:mypassword
0 Sorted pair 0 type: 1 local:192.168.1.16:53667 Remote:192.168.1.17:38474
I (29287) MQTT_SIGNALING: Received SDP from peer
I (29297) MQTT_SIGNALING: Sending SDP message v=0
o=- 7801614086411001801 2 IN IP4 0.0.0.0
s=-
t=0 0
a=msid-semantic: esp-webrtc
a=group:BUNDLE 0
m=audio 9 UDP/TLS/RTP/SAVPF 8
c=IN IP4 0.0.0.0
a=mid:0
a=sendrecv
a=rtcp-mux
a=rtpmap:8 PCMA/8000
a=ssrc:4 cname:EspSSpbA0
a=inactive
a=fingerprint:sha-256 29:B0:C0:EA:A5:75:C4:61:F2:79:02:7A:FE:DF:D8:FD:51:7D:87:CF:1E:A6:DF:61:0B:89:DD:D7:92:2D:8F:E2
a=setup:passive
a=ice-ufrag:WwUl
a=ice-pwd:Pd9ZwfQgpeN3WiStdYMx5Ru0
a=candidate:0 1 UDP 1691460607 192.168.1.16 53667 typ srflx raddr 0.0.0.0 rport 0
a=candidate:1 1 UDP 16370175 192.168.1.17 63945 typ relay raddr 0.0.0.0 rport 0
a=candidate:2 1 UDP 2127668223 192.168.1.16 53667 typ host

W (29377) MQTT_SIGNALING: Unknown event
I (29387) AGENT: 0 Send binding request (cand:0) local:192.168.1.16:53667 remote:192.168.1.17:38474 id:329b54fb381fc0385378d5a8
I (29757) MQTT_SIGNALING: MQTT_DATA: Topic=/f47ac10b-58cc-4372-a567-0e02b2c3d479/123e4567-e89b-12d3-a456-426614174000
I (29757) MQTT_SIGNALING: Received Data: {"type":"answer","sdp":"v=0\r\no=- 7801614086411001801 2 IN IP4 0.0.0.0\r\ns=-\r\nt=0 0\r\na=msid-semantic: esp-webrtc\r\na=group:BUNDLE 0\r\nm=audio 9 UDP/TLS/RTP/SAVPF 8\r\nc=IN IP4 0.0.0.0\r\na=mid:0\r\na=sendrecv\r\na=rtcp-mux\r\na=rtpmap:8 PCMA/8000\r\na=ssrc:4 cname:EspSSpbA0\r\na=inactive\r\na=fingerprint:sha-256 1D:79:12:DF:BD:0F:B1:70:FA:88:67:E9:00:69:04:6F:7A:3F:92:AE:35:20:39:42:9E:88:B3:1B:32:98:7C:7F\r\na=setup:passive\r\na=ice-ufrag:WwUl\r\na=ice-pwd:Pd9ZwfQgpeN3WiStdYMx5Ru0\r\na=candidate:0 1 UDP 1693908223 192.168.1.19 63228 typ srflx raddr 0.0.0.0 rport 0\r\na=candidate:1 1 UDP 16233471 192.168.1.17 63411 typ relay raddr 0.0.0.0 rport 0\r\na=candidate:2 1 UDP 2130115839 192.168.1.19 63228 typ host\r\n","peer_id":"ba310f14-f23c-43ca-a635-8920b2c81403"}
I (29827) MQTT_SIGNALING: *************************************
E (29827) AGENT: 0 Not found for bind remote 192.168.1.19:63228
I (29827) MQTT_SIGNALING: SDP: v=0
o=- 7801614086411001801 2 IN IP4 0.0.0.0
[... 329 lines cut ...]
try to construct connection between two esp32 peers

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

UBUNTU 22.04

### Shell

Bash

### Additional context

I can try to help if you provide libpeer default implementation source code

TempoTian, 2025-02-13:
You can check peer_demo as refer (this demo use 2 esp32 series boards), one as controlling one as controlled. Pay attention to the field `.role = info->is_initiator ? ESP_PEER_ROLE_CONTROLLING : ESP_PEER_ROLE_CONTROLLED,`
For client your signaling need report correct roles, if use esp-webrtc API need set is_initiator to false, if use peer default directly need set correct role.
https://github.com/espressif/esp-webrtc-solution/blob/main/solutions/peer_demo/main/webrtc.c#L179

amr-kasem, 2025-02-13:
Thank you for your response
I did modify info->is_initiator but the sdp generated always have passive role
I also tried to use libpeer directly but I got handshake error ERROR  
 /home/amr/my-apps/oip/libs/libpeer/src/dtls_srtp.c      408     failed! mbedtls_ssl_handshake returned -0x7280 
any ideas ?

TempoTian, 2025-02-14:
libpeer need modify many things to support controlled role. you can attach your code if not secure, I can help to check the possible reason, you can run [peer_demo](https://github.com/espressif/esp-webrtc-solution/tree/main/solutions/peer_demo) directly to check the flow for controlled.

amr-kasem, 2025-02-14:
Yes I've modified libpeer to handle this I've made a fork and requested a pull. 
But I've another problem I'm trying to build with limited memory without psram because I need to connect simultaneously with 2 spi devices 
the default data buffer size is 100k I need to lower that to 50k 
but this miss up the ice exchange I don't know why 
any ideas?
I'll make a repo with the code shortly.

TempoTian, 2025-02-14:
I have check the PR https://github.com/sepfy/libpeer/pull/199/files, still need other change to support controlled mainly in dtls_srtp.c to support data channel. Run webrtc without PSRAM is challenge, but it's possible to send only simple text without audio data (which without encoding and decoding). We are working on it, but it may take some time, for it's not so urgent currently. You can develop with PSRAM version, and update the code once we finish the refinement.

TempoTian, 2025-07-14:
@voluntas Currently already have configuration for it, and also have been verified by livekit (https://github.com/livekit-examples/esp-webrtc-solution/blob/sub-connection/solutions/livekit_demo/components/livekit/core/peer.c#L302)
You can set and can verify whether it meets your needs, if meets any issue, please also let me know.

closed 2025-09-16 by wujiangang