espressif/esp-webrtc-solution #15: Input codec not exists on board

AllenXPC, 2025-02-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello, I am encountering an issue while running the OpenAI demo project on an ESP32-S3 board. After compiling and flashing the code, I receive the following error in the logs:
### Hardware **Used:**
ESP32-S3 WROOM development board
INMP441 microphone (I2S)
PCM5100A speaker (I2S)
### **Error logs or terminal output**
```
I (967) Board: Init board.
i2s: {mclk: -1, bclk: 5, ws: 46, dout: 4, din: -1}
i2s: {mclk: -1, bclk: 41, ws: 42, dout: -1, din: 2}
out: {codec: DUMMY, pa: 15, i2c_port: -1}
Codec 0 dir 2 type:5
in: {codec: DUMMY, use_mclk: 0, i2c_port: -1}
Codec 1 dir 1 type:5
sdcard: {clk: 39, cmd: 38, d0: 40}
Sdcard clk:39 cmd:38 d0:40
camera: {type: dvp, xclk: 15, pclk: 13, vsync: 6, href: 7, d0: 11, d1: 9, d2: 8, d3: 10, d4: 12, d5: 18, d6: 17, d7: 16}
D (1007) i2c.common: new bus(0) at 0x3c220c94
D (1007) i2c.common: bus clock source frequency: 40000000hz
I (1017) CODEC_INIT: Set mater handle 0 0x3c2209a0
E (1017) CODEC_INIT: in:1 out:1 port: 1
I (1027) CODEC_INIT: Success to int i2c: -1
I (1027) CODEC_INIT: Init i2s 0 type: 3 mclk:-1 bclk:5 ws:46 din:-1 dout:4
D (1037) i2s_common: tx channel is registered on I2S0 successfully
D (1037) i2s_common: rx channel is registered on I2S0 successfully
I (1047) CODEC_INIT: tx:0x3c220e20 rx:0x3c220fe4
D (1047) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 1920
D (1057) i2s_std: Clock division info: [sclk] 160000000 Hz [mdiv] 39 [mclk] 4096000 Hz [bdiv] 4 [bclk] 1024000 Hz
D (1067) i2s_std: The tx channel on I2S0 has been initialized to STD mode successfully
I (1077) CODEC_INIT: output init std ret 0
D (1087) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 3840
W (1097) i2s_tdm: the current mclk multiple is too small, adjust the mclk multiple to 384
D (1097) i2s_tdm: Clock division info: [sclk] 160000000 Hz [mdiv] 26 [mclk] 6144000 Hz [bdiv] 3 [bclk] 2048000 Hz
D (1117) i2s_tdm: The rx channel on I2S0 has been initialized to TDM mode successfully
I (1117) CODEC_INIT: Input init tdm ret 0
D (1127) i2s_common: i2s tx channel enabled
D (1127) i2s_common: i2s rx channel enabled
I (1127) CODEC_INIT: Init i2s 0 ok
I (1137) CODEC_INIT: Success to init i2s: 0
I (1137) CODEC_INIT: Success to int i2c: -1
I (1137) CODEC_INIT: Success to init i2s: 0
I (1147) CODEC_INIT: Get out handle 0x3c220e20 port 0
I (1147) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (1157) BOARD: Input codec not exits on board
```
Please let me know if you need more information or if I can provide any additional logs to help resolve this issue.
Thanks!

TempoTian, 2025-02-17:
Old code is wrong and fix to correct log output in https://github.com/espressif/esp-webrtc-solution/commit/8634ac620decc587aaa6ab248529dbbe3526985e
I have use following test config seems work use `rec2play` cli.
```
Board: ESP32_TEST
i2s: {mclk: -1, bclk: 5, ws: 46, dout: 4, din: -1}
i2s: {mclk: -1, bclk: 41, ws: 42, dout: -1, din: 2}
out: {codec: DUMMY, pa: 15}
in: {codec: DUMMY, i2s_port:1}
```

AllenXPC, 2025-02-17:
Thank U

closed 2025-09-16 by wujiangang