espressif/esp-webrtc-solution #17: OpenAI demo error invalid SDP offer

kungfupizza, 2025-02-19:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Run RTC via OpenAI

### Actual behavior (suspected bug)

ON `start`, I see following error
```
esp> I (36290) esp-x509-crt-bundle: Certificate validated
Get remote SDP {"error":{"message":"Invalid SDP offer. Please try again with a WHIP-compliant offer.","type":"invalid_request_error","param":null,"code":"invalid_offer"}}
I (37420) HTTPS_CLIENT: HTTP POST Status = 400, content_length = 155
SDP not contain candidate continue
```

I checked `rec2play` and works ok

### Error logs or terminal output

```shell
esp> I (36290) esp-x509-crt-bundle: Certificate validated
Get remote SDP {"error":{"message":"Invalid SDP offer. Please try again with a WHIP-compliant offer.","type":"invalid_request_error","param":null,"code":"invalid_offer"}}
I (37420) HTTPS_CLIENT: HTTP POST Status = 400, content_length = 155
SDP not contain candidate continue
```

### Steps to reproduce the behavior

Run OpenAI example on Korvo2 board with OpenAI 4o model API key and WiFi creds

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

Bash

tyoh-love, 2025-02-19:
I got the same error for me as well !!!

kungfupizza, 2025-02-19:
@TempoTian can you have a look at this?

TempoTian, 2025-02-19:
@kungfupizza I have found the reason and will commit the code soon. wait for a moment.

kungfupizza, 2025-02-19:
Thank you! I'll keep an eye.

closed 2025-02-19 by wujiangang

sskelapure, 2025-02-19:
@TempoTian still an error while building, as below

```shell
/Users/sarthak.kelapure/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/sarthak.kelapure/GitRepos/esp-rtc/esp-webrtc-solution/components/esp_webrtc/impl/peer_default/libs/esp32s3/libpeer_default.a(peer_default.c.obj):(.literal.peer_query+0xc): undefined reference to `esp_log'
/Users/sarthak.kelapure/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/sarthak.kelapure/GitRepos/esp-rtc/esp-webrtc-solution/components/esp_webrtc/impl/peer_default/libs/esp32s3/libpeer_default.a(peer_default.c.obj): in function `peer_query':
/home/tempo/test/esp-webrtc/components/esp_webrtc/impl/peer_default/core/peer_default.c:1064:(.text.peer_query+0x3d): undefined reference to `esp_log'
/Users/sarthak.kelapure/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/sarthak.kelapure/GitRepos/esp-rtc/esp-webrtc-solution/components/esp_webrtc/impl/peer_default/libs/esp32s3/libpeer_default.a(peer_default.c.obj): in function `peer_prepare_connection':
/home/tempo/test/esp-webrtc/components/esp_webrtc/impl/peer_default/core/peer_default.c:667:(.text.peer_prepare_connection+0x71): undefined reference to `esp_log'
/Users/sarthak.kelapure/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /home/tempo/test/esp-webrtc/components/esp_webrtc/impl/peer_default/core/peer_default.c:680:(.text.peer_prepare_connection+0xde): undefined reference to `esp_log'
/Users/sarthak.kelapure/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/sarthak.kelapure/GitRepos/esp-rtc/esp-webrtc-solution/components/esp_webrtc/impl/peer_default/libs/esp32s3/libpeer_default.a(peer_default.c.obj): in function `peer_prepare':
/home/tempo/test/esp-webrtc/components/esp_webrtc/impl/peer_default/core/peer_default.c:748:(.text.peer_prepare+0x38): undefined reference to `esp_log'
/Users/sarthak.kelapure/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/sarthak.kelapure/GitRepos/esp-rtc/esp-webrtc-solution/components/esp_webrtc/impl/peer_default/libs/esp32s3/libpeer_default.a(peer_default.c.obj):/home/tempo/test/esp-webrtc/components/esp_webrtc/impl/peer_default/core/peer_default.c:814: more undefined references to `esp_log' follow
```

kungfupizza, 2025-02-19:
Seeing the same as above

TempoTian, 2025-02-19:
@kungfupizza Let me check

TempoTian, 2025-02-19:
Seems new IDF have changed some API name, I will try to build use old IDF and have a try.

TempoTian, 2025-02-19:
try this commit https://github.com/espressif/esp-webrtc-solution/commit/1958ec4ea10076e54447ab7c82a3c98b6044354e

kungfupizza, 2025-02-19:
Builds well now and does not have that issue as before, thank you! Although the speaker doesn't work for me in Korvo2 but it could be the speaker so I am getting a speaker that looks like the one in your video.