espressif/esp-webrtc-solution #18: SCTP recv negative openai_demo

kungfupizza, 2025-02-23:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

SCTP: Recv: <positive number> ACK:<positive number>
And audio being played from OpenAI

### Actual behavior (suspected bug)

No audio being played from the speaker, seeing
SCTP: Recv:<negative number> ACK:<negative number> Cached:0

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Build latest of main, flash onto Korvo2 board

### Project release version

Main

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Mac

### Shell

ZSH

TempoTian, 2025-02-24:
You can try to use CLI `rec2play` (It will play the record PCM directly) to check whether the record to play path works all right or not. From the log, P2P connection is already built up. Server return data but all data is mute frame (100 packets each 26bytes). Suspect record data is abnormal for your case.

kungfupizza, 2025-02-24:
Thank you for the response. `rec2play` worked fine andi also found my problem. I had low OpenAI credits so it wouldn't do it's thing, i had an email from OpenAI which I ignored but after i put more credits into it, I have got it working.

Thank you for the great work. Although, I am planning to get a PR for some sanity changes in the example.

closed 2025-02-24 by kungfupizza