espressif/esp-webrtc-solution #19: Microphone doesn't work

alikian, 2025-02-24:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I have Xiao ESP32 S3 and Using MAX98357 as i2s speaker and INMP441 as i2c microphone
I can hear OpenAI no issue but there is issue in Mic, here is my config

```
Board: SIMPLE
i2s: {mclk: -1, bclk: 6, ws: 43, dout: 5, din: -1}
i2s: {mclk: -1, bclk: 3, ws: 1, dout: -1, din: 2}
out: {codec: DUMMY, use_mclk: 0}
in: {codec: DUMMY, i2s_port: 1} 
```

First question is good, but after first question, OpenAI doesn't receive my voice

Thanks for your support in advance

### Actual behavior (suspected bug)

OpenAI doesn't receive my voice correctly, is this due to noise to some buffering issue?
times to times works fine

### Error logs or terminal output

```shell
i2s: {mclk: -1, bclk: 6, ws: 43, dout: 5, din: -1}
i2s: {mclk: -1, bclk: 3, ws: 1, dout: -1, din: 2}
out: {codec: DUMMY, use_mclk: 0}
Codec 0 dir 2 type:5
in: {codec: DUMMY, i2s_port: 1}
Codec 1 dir 1 type:5
I (985) CODEC_INIT: Set mater handle 0 0x3c2209a0
I (985) CODEC_INIT: in:1 out:1 port: 0
I (985) CODEC_INIT: Success to int i2c: 0
I (995) CODEC_INIT: Init i2s 0 type: 2 mclk:-1 bclk:6 ws:43 din:-1 dout:5
I (995) CODEC_INIT: tx:0x3c220e20 rx:0x0
I (1005) CODEC_INIT: output init std ret 0
I (1005) CODEC_INIT: Init i2s 0 ok
I (1005) CODEC_INIT: Success to init i2s: 1
I (1015) CODEC_INIT: Success to int i2c: 0
I (1015) CODEC_INIT: Init i2s 1 type: 1 mclk:-1 bclk:3 ws:1 din:2 dout:-1
W (1025) i2s_platform: i2s controller 0 has been occupied by i2s_driver
I (1025) CODEC_INIT: tx:0x0 rx:0x3c22101c
W (1035) i2s_tdm: the current mclk multiple is too small, adjust the mclk multiple to 384
I (1045) CODEC_INIT: Input init tdm ret 0
I (1045) CODEC_INIT: Init i2s 1 ok
I (1045) CODEC_INIT: Success to init i2s: 1
I (1055) CODEC_INIT: Get out handle 0x3c220e20 port 0
I (1055) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1065) CODEC_INIT: Get in handle 0x3c22101c port 1
I (1065) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```

### Steps to reproduce the behavior

Just build and flash

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 24.04

### Shell

Bash

alikian, 2025-02-24:
And negative Rec and ACK
```
I (107265) webrtc: Send A:99860 [101:22725] Recv A:130222745 [101:2626]
I (107265) PEER_DEF: Send 102:25194 d_out:0 receive 210:5072 din:0
I (107265) SCTP: Send: 1238 Ack: 1237 Cached: 0 Receive:0
I (107285) SCTP: Recv: -1221715405 ACK:-1221715405 Cached: 0

```

TempoTian, 2025-02-24:
Does your board have AEC reference feedback which means the demo need I2S input one channel to be speaker output signal ?  If possible you can feed back speaker output signal to the I2S other channel input.  
If you do not want to change the circuit, can change the code to disable AEC, but the communication is not so good for maybe openai will talk to himself. Below is the code to disable AEC: https://github.com/espressif/esp-webrtc-solution/blob/main/solutions/openai_demo/main/media_sys.c#L63
```c
    esp_capture_audio_aec_src_cfg_t codec_cfg = {
        .record_handle = get_record_handle(),
    };
    capture_sys.aud_src = esp_capture_new_audio_codec_src(&codec_cfg);
```

TempoTian, 2025-02-24:
Also you can use `rec2play` CLI to confirm whether the echo sound can play normally.

alikian, 2025-02-24:
Thanks @TempoTian but is not clear what should change here, it would be great if you add a flag to disable AEC
```
    esp_capture_audio_aec_src_cfg_t codec_cfg = {
        .record_handle = get_record_handle(),
#if CONFIG_IDF_TARGET_ESP32S3
        .channel = 4,
        .channel_mask = 1 | 2,
#endif
    };
    // capture_sys.aud_src = esp_capture_new_audio_codec_src(&codec_cfg);
    capture_sys.aud_src = esp_capture_new_audio_aec_src(&codec_cfg);
```

closed 2025-09-16 by wujiangang