espressif/esp-webrtc-solution #21: source code?

richlegrand, 2025-02-24:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Nice work y'all :) 
I've been using libpeer and using my WebRTC knowledge where I'm most needed.  I'm excited to explore and contribute to esp_webrtc.  However, the core source code is missing.  Can you share what your plans are for the code?

thanks

### Use cases

Having the core source code is necessary to extend features, find issues and generally improve things, but perhaps more importantly, I won't invest time in a solution that I may be unable to fix when something doesn't quite work for my application.

Sean-Der, 2025-02-24:
Would the modifications Espressif made to libpeer be upstreamable? 

libpeer will continue to improve. It would make it easier for Espressif's fork to stay up to date if you have less patches!

jason-mao, 2025-02-25:
Thank you for your interest in esp_webrtc. At this time, we do not have plans to open source the code; however, future decisions will be made based on evolving circumstances. If you have any specific customization requirements, please feel free to send them our way via GitHub Issues or contact our sales team at sales@espressif.com. We appreciate your understanding and look forward to potentially exploring further opportunities together.