espressif/esp-webrtc-solution #22: Option to change voice

sskelapure, 2025-02-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Have an API to set the voice as made available via OpenAI RTCs: https://platform.openai.com/docs/api-reference/realtime-sessions/create

### Use cases

Change voices on the go

TempoTian, 2025-02-26:
From the DOC you can not change voice in the middle of communication. I can commit one code to let user change voice type during init. Wait for some minutes.
```
Send this event to update the session’s default configuration. The client may send this event at any time to update any field, except for voice. However, note that once a session has been initialized with a particular model, it can’t be changed to another model using session.update
```

closed 2025-02-26 by wujiangang

TempoTian, 2025-02-26:
Code commited now you can change voice type in [ webrtc.c](https://github.com/espressif/esp-webrtc-solution/blob/main/solutions/openai_demo/main/webrtc.c)
```c
    openai_signaling_cfg_t openai_cfg = {
        .token = OPENAI_API_KEY,
        .voice = "alloy",
    };
```