espressif/esp-webrtc-solution #23: No speaker on ESP32_LYRAT_V43

nguyenhuy5997, 2025-03-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

When running example video_call, it should have RINGING SOUND on the board when receive COMMAND RING 

### Actual behavior (suspected bug)

I have test example videocall_demo with ESP32_LYRAT_V43
As readme, one board will play ringing sound when the other board send a call. But i can not hear anythings

### Error logs or terminal output

```shell
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0030,len:6276

load:0x40078000,len:15716

load:0x40080400,len:4

ho 8 tail 4 room 4

load:0x40080404,len:3860

entry 0x4008063c

I (31) boot: ESP-IDF v5.4 2nd stage bootloader

I (31) boot: compile time Mar  4 2025 14:07:05

I (31) boot: Multicore bootloader

I (32) boot: chip revision: v3.0

I (35) boot.esp32: SPI Speed      : 40MHz

I (39) boot.esp32: SPI Mode       : DIO

I (42) boot.esp32: SPI Flash Size : 4MB

I (46) boot: Enabling RNG early entropy source...

I (50) boot: Partition Table:

I (53) boot: ## Label            Usage          Type ST Offset   Length

I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000

I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000

I (72) boot:  2 factory          factory app      00 00 00010000 00300000

[... 271 lines cut ...]
```

### Steps to reproduce the behavior

always

### Project release version

v5.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

CMD

TempoTian, 2025-03-05:
There are some configuration error need correct in [board_cfg.txt](https://github.com/espressif/esp-webrtc-solution/blob/main/components/codec_board/board_cfg.txt), for esp32 need correct as below:
```
Board: ESP32_LYRAT_V43
i2c: {sda: 18, scl: 23}
i2s: {mclk: 0, bclk: 5, ws: 25, dout: 26, din: 35}
in_out: {codec: ES8388, pa: 21, use_mclk: 0, pa_gain:6}
```
To run on esp32, you need comment out video related code in [media_sys.c](https://github.com/espressif/esp-webrtc-solution/blob/main/solutions/videocall_demo/main/media_sys.c)
Also comment out video related things in [webrtc.c](https://github.com/espressif/esp-webrtc-solution/blob/main/solutions/videocall_demo/main/webrtc.c)
```
    esp_webrtc_cfg_t cfg = {
        .peer_cfg = {
            .audio_info = {
                .codec = ESP_PEER_AUDIO_CODEC_G711A,
            },
            .audio_dir = ESP_PEER_MEDIA_DIR_SEND_RECV,
            .on_custom_data = video_call_on_cmd,
            .no_auto_reconnect = true,       // No auto connect peer when signaling connected
        },
        .signaling_cfg = {
            .signal_url = url,
        },
        .peer_impl = esp_peer_get_default_impl(),
        .signaling_impl = esp_signaling_get_apprtc_impl(),
    };
```
There are some menuconfig need turn on to move some huge memory into psram.
After these change the demo can run on ESP32 also.
```
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
```

TempoTian, 2025-03-11:
@nguyenhuy5997 I have not tested OPUS on the ESP32 board yet, as the OPUS encoder and decoder do not have ASM optimization. This can result in high CPU consumption, so it is not recommended to use it. For better performance, you can use the ESP32-S3 or ESP32-P4, as they have better OPUS support. I will have someone verify whether there are any issues with the OPUS decoder or encoder on the ESP32. Thanks for the information!

closed 2025-03-13 by nguyenhuy5997