espressif/esp-webrtc-solution #25: No Audio Input/Output with INMP441 and MAX98357 on ESP32-S3 WebRTC Project

lafvintech, 2025-03-07:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I'm experiencing an issue with my ESP32-S3 WebRTC project where there is no audio input from the microphone or output through the speaker.  The WebRTC connection establishes successfully, but the audio functionality doesn't work.

### Actual behavior (suspected bug)

Board: ESP32-S3
Microphone: INMP441 (I2S microphone)
Speaker: MAX98357 (I2S amplifier)

I have configured the board settings in board_cfg as follows:

i2s: {mclk: -1, bclk: 5, ws: 4, din: -1, dout: 6}  # INMP441
i2s: {mclk: -1, bclk: 15, ws: 16, dout: 7}         # MAX98357
in: {codec: DUMMY, i2c_port: -1}                   # INMP441
out: {codec: DUMMY, pa: -1}                        # MAX98357

The ESP32-S3 boots successfully
WiFi connects and obtains an IP address
WebRTC connection establishes successfully
DTLS handshake completes
Audio front-end (AFE) initializes
But no audio input is detected from the microphone
No audio output is heard from the speaker

### Error logs or terminal output

```shell
PS F:\Github\esp-webrtc-solution\solutions\openai_demo> idf.py -p COM6 monitor
Executing action: monitor
Running idf_monitor in directory F:\Github\esp-webrtc-solution\solutions\openai_demo
Executing "D:\Espressif\python_env\idf5.4_py3.11_env\Scripts\python.exe D:\Espressif\frameworks\esp-idf-v5.4\tools/idf_monitor.py -p COM6 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 F:\Github\esp-webrtc-solution\solutions\openai_demo\build\openai_demo.elf --force-color -m 'D:\Espressif\python_env\idf5.4_py3.11_env\Scripts\python.exe' 'D:\Espressif\frameworks\esp-idf-v5.4\tools\idf.py' '-p' 'COM6'"...
--- Warning: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM6 instead...
--- esp-idf-monitor 1.5.0 on \\.\COM6 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
.뢁been supportESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (27) boot: compile time Mar  7 2025 10:58:11
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 16MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00400000
[... 381 lines cut ...]
```

### Steps to reproduce the behavior

audio functionality doesn't work.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

windows10

### Shell

PowerShell

TempoTian, 2025-03-07:
The demo has a test CLI command `rec2play`. It will play the record data directly. You can use it to detect the record path to play path works or not. From the log I can see that read from I2S OK and data size is also right. Pay attention to whether the PIN is correct or not. 16bits PCM data readed can be print in code https://github.com/espressif/esp-webrtc-solution/blob/main/components/esp_capture/src/impl/capture_audio_src/capture_aud_aec_src.c#L281

PCM write can be print in 16bits PCM at https://github.com/espressif/esp-webrtc-solution/blob/main/components/av_render/render_impl/i2s_render.c#L126

Hopes these can help you to find the causes. If data not all right, can use Logic analyzer to check whether I2S data signal on line correct or not.

lafvintech, 2025-03-07:
Hello, I've checked my hardware connections and software definitions.   After running the project, I can hear some noise and successfully connect to OpenAI's RTC service, but the microphone and speaker still aren't functioning properly.

I tried using rec2play to diagnose the issue, but it didn't help - I'm still getting noise.  I also attempted to disable AEC with the following configuration:

```c
esp_capture_audio_aec_src_cfg_t codec_cfg = {
.record_handle = get_record_handle(),
};
capture_sys.  aud_src = esp_capture_new_audio_codec_src(&codec_cfg);
```

However, this approach resulted in compilation errors.

Would you happen to have a wiring diagram and sample code for a similar hardware setup to mine?   If available, I would greatly appreciate if you could share it with me.   Thank you very much for your assistance!

lafvintech, 2025-03-07:
[image]

Looking at the image you shared of your wiring connections, I'd like to check if there might be any issues with your setup.  Could you please help me understand:

I can see your ESP32S3 development board connected to both the INMP441 microphone and MAX98357A amplifier
The connections appear to follow the pinout table you shared earlier
I notice some wiring between the components that I'd like to verify is correct
Would you mind if I compare your actual connections with the recommended pinout to identify any potential issues?  I'm particularly interested in checking the I2S connections between the ESP32S3 and both audio components, as well as confirming the power and ground connections are properly made.

TempoTian, 2025-03-10:
@lafvintech Sorry, I miss one point, please add `i2s_port` configuration for output codec, also din means read i2s data from device, dout means write data to device, you need change it accordingly.
```
i2s: {mclk: -1, bclk: 5, ws: 4, din: 6, dout: -1} # INMP441
i2s: {mclk: -1, bclk: 15, ws: 16, dout: 7} # MAX98357
in: {codec: DUMMY, i2c_port: -1} # INMP441
out: {codec: DUMMY, pa: -1, , i2s_port: 1} # MAX98357
```
If you want to disable AEC, can use following code
```c
    esp_capture_audio_codec_src_cfg_t codec_cfg = {
        .record_handle = get_record_handle(),
    };
    capture_sys.aud_src = esp_capture_new_audio_codec_src(&codec_cfg);
```

tariqalikhan, 2026-09-08:
I am a beginner (although have an idea about Arduino IDE and used ESP32 CAM and PHP and JS Programming ) and need help that how I can use ESP32-S3 WROOM CAM with Espressif webrtc for Audio & Video with INMP441 and MAX98357 to web browser with custom websocket server signaling like node.js. Tried ESP-IDF V6.0 and v5.4 but doorbell_demo or openai_demo are gives error on build...is there any guide or tutorial so a beginner can learn in easy way.

TempoTian, 2026-09-09:
@tariqalikhan You can attach the build error log, firstly let build pass and test initial example OK, then you can use AI to write the signaling to replace apprtc signaling.   
To feed video from `ESP32-S3 WROOM CAM` for it only support JPEG, if send to browser only support video over data channel, can refer https://github.com/espressif/esp-webrtc-solution/blob/main/solutions/local_jpeg_stream/README.md.
In recent update, all example code use esp_board_manager to implement the board, however you can still use the codec_board use release/v1.3 version example code.

tariqalikhan, 2026-09-10:
Hi Sir, please check below

1. Downloaded fresh v1.3 release
2. for **doorbell_demo** in settings.h I set my Wifi and PWD (nothing else changed)
3. exec statement (using IDF v5.4.4 Powershell) idf.py set-target esp32p4 (which is default) executed succesffuly
4. exec statement idf.py build and got below **Errors**

-- Configuring done (83.5s)
-- Generating done (1.1s)
-- Build files have been written to: C:/Users/alle/esp-webrtc-solution-release-v1.3/solutions/doorbell_demo/build/bootloader
[1049/1775] Building C object esp-idf/espressif__esp_asrc/...les/__idf_espressif__esp_asrc.dir/hw/src/asrc_hw_ops.c.obj
ninja: build stopped: subcommand failed.
HINT: The issue is better to resolve by replacing format specifiers to 'PRI'-family macros (include <inttypes.h> header file).
ninja failed with exit code 1, output of the command is in the C:\Users\alle\esp-webrtc-solution-release-v1.3\solutions\doorbell_demo\build\log\idf_py_stderr_output_1948 and C:\Users\alle\esp-webrtc-solution-release-v1.3\solutions\doorbell_demo\build\log\idf_py_stdout_output_1948