espressif/esp-webrtc-solution #26: if it supports http/https proxy connect?

yuanfangcn, 2025-03-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

since my esp32s3 wifi cann't connetct https://api.openai.com/xxx directly,
and so cann't test openai demo.
so need esp32s3 can connect http/https proxy，and then can use/test the demo

### Use cases

1. esp32s3 box + idf 5.4
2. box wifi: 192.168.0.11
3. proxy server ip: 192.168.0.6, http port:8888, sock5 port:1081
then how to configure or use esp32s3 wifi to connect proxy server?

TempoTian, 2025-03-17:
For proxy support you need manually change http_client code for it not support proxy setting currently.
For test you can add some hardcode like:

1:  force connection_info.path use full path
   change code in esp_http_client_set_url
```
    //Reset path and query if there are no information
    if (purl.field_data[UF_PATH].len) {
        http_utils_assign_string(&client->connection_info.path, url + purl.field_data[UF_PATH].off, purl.field_data[UF_PATH].len);
    } else {
        http_utils_assign_string(&client->connection_info.path, "/", -1);
    }
   // Add this line to force use full url
    http_utils_assign_string(&client->connection_info.path, url, strlen(url));
    HTTP_MEM_CHECK(TAG, client->connection_info.path, return ESP_ERR_NO_MEM);
```

2: change ip address and port to proxy server settings
   change code in esp_http_client_connect
```
        if (!client->is_async) {
            if (esp_transport_connect(client->transport, "192.168.0.6", 8888, client->timeout_ms) < 0) {
            //if (esp_transport_connect(client->transport, client->connection_info.host, client->connection_info.port, client->timeout_ms) < 0) {
                ESP_LOGE(TAG, "Connection failed, sock < 0");
                return ESP_ERR_HTTP_CONNECT;
            }
        } else {
            int ret = esp_transport_connect_async(client->transport, "192.168.0.6", 8888, client->timeout_ms);
            //int ret = esp_transport_connect_async(client->transport, client->connection_info.host, client->connection_info.port, client->timeout_ms);
            if (ret == ASYNC_TRANS_CONNECT_FAIL) {
```

yuanfangcn, 2025-03-24:
Added above your codes in esp_http_client.c, run it and got below:

I (2514) wifi:<ba-add>idx:0 (ifx:0, 8c:21:0a:85:9b:f0), tid:0, ssn:0, winSize:64
I (4014) esp_netif_handlers: sta ip: 192.168.0.107, mask: 255.255.255.0, gw: 192.168.0.1
I (4014) NETWORK: got ip:192.168.0.107
.......
.......
Receive local SDP
E (5884) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7280
I (5884) esp-tls-mbedtls: Certificate verified.
E (5884) esp-tls: Failed to open new connection
E (5894) transport_base: Failed to open a new connection
E (5894) HTTP_CLIENT: Connection failed, sock < 0
I (5894) HTTPS_CLIENT: HTTP_EVENT_ERROR
E (5904) HTTPS_CLIENT: HTTP POST request failed: ESP_ERR_HTTP_CONNECT
I (5904) HTTPS_CLIENT: HTTP_EVENT_DISCONNECTED
I (5904) HTTPS_CLIENT: Last esp error code: 0x801a
I (5914) HTTPS_CLIENT: Last mbedtls failure: 0x7280
E (5924) OPENAI_SIGNALING: Fail to post data to https://api.openai.com

TempoTian, 2025-03-24:
The demo code should work for http, for https support need a lot more change. As another approach, you can change PC to a vpn server and add a bridge to loop the vpn traffic into the proxy server. Then can use esp32 to connect to PC hotport directly. Support UDP for sock5 and HTTPs proxy client on esp32 need too many modifications of the code, I think the vpn gateway approach is more feasible one.

closed 2025-03-25 by yuanfangcn