espressif/esp-webrtc-solution #29: DTLS: Server handshake that's not successful

yuanfangcn, 2025-03-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I (10284) DTLS: Start to do server handshake

Works as 1
I (11824) DTLS: SRTP connected OK
I (11824) DTLS: Server handshake success
I (11824) PEER_DEF: DTLS handshake success
I (11824) webrtc: PeerConnectionState: 6

### Actual behavior (suspected bug)

idf5.4, main branch git log :2025 03 14, it's almost initial codes

### Error logs or terminal output

```shell
I (22) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (22) boot: compile time Mar 24 2025 16:57:35
I (23) boot: Multicore bootloader
I (23) boot: chip revision: v0.2
I (26) boot: efuse block revision: v1.3
I (30) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 8MB
I (41) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (54) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (61) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (67) boot:  2 factory          factory app      00 00 00010000 00300000
I (74) boot: End of partition table
I (77) esp_image: segment 0: paddr=00010020 vaddr=3c1c0020 size=84148h (541000) map
I (180) esp_image: segment 1: paddr=00094170 vaddr=3fca3f00 size=062bch ( 25276) load
I (185) esp_image: segment 2: paddr=0009a434 vaddr=40378000 size=05be4h ( 23524) load
I (191) esp_image: segment 3: paddr=000a0020 vaddr=42000020 size=1b844ch (1803340) map
I (509) esp_image: segment 4: paddr=00258474 vaddr=4037dbe4 size=16210h ( 90640) load
I (534) esp_image: segment 5: paddr=0026e68c vaddr=600fe100 size=0001ch (    28) load
I (545) boot: Loaded app from partition at offset 0x10000
I (545) boot: Disabling RNG early entropy source...
I (556) octal_psram: vendor id    : 0x0d (AP)
I (556) octal_psram: dev id       : 0x03 (generation 4)
I (556) octal_psram: density      : 0x05 (128 Mbit)
I (558) octal_psram: good-die     : 0x01 (Pass)
I (562) octal_psram: Latency      : 0x01 (Fixed)
I (567) octal_psram: VCC          : 0x00 (1.8V)
I (571) octal_psram: SRF          : 0x01 (Fast Refresh)
I (576) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (581) octal_psram: BurstLen     : 0x01 (32 Byte)
I (585) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (590) octal_psram: DriveStrength: 0x00 (1/1)
I (595) MSPI Timing: PSRAM timing tuning index: 6
I (599) esp_psram: Found 16MB PSRAM device
I (603) esp_psram: Speed: 80MHz
I (606) cpu_start: Multicore app
I (1205) esp_psram: SPI SRAM memory test OK
I (1214) cpu_start: Pro cpu start user code
I (1214) cpu_start: cpu freq: 240000000 Hz
I (1214) app_init: Application information:
I (1214) app_init: Project name:     openai_demo
[... 225 lines cut ...]
### Steps to reproduce the behavior

1.compile initial codes
2.idf.py -p port flash
3.reset

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win11

### Shell

PowerShell

TempoTian, 2025-03-25:
Can you use `idf.py -p xxxx monitor` to fetch log again, when crashed, it will enter gdb, you can use `bt` to get the corrupt callstack. I have not meet such crash stack before during handshake, hope you can give more info, Thanks!

yuanfangcn, 2025-03-25:
I (11555) AGENT: 0 Connection OK 40.86.182.178:3478
I (11555) webrtc: PeerConnectionState: 5
I (11575) DTLS: Start to do server handshake

Guru Meditation Error: Core  1 panic'ed (InstructionFetchError). Exception was unhandled.

Core  1 register dump:
PC      : 0x3fcd8640  PS      : 0x00060430  A0      : 0x82129533  A1      : 0x3c2c5150
A2      : 0x3fcd8648  A3      : 0x3fcac888  A4      : 0x00000000  A5      : 0x0000000a
A6      : 0x3c25a001  A7      : 0x3fcd8648  A8      : 0x8212883d  A9      : 0x3fcd8640
A10     : 0x3fcd8c70  A11     : 0x3c25a00b  A12     : 0x00000014  A13     : 0x00000001
A14     : 0x3fcd8648  A15     : 0x00000000  SAR     : 0x0000000c  EXCCAUSE: 0x00000002
EXCVADDR: 0x3fcd8640  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff
--- 0x400570e8: memset in ROM
0x400570f3: memset in ROM

Backtrace: 0x3fcd863d:0x3c2c5150 |<-CORRUPTED

ELF file SHA256: 2d971618b

Entering gdb stub now.
$T0b#e6GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from C:\work\mifi\esp-webrtc-solution-new\solutions\openai_demo\build\openai_demo.elf...

warning: could not convert 'main' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.
Remote debugging using \\.\COM3
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
0x40378400 in _init ()
(gdb)
(gdb)
(gdb) bt
#0  0x40378400 in _init ()
#1  0x4212883d in ssl_write_server_hello (ssl=0x3fcd8648)
    at C:/work/Espressif/frameworks/esp-idf-v5.4/components/mbedtls/mbedtls/library/ssl_tls12_server.c:2214
#2  0x42129533 in mbedtls_ssl_handshake_server_step (ssl=<optimized out>)
    at C:/work/Espressif/frameworks/esp-idf-v5.4/components/mbedtls/mbedtls/library/ssl_tls12_server.c:4353
#3  0x42124563 in mbedtls_ssl_handshake_step (ssl=0x3fcd8648)
    at C:/work/Espressif/frameworks/esp-idf-v5.4/components/mbedtls/mbedtls/library/ssl_tls.c:4560
#4  0x421245bb in mbedtls_ssl_handshake (ssl=0x3fcd8648)
    at C:/work/Espressif/frameworks/esp-idf-v5.4/components/mbedtls/mbedtls/library/ssl_tls.c:4610
#5  0x420539d5 in dtls_srtp_do_handshake (dtls_srtp=0x3fcd8640)
    at /home/tempo/test/esp-webrtc/components/esp_webrtc/impl/peer_default/dtls/dtls_srtp.c:338
#6  0x42053a44 in dtls_srtp_handshake_server (dtls_srtp=0x3fcd8640)
    at /home/tempo/test/esp-webrtc/components/esp_webrtc/impl/peer_default/dtls/dtls_srtp.c:354
#7  0x42053f37 in dtls_srtp_handshake (dtls_srtp=0x3fcd8640, addr=0x0)
    at /home/tempo/test/esp-webrtc/components/esp_webrtc/impl/peer_default/dtls/dtls_srtp.c:390
#8  0x420501af in peer_on_connecting (pc=0x3fcd85d0)
    at /home/tempo/test/esp-webrtc/components/esp_webrtc/impl/peer_default/core/peer_default.c:885
#9  0x42050735 in peer_main_loop (handle=0x3fcd85d0)
    at /home/tempo/test/esp-webrtc/components/esp_webrtc/impl/peer_default/core/peer_default.c:1103
#10 0x421a2bda in esp_peer_main_loop (handle=0x3fcd6d08)
    at C:/work/mifi/esp-webrtc-solution-new/components/esp_webrtc/src/esp_peer.c:132
#11 0x42012a09 in pc_task (arg=0x3fcd54a0)
    at C:/work/mifi/esp-webrtc-solution-new/components/esp_webrtc/src/esp_webrtc.c:251
#12 0x40385ea8 in vPortTaskWrapper (pxCode=0x420129bc <pc_task>,
    pvParameters=0x3fcd54a0)
    at C:/work/Espressif/frameworks/esp-idf-v5.4/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
(gdb)

yuanfangcn, 2025-03-25:
And it's esp-box-3, psram 8MB, flash 16MB
if you have correct sdkconfig, thx

TempoTian, 2025-03-26:
You can use `idf.py set-target esp32s3` and it will auto generate sdkconfig for you. From the crash point.
https://github.com/espressif/mbedtls/blob/98fcfd6d2cea90d306e8fde8e5bffd6087c9cda8/library/ssl_tls12_server.c#L2214C5-L2214C64
Seems code dismatched, if use auto gen sdkconfig still fails, can use master branch to have a try. I tested on
`
cfe7021801 (HEAD -> master, origin/master, origin/HEAD) Merge branch 'bugfix/h2_wifi_intr' into 'master'
09200d7df4 Merge branch 'bugfix/earse_flash_after_pairing_successfully' into 'master'
587351ec3d Merge branch 'contrib/github_pr_15119' into 'master'
fa66ebec27 Merge branch 'feat/c5_ocode_support' into 'master'
`

closed 2025-03-30 by yuanfangcn