espressif/esp-webrtc-solution #33: Support for voice ai frameworks like pipecat

nischalj10, 2025-04-11:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Thanks for the amazing work. I am curious if there are any plans to support frameworks like pipecat that make it easy to build voice AI agents? The demo works for the Open AI realtime API (which is speech to speech model) but there are a lot of great "text to speech" models with great steerability coming which have to be used in a pipeline of STT-LLM-TTS. Having support for frameworks like pipecat which are open source would be great to bring any state of the art models to embedded devices. 

Note - I am not associated with pipecat in any ways. I have been using the webrtc solution with open ai realtime api as demonstrated in the examples but we are facing some constraints with function calling, voice expresiveness etc. which moving to a  STT-LLM-TTS pipeline could solve. 

The already present webrtc stack would be readily able to support these frameworks and we would be able to use ESP32 devices as first class clients of all the innovations that are happening in the voice ai world. Pipecat has support for SmallWebRTCTransport on the server side - so if the modification of espressif client could be done - it should be able to connect to pipecat. 

### Use cases

Better function calling with powerful LLMs instead of speech to speech model of open ai realtime.
Ability to use new range of steerable text to speech models recently launched by open ai - https://openai.fm/

Sean-Der, 2025-04-12:
Hey @nischalj10 

I would love to help with this. It would be great if I could make an example with pipecat/pipecat cloud. Then users could switch between all the different providers without lock in.

Maybe we just start a new example in a PR? I have a few other things I need to finish, but if this sounds like the right path we can do this together

nischalj10, 2025-04-12:
That's awesome @Sean-Der - I am all in! I would also try actively working in this direction next week on and post updates here. PS - Thanks for your amazing work on the open ai embedded sdk and kicking off all this!

tyoh-love, 2025-04-12:
This sounds really amazing!!!! I'm also trying to make s3 working against Livekit !

nischalj10, 2025-04-12:
Awesome @tyoh-love. Any learnings/ example implementation from your work that you can post here?

Sean-Der, 2025-04-12:
I wrote a esp-idf implementation for LiveKit https://github.com/Sean-Der/embedded-sdk

The issue was two PeerConnections and a websocket was hard to manage. The OpenAI API was designed with that in mind 

Would be cool to upstream LiveKit if it would be useful !

nischalj10, 2025-04-13:
I stumbled on this - can't wait to try out. Someone made livekit work on esp32s3 (inside seeed sensecap)
https://github.com/pham-tuan-binh/wheatley-ai

Sean-Der, 2025-04-13:
Tell me how it goes @nischalj10! That repo is a superset of https://github.com/Sean-Der/embedded-sdk

So if you have issues with either tell me and would love to make it work for you :)

tyoh-love, 2025-04-13:
Dear @Sean-Der 

Yes!  could you help us to combine wheatley-ai's   Livekit protocol process into current ESP-Webrtc-Solution?? 

Wheatly-ai is customized to support Senscap Watcher device, and I'm having difficulty to combine both project together to make my esp32s3 board working!!

jason-mao, 2025-04-18:
Hi everyone, hope you don’t mind me bringing up a different topic — I’m currently thinking about how to get our device connected to the MCP protocol. Does anyone have any thoughts or suggestions?

tyoh-love, 2025-04-22:
@Sean-Der 
Hi Sean,  do you have any update this?   I'm anxiously waiting to hear from you!!!