espressif/esp-webrtc-solution #34: add a custom board support fail with i2c/io expander

hutualive, 2025-04-12:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I'm using a lichuang_dev_board(https://wiki.lckfb.com/zh-hans/szpi-esp32s3/beginner/introduction.html) to try this webrtc solution with openai demo, after figuring out some errors by cross checking, i'm stucked with i2c/io expander errors as attached error logs.
the board is using pca9557 instead of tca9554 for io expander, so i try to reuse your driver code on tca9554 to bring up the demo as soon as possible. but i do not figure how to explicitly set the i2c port and i2c address for "fake" tca9554 for the custom hardware, and the custom board follow your korvo-2 v3.0 approach, which put the pa ctrl pin under io expander(here ext1), so to make the pa/speaker work, i do not have idea how to config the file: components/codec_board/board_cfg.txt to reflect my board's implementation.

### Actual behavior (suspected bug)

by cross checking with a project working perfectly on this custom board, i suspect the custom hardware using i2c port 1, though sda and scl pin is still io1 and io2 respectively, same as i2c for codec:

void InitializeI2c() {
        // Initialize I2C peripheral
        i2c_master_bus_config_t i2c_bus_cfg = {
            .i2c_port = (i2c_port_t)1,
            .sda_io_num = AUDIO_CODEC_I2C_SDA_PIN,
            .scl_io_num = AUDIO_CODEC_I2C_SCL_PIN,
            .clk_source = I2C_CLK_SRC_DEFAULT,
            .glitch_ignore_cnt = 7,
            .intr_priority = 0,
            .trans_queue_depth = 0,
            .flags = {
                .enable_internal_pullup = 1,
            },
        };
        ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_bus_cfg, &i2c_bus_));

        // Initialize PCA9557
        pca9557_ = new Pca9557(i2c_bus_, 0x19);
    }

i have revised the code portion of components/codec_board/drv/tca9554.c as below:

static tca9554_dev_t dev_list[] = {
    {0x19, "PCA9557"},  // Updated name to reflect actual hardware
};

const static audio_codec_ctrl_if_t *i2c_ctrl;

static esp_err_t expander_dev_prob(uint8_t port)
{
    // Force port 1 for PCA9557
    port = 1;

    // Use existing configuration approach
    audio_codec_i2c_cfg_t i2c_cfg = {
        .port = port,
        .addr = dev_list[0].addr,
        .bus_handle = get_i2c_bus_handle(0)  // IMPORTANT: Use port 0's handle temporarily
    };

    i2c_ctrl = audio_codec_new_i2c_ctrl(&i2c_cfg);
    if (i2c_ctrl == NULL) {
        ESP_LOGE(TAG, "Failed to create I2C controller");
        return ESP_FAIL;
    }

    // Initialize PCA9557 based on the working code
    // First, set output port register (0x01) to 0x03
    uint8_t output_data = 0x03;
    esp_err_t ret = i2c_ctrl->write_reg(i2c_ctrl, TCA9554_OUTPUT_PORT, 1, &output_data, 1);
    if (ret != 0) {
        ESP_LOGE(TAG, "Failed to set output port, error: %d", ret);
        audio_codec_delete_ctrl_if(i2c_ctrl);
        i2c_ctrl = NULL;
        return ESP_FAIL;
    }

    // Then, set configuration register (0x03) to 0xf8
    uint8_t config_data = 0xf8;
ret = i2c_ctrl->write_reg(i2c_ctrl, TCA9554_CONFIGURATION_PORT, 1, &config_data, 1);
    if (ret != 0) {
        ESP_LOGE(TAG, "Failed to set configuration, error: %d", ret);
        audio_codec_delete_ctrl_if(i2c_ctrl);
        i2c_ctrl = NULL;
        return ESP_FAIL;
    }

    // Verify device
    uint8_t data = 0;
    if (i2c_ctrl->read_reg(i2c_ctrl, TCA9554_OUTPUT_PORT, 1, &data, 1) == 0) {
        ESP_LOGI(TAG, "Detected IO expander device at 0x%02X, name is: %s, output value: 0x%02X",
                 dev_list[0].addr, dev_list[0].name, data);
        return ESP_OK;
    }

    audio_codec_delete_ctrl_if(i2c_ctrl);
    i2c_ctrl = NULL;

    ESP_LOGE(TAG, "IO expander device has not detected");
    return ESP_ERR_NOT_FOUND;
}

### Error logs or terminal output

```shell
I (986) Board: Init board.
i2c: {sda: 1, scl: 2}
i2s: {mclk: 38, bclk: 14, ws: 13, din: 12, dout: 45}
out: {codec: ES8311, i2c_addr: 48, use_mclk: 1, pa_gain: 6, pa: ext1}
Codec 0 dir 2 type:1
in: {codec: ES7210, i2c_addr: 130}
Codec 1 dir 1 type:2
sdcard: {clk: 47, cmd: 48, d0: 21}
Sdcard clk:47 cmd:48 d0:21
camera: {type: dvp, xclk: 5, pclk: 7, vsync: 3, href: 46, d0: 16, d1: 18, d2: 8, d3: 17, d4: 15, d5: 6, d6: 4, d7: 39}
lcd: {bus: spi, extend_io: tca9554, controller: st7789, spi_bus: 2, mirror_x: 1, mirror_y: 0, swap_xy: 1, color_inv: 1, width: 320, height: 240, ctrl: 42, rst: -1, cs: ext0, dc: 39, clk: 41, mosi: 40, cmd_bits: 8, param_bits: 8}
I (1046) CODEC_INIT: Set master handle 0 0x3c240998
I (1046) CODEC_INIT: in:1 out:1 port: 1
I (1056) CODEC_INIT: Success to int i2c: 0
I (1056) CODEC_INIT: Init i2s 0 type: 3 mclk:38 bclk:14 ws:13 din:12 dout:45
I (1066) CODEC_INIT: tx:0x3c240e18 rx:0x3c240fec
I (1066) CODEC_INIT: output init std ret 0
W (1076) i2s_tdm: the current mclk multiple is too small, adjust the mclk multiple to 384
I (1076) CODEC_INIT: Input init tdm ret 0
I (1086) CODEC_INIT: Init i2s 0 ok
I (1086) CODEC_INIT: Success to init i2s: 0
I (1086) CODEC_INIT: Success to int i2c: 0
I (1096) CODEC_INIT: Success to init i2s: 0
I (1096) CODEC_INIT: Get out handle 0x3c240e18 port 0
I (1106) ES8311: Work in Slave mode
I (1106) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1116) ES7210: Work in Slave mode
I (1126) ES7210: Enable ES7210_INPUT_MIC1
I (1126) ES7210: Enable ES7210_INPUT_MIC2
I (1126) ES7210: Enable ES7210_INPUT_MIC3
I (1136) ES7210: Enable ES7210_INPUT_MIC4
I (1136) ES7210: Enable TDM mode
I (1146) CODEC_INIT: Get master handle 0 0x3c240998
E (1146) i2c.master: I2C transaction unexpected nack detected
E (1146) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (1156) i2c.master: i2c_master_multi_buffer_transmit(1193): I2C transaction failed
E (1156) I2C_If: Fail to write to dev 19
E (1166) TCA9554: Failed to set output port, error: 269

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
^@I (1276) pp: pp rom version: e7ae62f
I (1276) net80211: net80211 rom version: e7ae62f
I (1286) wifi:wifi driver task: 3fccb254, prio:23, stack:6656, core=0
I (1296) wifi:wifi firmware version: 7086521f2a
I (1296) wifi:wifi certification version: v7.0
I (1296) wifi:config NVS flash: enabled
I (1296) wifi:config nano formatting: disabled
```

### Steps to reproduce the behavior

if you have a similiar board like this lichuang_dev_board:
https://wiki.lckfb.com/zh-hans/szpi-esp32s3/beginner/introduction.html
or people want to replicate the project with similiar custom hardware
typically we will have such issue(how to adapt the project to different hardware/hardware placement)

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Ubuntu 24.04

### Shell

Bash

### Additional context

for LCD, i'm also confused how config the custom board to make it work properly, especially the parameter like backlight control, from the docs and cross checking with working code, seems the backlight control is inverted, how to set up such parameter under components/codec_board/board_cfg.txt? below my board_cfg.txt FYI:
# support in, out, in_out type
# support i2c_port, i2s_port settings
# support pa_gain, i2c_addr setting

Board: S3_Korvo_V2
i2c: {sda: 1, scl: 2}
i2s: {mclk: 38, bclk: 14, ws: 13, din: 12, dout: 45}
out: {codec: ES8311, i2c_addr: 48, use_mclk: 1, pa_gain:6, pa: ext1}
in: {codec: ES7210, i2c_addr: 130}
sdcard: {clk: 47, cmd: 48, d0: 21}
camera: {type: dvp, xclk: 5, pclk: 7, vsync: 3, href: 46, d0: 16, d1: 18, d2: 8, d3: 17, d4: 15, d5: 6, d6: 4, d7: 39}
lcd: {
    bus: spi, extend_io: tca9554, controller: st7789, spi_bus: 2,
    mirror_x: 1, mirror_y: 0, swap_xy: 1, color_inv: 1,
    width: 320, height: 240,
    ctrl: 42, rst: -1,
    cs: ext0, dc: 39, clk: 41, mosi: 40, cmd_bits: 8, param_bits: 8
}

hutualive, 2025-04-13:
update:
by design an i2c scan with trial and error process, i find the addressable i2c address is not aligned with the document: 0x30/0x31/0x32/0x33 and 0x70/0x71, from the document, 0x30 is es8311, es7210 is 0x82 and pca9557 is 0x19. so i tried with 0x31/32/33/70/71 by write output register with 0x03 and config register with 0xf8 then read back output to veify, 0x70/71  output value is 0x00 and 0x31/32/33 is 0x03, which aligned with the config, but strange why we have 3 address is working...

I (991) Board: Init board.
i2c: {sda: 1, scl: 2}
i2s: {mclk: 38, bclk: 14, ws: 13, din: 12, dout: 45}
out: {codec: ES8311, i2c_addr: 48, use_mclk: 1, pa_gain: 6, pa: ext1}
Codec 0 dir 2 type:1
in: {codec: ES7210, i2c_addr: 130}
Codec 1 dir 1 type:2
sdcard: {clk: 47, cmd: 48, d0: 21}
Sdcard clk:47 cmd:48 d0:21
camera: {type: dvp, xclk: 5, pclk: 7, vsync: 3, href: 46, d0: 16, d1: 18, d2: 8, d3: 17, d4: 15, d5: 6, d6: 4, d7: 39}
lcd: {bus: spi, extend_io: tca9554, controller: st7789, spi_bus: 2, mirror_x: 1, mirror_y: 0, swap_xy: 1, color_inv: 1, width: 320, height: 240, 
ctrl: 42, rst: -1, cs: ext0, dc: 39, clk: 41, mosi: 40, cmd_bits: 8, param_bits: 8}
I (1051) CODEC_INIT: Set master handle 0 0x3c240998
I (1051) CODEC_INIT: in:1 out:1 port: 1
I (1061) CODEC_INIT: Success to int i2c: 0
I (1061) CODEC_INIT: Init i2s 0 type: 3 mclk:38 bclk:14 ws:13 din:12 dout:45
I (1071) CODEC_INIT: tx:0x3c240e18 rx:0x3c240fec
I (1071) CODEC_INIT: output init std ret 0
W (1081) i2s_tdm: the current mclk multiple is too small, adjust the mclk multiple to 384
I (1081) CODEC_INIT: Input init tdm ret 0
I (1091) CODEC_INIT: Init i2s 0 ok
I (1091) CODEC_INIT: Success to init i2s: 0
I (1091) CODEC_INIT: Success to int i2c: 0
I (1101) CODEC_INIT: Success to init i2s: 0
I (1101) CODEC_INIT: Get out handle 0x3c240e18 port 0
I (1111) ES8311: Work in Slave mode
I (1111) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1121) ES7210: Work in Slave mode
I (1131) ES7210: Enable ES7210_INPUT_MIC1
I (1131) ES7210: Enable ES7210_INPUT_MIC2
I (1131) ES7210: Enable ES7210_INPUT_MIC3
I (1141) ES7210: Enable ES7210_INPUT_MIC4
I (1141) ES7210: Enable TDM mode
I (1151) CODEC_INIT: Get master handle 0 0x3c240998
I (1151) TCA9554: Detected IO expander device at 0x33, name is: PCA9557, output value: 0x03
I (1271) LCD_INIT: CLK 4096 MOSI 4096 CS:-1 DC: 4096 Bus:1
I (1271) LCD_INIT: Init driver ST7789 finished

seems LCD config is not working too.

and the app crushed entering into gdb session, seems due to other errors with openai signaling:

 (2731) NETWORK: got ip:10.42.0.43
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42010f86  PS      : 0x00060c30  A0      : 0x82014034  A1      : 0x3c2434b0  
A2      : 0x3c243500  A3      : 0x3fcc73ac  A4      : 0x3fcc9090  A5      : 0x3fcc9050  
A6      : 0x00000000  A7      : 0x3fcd330c  A8      : 0x82010f86  A9      : 0x3c2434a0  
A10     : 0x3fcd330c  A11     : 0x3c243520  A12     : 0x00000020  A13     : 0x3fcd332c  
A14     : 0x00000000  A15     : 0x3fcd356c  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x42010f83:0x3c2434b0 0x42014031:0x3c2434e0 0x42013aef:0x3c243500 0x42010d1b:0x3c243540 0x4200fe8f:0x3c2435d0 0x40386c3d:0x3c2435f0

ELF file SHA256: 3bba69785

Entering gdb stub now.
$T0b#e6GNU gdb (esp-gdb) 15.2_20241112
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/dp/code/webrtc/solutions/openai_demo/build/openai_demo.elf...
Remote debugging using /dev/ttyUSB0
openai_signaling_start (cfg=0x3c243500, h=0x3fcc73ac) at /home/dp/code/webrtc/solutions/openai_demo/main/openai_signaling.c:104

hutualive, 2025-04-13:
update:
by cross checking with a working project on this board, i confirmed the config should be like this:

i2c: {sda: 1, scl: 2}
i2s: {mclk: 38, bclk: 14, ws: 13, din: 12, dout: 45}
out: {codec: ES8311, i2c_addr: 48, use_mclk: 1, pa_gain: 6, pa: ext1}
Codec 0 dir 2 type:1
in: {codec: ES7210, i2c_addr: 130}
Codec 1 dir 1 type:2
sdcard: {clk: 47, cmd: 48, d0: 21}
Sdcard clk:47 cmd:48 d0:21
camera: {type: dvp, xclk: 5, pclk: 7, vsync: 3, href: 46, d0: 16, d1: 18, d2: 8, d3: 17, d4: 15, d5: 6, d6: 4, d7: 39}
lcd: {bus: spi, extend_io: tca9554, controller: st7789, spi_bus: 3, mirror_x: 1, mirror_y: 0, swap_xy: 1, color_inv: 1, width: 320, height: 240, ctrl: 42, rst: -1, cs: ext0, dc: 39, clk: 41, mosi: 40, cmd_bits: 8, param_bits: 8}

but back to the original error:
I (991) Board: Init board.
i2c: {sda: 1, scl: 2}
i2s: {mclk: 38, bclk: 14, ws: 13, din: 12, dout: 45}
out: {codec: ES8311, i2c_addr: 48, use_mclk: 1, pa_gain: 6, pa: ext1}
Codec 0 dir 2 type:1
in: {codec: ES7210, i2c_addr: 130}
Codec 1 dir 1 type:2
sdcard: {clk: 47, cmd: 48, d0: 21}
Sdcard clk:47 cmd:48 d0:21
camera: {type: dvp, xclk: 5, pclk: 7, vsync: 3, href: 46, d0: 16, d1: 18, d2: 8, d3: 17, d4: 15, d5: 6, d6: 4, d7: 39}
lcd: {bus: spi, extend_io: tca9554, controller: st7789, spi_bus: 3, mirror_x: 1, mirror_y: 0, swap_xy: 1, color_inv: 1, width: 320, height: 240, ctrl: 42, rst: -1, cs: ext0, dc: 39, clk: 41, mosi: 40, cmd_bits: 8, param_bits: 8}
I (1051) CODEC_INIT: Set master handle 0 0x3c240998
I (1051) CODEC_INIT: in:1 out:1 port: 1
I (1061) CODEC_INIT: Success to int i2c: 0
I (1061) CODEC_INIT: Init i2s 0 type: 3 mclk:38 bclk:14 ws:13 din:12 dout:45
I (1071) CODEC_INIT: tx:0x3c240e18 rx:0x3c240fec
I (1071) CODEC_INIT: output init std ret 0
W (1081) i2s_tdm: the current mclk multiple is too small, adjust the mclk multiple to 384
I (1081) CODEC_INIT: Input init tdm ret 0
I (1091) CODEC_INIT: Init i2s 0 ok
I (1091) CODEC_INIT: Success to init i2s: 0
I (1091) CODEC_INIT: Success to int i2c: 0
I (1101) CODEC_INIT: Success to init i2s: 0
I (1101) CODEC_INIT: Get out handle 0x3c240e18 port 0
I (1111) ES8311: Work in Slave mode
I (1111) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1121) ES7210: Work in Slave mode
I (1131) ES7210: Enable ES7210_INPUT_MIC1
I (1131) ES7210: Enable ES7210_INPUT_MIC2
I (1131) ES7210: Enable ES7210_INPUT_MIC3
I (1141) ES7210: Enable ES7210_INPUT_MIC4
I (1141) ES7210: Enable TDM mode
I (1151) CODEC_INIT: Get master handle 0 0x3c240998
E (1151) i2c.master: I2C transaction unexpected nack detected
E (1151) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
E (1161) i2c.master: i2c_master_multi_buffer_transmit(1193): I2C transaction failed
E (1161) I2C_If: Fail to write to dev 19
E (1171) TCA9554: Failed to set output port, error: 269

i have modified tca9554.c portion as below:

static tca9554_dev_t dev_list[] = {
    {0x19, "PCA9557"},  // Updated name to reflect actual hardware
};

static esp_err_t expander_dev_prob(uint8_t port)
{
     // Scan both I2C ports to see which devices are available
//    i2c_simple_scan();

    // Force port 1 for PCA9557
//    port = 1;

    // Use existing configuration approach
    audio_codec_i2c_cfg_t i2c_cfg = {
        .port = port,
        .addr = dev_list[0].addr,
        .bus_handle = get_i2c_bus_handle(0)  // IMPORTANT: Use port 0's handle temporarily
    };

    i2c_ctrl = audio_codec_new_i2c_ctrl(&i2c_cfg);
    if (i2c_ctrl == NULL) {
        ESP_LOGE(TAG, "Failed to create I2C controller");
        return ESP_FAIL;
    }

    // Initialize PCA9557 based on the working code
    // First, set output port register (0x01) to 0x03
    uint8_t output_data = 0x03;
    esp_err_t ret = i2c_ctrl->write_reg(i2c_ctrl, TCA9554_OUTPUT_PORT, 1, &output_data, 1);
    if (ret != 0) {
        ESP_LOGE(TAG, "Failed to set output port, error: %d", ret);
        audio_codec_delete_ctrl_if(i2c_ctrl);
        i2c_ctrl = NULL;
        return ESP_FAIL;
    }

    // Then, set configuration register (0x03) to 0xf8
    uint8_t config_data = 0xf8;
    ret = i2c_ctrl->write_reg(i2c_ctrl, TCA9554_CONFIGURATION_PORT, 1, &config_data, 1);
    if (ret != 0) {
        ESP_LOGE(TAG, "Failed to set configuration, error: %d", ret);
        audio_codec_delete_ctrl_if(i2c_ctrl);
        i2c_ctrl = NULL;
        return ESP_FAIL;
    }

    // Verify device
    uint8_t data = 0;
    if (i2c_ctrl->read_reg(i2c_ctrl, TCA9554_OUTPUT_PORT, 1, &data, 1) == 0) {
        ESP_LOGI(TAG, "Detected IO expander device at 0x%02X, name is: %s, output value: 0x%02X",
                 dev_list[0].addr, dev_list[0].name, data);
        return ESP_OK;
    }

    audio_codec_delete_ctrl_if(i2c_ctrl);
    i2c_ctrl = NULL;

    ESP_LOGE(TAG, "IO expander device has not detected");
    return ESP_ERR_NOT_FOUND;
}

TempoTian, 2025-04-14:
I think your issue is caused by the RW bit of the i2c addr. The codec configuration is based on `esp-adf` which shift one bit for the normal devices address, for es8311 it's 0x18 and trans to 0x30, for 0x19 need trans to 0x32. You can try and change it.
```c
#ifdef USE_IDF_I2C_MASTER
    if (i2c_cfg->bus_handle == NULL) {
        return ESP_CODEC_DEV_INVALID_ARG;
    }
    i2c_device_config_t dev_cfg = {
        .dev_addr_length = I2C_ADDR_BIT_LEN_7,
        .device_address = (i2c_cfg->addr >> 1),
        .scl_speed_hz = DEFAULT_I2C_CLOCK,
    };
    int ret = i2c_master_bus_add_device(i2c_cfg->bus_handle, &dev_cfg, &i2c_ctrl->dev_handle);
```

hutualive, 2025-04-14:
thanks for your prompt reply and hint. all the i2c address need shift or just pca9557(io expander)? i'm using 0x30 for 8311 and 0x82 for 7210, from the log seems it's init properly.

i confirmed 0x32 is working for pca9557 as below log
does the lcd/io expander init correctly? as CS is -1, the lcd is not working.

I (989) Board: Init board.
i2c: {sda: 1, scl: 2}
i2s: {mclk: 38, bclk: 14, ws: 13, din: 12, dout: 45}
out: {codec: ES8311, i2c_addr: 48, use_mclk: 1, pa_gain: 6, pa: ext1}
Codec 0 dir 2 type:1
in: {codec: ES7210, i2c_addr: 130}
Codec 1 dir 1 type:2
sdcard: {clk: 47, cmd: 48, d0: 21}
Sdcard clk:47 cmd:48 d0:21
camera: {type: dvp, xclk: 5, pclk: 7, vsync: 3, href: 46, d0: 16, d1: 18, d2: 8, d3: 17, d4: 15, d5: 6, d6: 4, d7: 39}
lcd: {bus: spi, extend_io: tca9554, controller: st7789, spi_bus: 3, mirror_x: 1, mirror_y: 0, swap_xy: 1, color_inv: 1, width: 320, height: 240, 
ctrl: 42, rst: -1, cs: ext0, dc: 39, clk: 41, mosi: 40, cmd_bits: 8, param_bits: 8}
I (1049) CODEC_INIT: Set mater handle 0 0x3c240998
I (1049) CODEC_INIT: in:1 out:1 port: 1
I (1059) CODEC_INIT: Success to int i2c: 0
I (1059) CODEC_INIT: Init i2s 0 type: 3 mclk:38 bclk:14 ws:13 din:12 dout:45
I (1069) CODEC_INIT: tx:0x3c240e18 rx:0x3c240fec
I (1069) CODEC_INIT: output init std ret 0
W (1079) i2s_tdm: the current mclk multiple is too small, adjust the mclk multiple to 384
I (1079) CODEC_INIT: Input init tdm ret 0
I (1089) CODEC_INIT: Init i2s 0 ok
I (1089) CODEC_INIT: Success to init i2s: 0
I (1089) CODEC_INIT: Success to int i2c: 0
I (1099) CODEC_INIT: Success to init i2s: 0
I (1099) CODEC_INIT: Get out handle 0x3c240e18 port 0
I (1109) ES8311: Work in Slave mode
I (1109) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1119) ES7210: Work in Slave mode
I (1129) ES7210: Enable ES7210_INPUT_MIC1
I (1129) ES7210: Enable ES7210_INPUT_MIC2
I (1129) ES7210: Enable ES7210_INPUT_MIC3
I (1139) ES7210: Enable ES7210_INPUT_MIC4
I (1139) ES7210: Enable TDM mode
I (1149) CODEC_INIT: Get mater handle 0 0x3c240998
I (1149) TCA9554: Detected IO expander device at 0x32, name is: PCA9557
I (1269) LCD_INIT: CLK 41 MOSI 40 CS:-1 DC: 39 Bus:2
I (1269) LCD_INIT: Init driver ST7789 finished

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
^@I (1429) pp: pp rom version: e7ae62f
I (1429) net80211: net80211 rom version: e7ae62f

TempoTian, 2025-04-15:
You can add some log to check the issue, [cfg_parse.c](https://github.com/espressif/esp-webrtc-solution/blob/main/components/codec_board/cfg_parse.c#L263) check pin have ext prefix add 0x1000 flag
```c
static int16_t get_pin(const char* s)
{
    char* ext = strstr(s, "ext");
    if (ext) {
        return 0x1000 + atoi(ext + 3);
    }
    return (int16_t)atoi(s);
}
```
in [lcd_init.c](https://github.com/espressif/esp-webrtc-solution/blob/main/components/codec_board/lcd_init.c#L108) read the cfg
```c
static int16_t get_hw_gpio(int16_t pin)
{
    if (pin == -1) {
        return pin;
    }
    if (pin & BOARD_EXTEND_IO_START) {
        return -1;
    }
    return pin;
}
```
in [lcd_init.c](https://github.com/espressif/esp-webrtc-solution/blob/main/components/codec_board/lcd_init.c#L101)
apply the setting
```c
static int set_pin_state(int16_t pin, bool high)
{
    if (pin & BOARD_EXTEND_IO_START) {
        extend_io_ops.set_gpio(pin, high);
    } else {
        gpio_set_level(pin, true);
    }
    return 0;
}
```

hutualive, 2025-04-15:
understood, i will try with another board as seems current board create a lot headache. they put lcd_cs and pa_ctrl on io expander, share the reset with system reset, but actually your korvo-v2 board also use io expander for ctrl(ext1), rst(ext2)and lcd_cs(ext3) which working with current code.

hutualive, 2025-04-16:
@TempoTian update:
now i can connect to the openai server and tried the demo, it works and confirmed the driver for codec(8311/7210) plus the pca9775 is working(as the pa ctrl is using ext1) , the only issue now is the lcd showed nothing.to recap again, my config as below:
i2c: {sda: 1, scl: 2}
i2s: {mclk: 38, bclk: 14, ws: 13, din: 12, dout: 45}
out: {codec: ES8311, i2c_addr: 48, pa: ext1, use_mclk: 1, pa_gain: 6}
Codec 0 dir 2 type:1
in: {codec: ES7210, i2c_addr: 130}
Codec 1 dir 1 type:2
lcd: {bus: spi, extend_io: pca9557, controller: st7789, spi_bus: 3, mirror_x: 1, mirror_y: 0, swap_xy: 1, color_inv: 0, width: 320, height: 240, ctrl: 42, rst: 0, cs: ext0, dc: 39, clk: 41, mosi: 40, cmd_bits: 8, param_bits: 8}

seems it's due to the config of lcd. as i catched the abnormal log as below:
I (1129) CODEC_INIT: Get mater handle 0 0x3c240998
I (1129) PCA9557: Detected IO expander device at 0x32, name is: PCA9557
I (1129) PCA9557: Initialized IO expander: output=0x03, config=0xf8
I (1259) LCD_INIT: CLK 4096 MOSI 4096 CS:-1 DC: 4096 Bus:2
I (1259) LCD_INIT: Init driver ST7789 finished

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
I (1419) pp: pp rom version: e7ae62f

the clk/mosi/dc pin is parsed/interpreted as 4096, here is where i do not have clue, do you have any idea? thanks

TempoTian, 2025-04-17:
I see the old log is OK, you can git diff to check whether the modification cause such issue
```
I (1269) LCD_INIT: CLK 41 MOSI 40 CS:-1 DC: 39 Bus:2
```

hutualive, 2025-04-18:
@TempoTian 
it's me again. to isolate the potential issue with lcd panel on lichuang board, i setup a bread board to do quick test with below config:
Board: BREAD_BOARD
i2s: {mclk: -1, din: 4, bclk: 5, ws: 6} #INMP441
i2s: {mclk: -1, dout: 7, bclk: 15, ws: 16} # MAX98357
in_out: {codec: DUMMY, i2s_port: 1, i2c_port: -1, pa: -1} #INMP441/MAX98357
lcd: {
    bus: spi, controller: st7789, spi_bus: 2,
    mirror_x: 0, mirror_y: 0, swap_xy: 0, color_inv: 0,
    width: 240, height: 240,
    ctrl: 18, cs: 8, dc: 19, rst: 20,
    mosi: 9, clk: 10, cmd_bits: 8, param_bits: 8
}
and disabled the aec as other issue discussed.
but the lcd seems still trouble and any hint to make such config/lcd working? thanks in advance
-->
I (937) Board: Init board.
i2s: {mclk: -1, din: 4, bclk: 5, ws: 6}
i2s: {mclk: -1, dout: 7, bclk: 15, ws: 16}
in_out: {codec: DUMMY, i2s_port: 1, i2c_port: -1, pa: -1}
Codec 0 dir 3 type:5
lcd: {bus: spi, controller: st7789, spi_bus: 2, mirror_x: 0, mirror_y: 0, swap_xy: 0, color_inv: 0, width: 240, height: 240, ctrl: 18, cs: 8, dc: 19, rst: 20, mosi: 9, clk: 10, cmd_bits: 8, param_bits: 8}
I (977) CODEC_INIT: Set mater handle 0 0x3c1f0998
I (977) CODEC_INIT: in:1 out:1 port: 1
I (977) CODEC_INIT: Success to int i2c: -1
I (987) CODEC_INIT: Init i2s 1 type: 3 mclk:-1 bclk:15 ws:16 din:-1 dout:7
I (987) CODEC_INIT: tx:0x3c1f0e18 rx:0x3c1f0fec
I (997) CODEC_INIT: output init std ret 0
W (997) i2s_tdm: the current mclk multiple is too small, adjust the mclk multiple to 384
I (1007) CODEC_INIT: Input init tdm ret 0
I (1007) CODEC_INIT: Init i2s 1 ok
I (1017) CODEC_INIT: Success to init i2s: 1
I (1017) CODEC_INIT: Success to int i2c: -1
I (1017) CODEC_INIT: Success to init i2s: 1
I (1027) CODEC_INIT: Get out handle 0x3c1f0e18 port 1
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x8201307b  A1      : 0x3fcb41d0  
A2      : 0x00001001  A3      : 0x00000001  A4      : 0x00000040  A5      : 0x3fcb4280  
A6      : 0xffffffff  A7      : 0xffffffff  A8      : 0x82012e19  A9      : 0x3fcb41b0  
A10     : 0x00000001  A11     : 0x00000001  A12     : 0x3fceeaa8  A13     : 0x3c17714e  
A14     : 0x00000001  A15     : 0x0000002c  SAR     : 0x00000016  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  
--- 0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM

Backtrace: 0xfffffffd:0x3fcb41d0 0x42013078:0x3fcb4210 0x42013175:0x3fcb4240 0x4201120f:0x3fcb42a0 0x420111b3:0x3fcb42d0 0x42167b30:0x3fcb42f0 0x403865f9:0x3fcb4320
--- 0x42013078: _init_lcd at /home/dp/code/esp-webrtc-solution/components/codec_board/lcd_init.c:354
0x42013175: board_lcd_init at /home/dp/code/esp-webrtc-solution/components/codec_board/lcd_init.c:398
0x4201120f: init_board at /home/dp/code/esp-webrtc-solution/solutions/openai_demo/main/board.c:24
0x420111b3: app_main at /home/dp/code/esp-webrtc-solution/solutions/openai_demo/main/main.c:233
0x42167b30: main_task at /home/dp/code/esp-idf/components/freertos/app_startup.c:208
0x403865f9: vPortTaskWrapper at /home/dp/code/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: e3c26e5b7

Entering gdb stub now.

hutualive, 2025-04-18:
@TempoTian
it's me again. to isolate the potential issue with lcd panel on lichuang board, i setup a bread board to do quick test with below config but comment out the lcd config first to make sure new setup is working with openai demo first:
Board: BREAD_BOARD
i2s: {mclk: -1, din: 4, bclk: 5, ws: 6} #INMP441
i2s: {mclk: -1, dout: 7, bclk: 15, ws: 16} # MAX98357
in_out: {codec: DUMMY, i2s_port: 1, i2c_port: -1, pa: -1} #INMP441/MAX98357
#lcd: {
#bus: spi, controller: st7789, spi_bus: 2,
#mirror_x: 0, mirror_y: 0, swap_xy: 0, color_inv: 0,
#width: 240, height: 240,
#ctrl: 18, cs: 8, dc: 19, rst: 20,
#mosi: 9, clk: 10, cmd_bits: 8, param_bits: 8
#}

i disabled the aec too. but i only heard the 1st greeting from openai server, seems the microphone is not working hence do not acquired my voice.

after changed to below config:
i2s: {mclk: -1, din: 4, bclk: 5, ws: 6} #INMP441
i2s: {mclk: -1, dout: 7, bclk: 15, ws: 16} #MAX98357
in: {codec: DUMMY, i2c_port: -1]]} #INMP441
out: {codec: DUMMY, i2c_port: -1, pa: -1, i2s_port: 1} #MAX98357

i can heard the greeting and chat with openai, but seems it's talk to himself directly, how can i fix this? i need use two digital mics(INMP441) to do/enable the AEC?

TempoTian, 2025-04-18:
You need add AEC to avoid it talk to himself.  You can feedback speaker output into mic input directly. let i2s input one channel is mic sound one channel is reference signal from speaker.

[image]

hutualive, 2025-04-18:
i use a jumper to short dout pin(pa) and din pin(mic), it's not working but big noise.
since both inmp441 and max98357 are i2s device, besides to use the same hardware as your board(es7210/es8311), do you have any other recommendation to make this hardware config work decently with your demo? this pa(max98357) and mic(inmp441) is simple and easy to get and mount on a bread board. thanks in advance.
i2s: {mclk: -1, din: 4, bclk: 5, ws: 6} #INMP441
i2s: {mclk: -1, dout: 7, bclk: 15, ws: 16} #MAX98357
in: {codec: DUMMY, i2c_port: -1} #INMP441
out: {codec: DUMMY, i2c_port: -1, pa: -1, i2s_port: 1}

hutualive, 2025-04-20:
i close the issue as the io expander i2c address shift one bit already solved my original issue.

closed 2025-04-20 by hutualive

TempoTian, 2025-04-22:
If you direct connect DAC output to mic input the signal is too huge, you need use resistor and capacity to let the voltage dec to accepted level as the mentioned circuit above.