espressif/esp-webrtc-solution #35: QRCode Scanning

iothub01, 2025-04-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Is there any example or reference on how to implement QR code Scanning in ESP32P4 to scan QR Code containing WiFi details and use that WiFi to connect 

### Use cases

In doorbell application, we can use it to make it more dynamic in nature.

TempoTian, 2025-04-16:
I see the component registry have https://components.espressif.com/components/espressif/quirc/versions/1.2.0?language=en but I have not tested it yet. I will ask whether some team optimized QR code scan on P4, and notify to you.

iothub01, 2025-04-21:
Any Update sir ?

TempoTian, 2025-04-21:
@iothub01 I see an example commited for esp32p4 now, you can have a try https://github.com/espressif/esp-who/tree/master/examples/qrcode_recognition

closed 2025-04-30 by iothub01

iothub01, 2025-09-15:
is there any module of qrcode scanning like that of pedestrain detect ?

TempoTian, 2025-09-15:
There is a component for it, I think can use it directly for it is C code only. https://components.espressif.com/components/espressif/quirc/versions/1.2.0/readme