espressif/esp-webrtc-solution #36: I encountered the following problem when using S3_Korvo_V2.

Eestghost, 2025-04-19:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I encountered the following problem when using S3_Korvo_V2. However, my API key has already been configured in the IDF environment.

entry 0x403c8928
I (27) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (27) boot: compile time Apr 19 2025 12:23:21
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 16MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00300000
I (78) boot: End of partition table
I (81) esp_image: segment 0: paddr=00010020 vaddr=3c1b0020 size=823d0h (533456) map
I (183) esp_image: segment 1: paddr=000923f8 vaddr=3fca5a00 size=0669ch ( 26268) load
I (189) esp_image: segment 2: paddr=00098a9c vaddr=40378000 size=0757ch ( 30076) load
I (195) esp_image: segment 3: paddr=000a0020 vaddr=42000020 size=1a4358h (1721176) map
I (499) esp_image: segment 4: paddr=00244380 vaddr=4037f57c size=163f4h ( 91124) load
I (519) esp_image: segment 5: paddr=0025a77c vaddr=600fe100 size=0001ch (    28) load
I (531) boot: Loaded app from partition at offset 0x10000
I (531) boot: Disabling RNG early entropy source...
W (541) flash HPM: HPM mode is optional feature that depends on flash model. Read Docs First!
W (541) flash HPM: HPM mode with DC adjustment is disabled. Some flash models may not be supported. Read Docs First!
W (548) flash HPM: High performance mode of this flash model hasn't been supported.
I (556) MSPI Timing: Flash timing tuning index: 2
I (560) octal_psram: vendor id    : 0x0d (AP)
I (564) octal_psram: dev id       : 0x02 (generation 3)
I (569) octal_psram: density      : 0x03 (64 Mbit)
I (573) octal_psram: good-die     : 0x01 (Pass)
I (578) octal_psram: Latency      : 0x01 (Fixed)
I (582) octal_psram: VCC          : 0x01 (3V)
I (586) octal_psram: SRF          : 0x01 (Fast Refresh)
I (591) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (596) octal_psram: BurstLen     : 0x01 (32 Byte)
I (600) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (606) octal_psram: DriveStrength: 0x00 (1/1)
I (616) MSPI Timing: PSRAM timing tuning index: 2
I (616) esp_psram: Found 8MB PSRAM device
I (618) esp_psram: Speed: 120MHz
I (621) cpu_start: Multicore app
I (858) esp_psram: SPI SRAM memory test OK
I (867) cpu_start: Pro cpu start user code
I (867) cpu_start: cpu freq: 240000000 Hz
I (867) app_init: Application information:
I (867) app_init: Project name:     openai_demo
I (871) app_init: App version:      3a52782-dirty
I (876) app_init: Compile time:     Apr 19 2025 12:22:25
I (881) app_init: ELF file SHA256:  25ab3cd5b...
I (885) app_init: ESP-IDF:          v5.4-dirty
I (889) efuse_init: Min chip rev:     v0.0
I (893) efuse_init: Max chip rev:     v0.99
I (897) efuse_init: Chip rev:         v0.2
I (901) heap_init: Initializing. RAM available for dynamic allocation:
[... 163 lines cut ...]
```

### Steps to reproduce the behavior

The compilation can pass normally, but the error occurs when running.

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

windows10

### Shell

ZSH

lorddashboard2025, 2025-04-20:
Yes, same issue here. Windows 11, latest release, Guru Meditation Error: Core 0 panic'ed (LoadProhibited). Exception was unhandled.

Eestghost, 2025-04-21:
I have a temporary solution for now. I will hardcode the API key into the `setting.h` file, and then use the hardcoded API key in `openai_signaling.c`.
When using the original firmware and hardware for communication, there is some lag. I'm not sure whether it's caused by performance issues from the Acoustic Echo Cancellation (AEC) or by network latency.

Eestghost, 2025-04-24:
> > I have a temporary solution for now. I will hardcode the API key into the `setting.h` file, and then use the hardcoded API key in `openai_signaling.c`. When using the original firmware and hardware for communication, there is some lag. I'm not sure whether it's caused by performance issues from the Acoustic Echo Cancellation (AEC) or by network latency.
> 
> Great that you found a work around. I tried to replicate the suggested temporary solution but did not succeed. I added the key to `setting.h` and changed various things in `openai_signaling.c.` but still get the `Guru Meditation Error: Core 0 panic'ed (LoadProhibited)` error. Could you let me know what you exactly changed in `openai_signaling.c.` to fix it?

Modify openai_signaling_start in openai_signaling.c

static int openai_signaling_start(esp_peer_signaling_cfg_t *cfg, esp_peer_signaling_handle_t *h)
{
    openai_signaling_t *sig = (openai_signaling_t *)calloc(1, sizeof(openai_signaling_t));
    if (sig == NULL) {
        return ESP_PEER_ERR_NO_MEM;
    }
    openai_signaling_cfg_t *openai_cfg = (openai_signaling_cfg_t *)cfg->extra_cfg;
    sig->cfg = *cfg;
    
    // Check if the token exists; otherwise, use a hard-coded token
    char *token = NULL;
    char *voice = NULL;
    
    if (openai_cfg == NULL || openai_cfg->token == NULL || strlen(openai_cfg->token) == 0) {
        ESP_LOGE(TAG, "Using hardcoded token instead of NULL token from config");
        token = "sk-svcacct-xxxxxxx";
    } else {
        token = openai_cfg->token;
    }
    
    if (openai_cfg && openai_cfg->voice) {
        voice = openai_cfg->voice;
    } else {
        voice = "alloy";
    }
    
    // alloy, ash, ballad, coral, echo sage, shimmer and verse
    get_ephemeral_token(sig, token, voice);
    // get_ephemeral_token(sig, openai_cfg->token, openai_cfg->voice ? openai_cfg->voice : "alloy");

    if (sig->ephemeral_token == NULL) {
        free(sig);
        return ESP_PEER_ERR_NOT_SUPPORT;
    }
    *h = sig;
    esp_peer_signaling_ice_info_t ice_info = {
        .is_initiator = true,
    };
    sig->cfg.on_ice_info(&ice_info, sig->cfg.ctx);
    sig->cfg.on_connected(sig->cfg.ctx);
    return ESP_PEER_ERR_NONE;
}

Also add this in setting.h

/**
 * @brief  Set OpenAI API Key
 */
#define OPENAI_API_KEY "sk-svcacct-xxxxxx"

Of course, this is a temporary plan

closed 2025-04-25 by wujiangang