espressif/esp-webrtc-solution #37: when changed ESP-BOX-3 wroom-1-N8R16 to wroom-1-N4R2,  I2C transaction timeout

yuanfangcn, 2025-04-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

bootup ok after changed esp32s3-wroom-1-N8R16 to esp32s3-wroom-1-N4R2

### Actual behavior (suspected bug)

E (1046) i2c.master: I2C transaction timeout detected
E (1046) i2c.master: s_i2c_synchronous_transaction(918): I2C transaction failed
E (1046) i2c.master: i2c_master_multi_buffer_transmit(1180): I2C transaction failed
E (1056) I2C_If: Fail to write to dev 30
E (1536) ES8311: Open fail
...
E (1746) I2C_If: Fail to write to dev 80
E (1746) ES7210: Write register fail
E (1746) ES7210: Open fail

### Error logs or terminal output

```shell
I (27) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (27) boot: compile time Apr 21 2025 18:15:14
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 002a0000
I (78) boot:  3 storage          Unknown data     01 82 002b0000 00100000
I (85) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c1d0020 size=7b8a0h (506016) map
I (184) esp_image: segment 1: paddr=0008b8c8 vaddr=3fca4a00 size=04750h ( 18256) load
I (188) esp_image: segment 2: paddr=00090020 vaddr=42000020 size=1ca874h (1878132) map
I (520) esp_image: segment 3: paddr=0025a89c vaddr=3fca9150 size=02b8ch ( 11148) load
I (522) esp_image: segment 4: paddr=0025d430 vaddr=40378000 size=1c9c4h (117188) load
I (549) esp_image: segment 5: paddr=00279dfc vaddr=600fe100 size=0001ch (    28) load
I (561) boot: Loaded app from partition at offset 0x10000
I (561) boot: Disabling RNG early entropy source...
I (571) esp_psram: Found 2MB PSRAM device
I (571) esp_psram: Speed: 80MHz
I (571) cpu_start: Multicore app
I (765) esp_psram: SPI SRAM memory test OK
I (774) cpu_start: Pro cpu start user code
I (774) cpu_start: cpu freq: 240000000 Hz
I (774) app_init: Application information:
I (774) app_init: Project name:     openai_demo
I (778) app_init: App version:      742ddf5-dirty
I (783) app_init: Compile time:     Apr 21 2025 18:15:49
I (788) app_init: ELF file SHA256:  cfee5ae8b...
I (792) app_init: ESP-IDF:          v5.4-dirty
I (796) efuse_init: Min chip rev:     v0.0
I (800) efuse_init: Max chip rev:     v0.99 
I (804) efuse_init: Chip rev:         v0.2
I (808) heap_init: Initializing. RAM available for dynamic allocation:
I (814) heap_init: At 3FCB15C0 len 00038150 (224 KiB): RAM
[... 176 lines cut ...]
1.changed module to esp32s3-wroom-1-N4R2
2.deleted LCD module
others same as esp-box-3(es8311, es7210...)
the compiled ,upgraded, bootup, 
error log as below.

### Project release version

idf 5.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win10

### Shell

PowerShell

TempoTian, 2025-04-22:
So you only replace the CPU other hardware keep unchanged? You can use IDF example to scan the i2s bus to probe which device are linked to CPU OK. https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2c/i2c_tools

yuanfangcn, 2025-04-22:
Below is that i2c_tools test result: (esp32s3-wroom-1-N4R2)

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x1120
load:0x403c8700,len:0x4
load:0x403c8704,len:0xb2c
load:0x403cb700,len:0x2d0c
entry 0x403c88a8
I (101) esp_psram: Found 2MB PSRAM device
I (101) esp_psram: Speed: 40MHz
I (101) cpu_start: Multicore app
I (520) esp_psram: SPI SRAM memory test OK
I (529) cpu_start: Pro cpu start user code
I (529) cpu_start: cpu freq: 160000000 Hz
I (529) app_init: Application information:
I (529) app_init: Project name:     i2c_tools
I (533) app_init: App version:      742ddf5-dirty
I (537) app_init: Compile time:     Apr 22 2025 17:56:18
I (542) app_init: ELF file SHA256:  d7672874a...
I (547) app_init: ESP-IDF:          v5.4-dirty
I (551) efuse_init: Min chip rev:     v0.0
I (555) efuse_init: Max chip rev:     v0.99 
I (559) efuse_init: Chip rev:         v0.2
I (563) heap_init: Initializing. RAM available for dynamic allocation:
I (569) heap_init: At 3FC97BE0 len 00051B30 (326 KiB): RAM
I (574) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (579) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (584) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (590) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
I (597) spi_flash: detected chip: gd
I (599) spi_flash: flash io: dio
I (602) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (609) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (616) main_task: Started on CPU0
I (646) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (646) main_task: Calling app_main()
W (796) vfs_fat_spiflash: f_mount failed (13)
I (796) vfs_fat_spiflash: Formatting FATFS partition, allocation unit size=4096
I (1036) vfs_fat_spiflash: Mounting again

 ==============================================================
 |             Steps to Use i2c-tools                         |
 |                                                            |
 |  1. Try 'help', check all supported commands               |
 |  2. Try 'i2cconfig' to configure your I2C bus              |
 |  3. Try 'i2cdetect' to scan devices on the bus             |
 |  4. Try 'i2cget' to get the content of specific register   |
 |  5. Try 'i2cset' to set the value of specific register     |
 |  6. Try 'i2cdump' to dump all the register (Experiment)    |
 |                                                            |
 ==============================================================

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
I (1176) main_task: Returned from app_main()
i2c-tools> 
i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: E (9306) i2c.master: I2C transaction timeout detected
E (9306) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up
UU E (9316) i2c.master: I2C hardware timeout detected
E (9316) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up
.....
.....
i2c-tools> i2cconfig --port=0 --sda=8 --scl=18 --freq=100000
i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: E (169836) i2c.master: I2C transaction timeout detected
E (169836) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up
.....

yuanfangcn, 2025-04-22:
BTW, found that in code:
#define ES8311_CODEC_DEFAULT_ADDR (0x30)
#define ES7210_CODEC_DEFAULT_ADDR (0x80)

but I test it on esp-box-3：
i2c-tools> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- 18 -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- 5d -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
i2c-tools> 

So if ES8311_CODEC_DEFAULT_ADDR and ES7210_CODEC_DEFAULT_ADDR are correct?

TempoTian, 2025-04-22:
If you have Logic Analyzer or Oscilloscope, you can connect to the codec input pin and check the signal OK or not. I think it is caused by the circuit, change software can not fix such issue.

closed 2025-04-23 by yuanfangcn