espressif/esp-webrtc-solution #38: DTLS Handshake Failure

DurhamSmith, 2025-04-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I have made some modifications to the openAI example to replace their server with my own server for the WebRTC demo, and to also use STUN.
I expect the DTLS handshake to pass and and the ESP32 to establish a WebRTC connection with my server. 
I know my server is working as I can connect to it over the browser.

### Actual behavior (suspected bug)

It fails at DTLS handshake and the connection never establishes.

### Error logs or terminal output

```shell
I (3382) WIFI: Connected, calling start_webrtc() asynchronously
I (3382) openai_signaling_start-1: h pointer: 0x3fcc0df8
I (3392) openai_signaling_start-1: sig pointer: 0x3fcccffc
I (3392) DTLS: Init SRTP OK
I (5262) DTLS_SRTP: dtls_srtp init done
I (5262) webrtc: peer_connection_task started
I (5262) ESP_CAPTURE: Audio source caps negotiate done: 16000Hz, 1 channels
I (5262) webrtc: PeerConnectionState: 2
End: openai_signaling_start
I (5272) openai_signaling_start2: h pointer: 0x3fcc0df8
I (5272) AGENT: Start agent as Controlling
I (5272) openai_signaling_start2: sig pointer: 0x3fcccffc
I (5282) PEER_DEF: Start DTLS role as 1
I (5302) AGENT: Send STUN binding request
I (5322) wifi:<ba-add>idx:1 (ifx:0, c8:e3:06:e3:9c:c6), tid:4, ssn:0, winSize:64
I (5322) AGENT: 0 Get candidate success user: psw:
Add audio codec 9 sample_rate 16000
I (5332) webrtc: Send client sdp: v=0
o=- 7801614086411001801 2 IN IP4 0.0.0.0
s=-
t=0 0
a=msid-semantic: esp-webrtc
a=group:BUNDLE 0 1
m=audio 9 UDP/TLS/RTP/SAVP 111
a=rtpmap:111 opus/48000/2
a=ssrc:6 cname:webrtc-opus
a=sendrecv
a=mid:0
c=IN IP4 0.0.0.0
a=rtcp-mux
a=fingerprint:sha-256 71:5D:7F:6D:77:0B:3E:8A:DB:06:3E:A1:CC:DC:A5:FF:20:54:C2:DB:BB:47:E3:75:AD:F8:9D:55:46:FE:87:8F
a=setup:passive
a=ice-ufrag:Xd4P
a=ice-pwd:oKWwUlPd9ZwfQgpeN3WiStdY
a=candidate:0 1 UDP 1690652159 135.180.121.3 50509 typ srflx raddr 0.0.0.0 rport 0
a=candidate:1 1 UDP 2126859775 192.168.4.48 50509 typ host
m=application 50712 UDP/DTLS/SCTP webrtc-datachannel
a=mid:1
a=sctp-port:5000
c=IN IP4 0.0.0.0
a=max-message-size:262144
a=fingerprint:sha-256 71:5D:7F:6D:77:0B:3E:8A:DB:06:3E:A1:CC:DC:A5:FF:20:54:C2:DB:BB:47:E3:75:AD:F8:9D:55:46:FE:87:8F
a=setup:passive
a=ice-ufrag:Xd4P
a=ice-pwd:oKWwUlPd9ZwfQgpeN3WiStdY
a=candidate:0 1 UDP 1690652159 135.180.121.3 50509 typ srflx raddr 0.0.0.0 rport 0
a=candidate:1 1 UDP 2126859775 192.168.4.48 50509 typ host
[... 647 lines cut ...]
    ice_servers[0].psw      = NULL;

)

3. run idf.py flash build

### Project release version

5.5

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu 20.04

### Shell

Bash

TempoTian, 2025-04-23:
The log is quite strange, the handshake code is like below:
It will call `mbedtls_ssl_handshake` to do handshake, but I can not see any handshake log output from mbedtls lib, I see that you already turn on the mbedtls log level, at least `I (5572) mbedtls: ssl_tls.c:4606 => handshake` should output.
```c
static int dtls_srtp_do_handshake(dtls_srtp_t *dtls_srtp)
{
    int ret;
    static mbedtls_timing_delay_context timer;

    mbedtls_ssl_set_timer_cb(&dtls_srtp->ssl, &timer, mbedtls_timing_set_delay, mbedtls_timing_get_delay);
    mbedtls_ssl_set_export_keys_cb(&dtls_srtp->ssl, dtls_srtp_key_derivation, dtls_srtp);
    mbedtls_ssl_set_bio(&dtls_srtp->ssl, dtls_srtp, dtls_srtp->udp_send, dtls_srtp->udp_recv, NULL);

    do {
        ret = mbedtls_ssl_handshake(&dtls_srtp->ssl);
    } while (ret == MBEDTLS_ERR_SSL_WANT_READ || ret == MBEDTLS_ERR_SSL_WANT_WRITE);
    if (ret == MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY) {
        mbedtls_ssl_session_reset(&dtls_srtp->ssl);
    }
    return ret;
}

static int dtls_srtp_handshake_server(dtls_srtp_t *dtls_srtp)
{
    int ret;
    ESP_LOGI(TAG, "Start to do server handshake\n");
    while (1) {
        unsigned char client_ip[] = "test";
        mbedtls_ssl_session_reset(&dtls_srtp->ssl);
        mbedtls_ssl_set_client_transport_id(&dtls_srtp->ssl, client_ip, sizeof(client_ip));
        ret = dtls_srtp_do_handshake(dtls_srtp);
        if (ret != MBEDTLS_ERR_SSL_HELLO_VERIFY_REQUIRED) {
            if (ret != 0) {
                ESP_LOGE(TAG, "Server handshake return -0x%.4x", (unsigned int)-ret);
            }
            break;
        }
    }
    return ret;
}

```

DurhamSmith, 2025-04-23:
I see `I (5572) mbedtls: ssl_tls.c:4606 => handshake` which I assume means handshake has been initiated? I assume I am not meant to call `dtls_srtp_do_handshake` and `dtls_srtp_handshake_server` manually?

Is there any way to turn on more debugging information for handshake? I already have mbedutils and DTLS debugging set to verbose.  Are there any suggestions on what may be causing this issue?

TempoTian, 2025-04-23:
`dtls_srtp_handshake_server` is called internally, but it also call api `mbedtls_ssl_handshake`, you can add more logs in it to check the reason.