espressif/esp-webrtc-solution #45: whip示例出现mbedtls_ssl_setup returned -0x7F00

OpenWor, 2025-05-05:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

我修改了whip示例，原先的示例中不支持S3_Korvo_V2，我添加了#define TEST_BOARD_NAME "S3_Korvo_V2"，将示例中服务器替换为我自己的 WebRTC 服务器（使用了[MediaMTX](https://github.com/bluenviron/mediamtx)），并参照readme添加了STUN。我使用浏览器访问https://192.168.xx.xxx:8889/whip/publish进行推流可以顺利进行。

### Actual behavior (suspected bug)

#define WHIP_SERVER "http://XXXX/whip"修改为https://192.168.xx.xxx:8889/whip/publish，ESP32 与我的服务器无法进行建立连接。

### Error logs or terminal output

```shell
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd24
load:0x403cb700,len:0x2f00
entry 0x403c8928
I (26) boot: ESP-IDF v5.4.1-368-g8e473c6bed 2nd stage bootloader
I (27) boot: compile time Apr 29 2025 17:24:55
I (27) boot: Multicore bootloader
I (28) boot: chip revision: v0.2
I (31) boot: efuse block revision: v1.3
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00300000
I (79) boot: End of partition table
I (82) esp_image: segment 0: paddr=00010020 vaddr=3c1c0020 size=8f410h (586768) map
I (193) esp_image: segment 1: paddr=0009f438 vaddr=3fcaee00 size=00be0h (  3040) load
I (194) esp_image: segment 2: paddr=000a0020 vaddr=42000020 size=1bc7c4h (1820612) map
I (518) esp_image: segment 3: paddr=0025c7ec vaddr=3fcaf9e0 size=062bch ( 25276) load
I (523) esp_image: segment 4: paddr=00262ab0 vaddr=40378000 size=26da0h (159136) load
I (558) esp_image: segment 5: paddr=00289858 vaddr=600fe100 size=0001ch (    28) load
I (573) boot: Loaded app from partition at offset 0x10000
I (574) boot: Disabling RNG early entropy source...
W (584) flash HPM: HPM mode is optional feature that depends on flash model. Read Docs First!
W (584) flash HPM: HPM mode with DC adjustment is disabled. Some flash models may not be supported. Read Docs First!
W (591) flash HPM: High performance mode of this flash model hasn't been supported.
I (599) MSPI Timing: Flash timing tuning index: 2
I (603) octal_psram: vendor id    : 0x0d (AP)
I (607) octal_psram: dev id       : 0x02 (generation 3)
I (612) octal_psram: density      : 0x03 (64 Mbit)
I (616) octal_psram: good-die     : 0x01 (Pass)
I (620) octal_psram: Latency      : 0x01 (Fixed)
I (625) octal_psram: VCC          : 0x01 (3V)
I (629) octal_psram: SRF          : 0x01 (Fast Refresh)
I (634) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (639) octal_psram: BurstLen     : 0x01 (32 Byte)
I (643) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (648) octal_psram: DriveStrength: 0x00 (1/1)
I (662) MSPI Timing: PSRAM timing tuning index: 6
I (662) esp_psram: Found 8MB PSRAM device
I (662) esp_psram: Speed: 120MHz
I (664) cpu_start: Multicore app
[... 209 lines cut ...]
```

### Steps to reproduce the behavior

使用ESP-IDF 5.4.1，在setting.h中添加了#define TEST_BOARD_NAME "S3_Korvo_V2"。在menuconfig中勾选了Allow potentially insecure options；Skip server certificate verification by default (WARNING: ONLY FOR TESTING PURPOSE, READ HELP)；

### Project release version

最新

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Window11

### Shell

ZSH

TempoTian, 2025-05-06:
从提供的log看是分不到内存了 
```
/** Memory allocation failed */
#define MBEDTLS_ERR_SSL_ALLOC_FAILED                      -0x7F00
```
可以把MBEDTLS放到外部PSRAM看看是否OK
```
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=4096
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
```

TempoTian, 2025-05-08:
按说使能Allow potentially insecure options；Skip server certificate verification by default 可以解决Failed to verify peer certificate! 这个问题的可以确认下看选项是否关掉了

OpenWor, 2025-05-08:
已确认Allow potentially insecure options；Skip server certificate verification by default已经使能，但仍有上述错误。在下次更新中希望对whip demo增加对S3_Korvo_V2的支持，感谢您的工作。

closed 2025-05-13 by wujiangang

TempoTian, 2025-05-13:
我跑了下发现是因为开起了bundle导致的，可以在menuconfig中关掉然后开启insecure选项就可以了，
新的版本会fix关闭bundle导致的build问题，可以抓下来测试下，我local测试S3可以推流成功只是分辨率较小。

[image]

OpenWor, 2025-05-16:
非常感谢，此问题已顺利解决。但遗憾的是紧接着遇到一个问题，目前终端输出为
Get remote SDP 404 page not found�?I
I (9879) HTTPS_CLIENT: HTTP POST Status = 404, content_length = 18
SDP not contain candidate continu

我已确认在setting.h文件中#define WHIP_SERVER "https://xx.xx.xx.xxx:8889/whip/publish"是可用，通过手机和电脑均可完成推流。但ESP32 whip demo未能成功。您是否有可参考的思路？完整日志如下：
I (26) boot: ESP-IDF v5.4.1-368-g8e473c6bed 2nd stage bootloader
I (27) boot: compile time May 14 2025 20:46:20
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (27) boot: efuse block revision: v1.3
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (28) boot.esp32s3: SPI Mode       : DIO
I (28) boot.esp32s3: SPI Flash Size : 4MB
I (28) boot: Enabling RNG early entropy source...
I (28) boot: Partition Table:
I (28) boot: ## Label            Usage          Type ST Offset   Length
I (29) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (29) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (29) boot:  2 factory          factory app      00 00 00010000 00300000
I (30) boot: End of partition table
I (30) esp_image: segment 0: paddr=00010020 vaddr=3c1d0020 size=89518h (562456) map
I (130) esp_image: segment 1: paddr=00099540 vaddr=3fcaee00 size=06ad8h ( 27352) load
I (135) esp_image: segment 2: paddr=000a0020 vaddr=42000020 size=1c8f38h (1871672) map
I (466) esp_image: segment 3: paddr=00268f60 vaddr=3fcb58d8 size=00424h (  1060) load
I (466) esp_image: segment 4: paddr=0026938c vaddr=40378000 size=26dech (159212) load
I (501) esp_image: segment 5: paddr=00290180 vaddr=600fe100 size=0001ch (    28) load
I (516) boot: Loaded app from partition at offset 0x10000
I (516) boot: Disabling RNG early entropy source...
W (517) flash HPM: HPM mode is optional feature that depends on flash model. Read Docs First!
W (517) flash HPM: HPM mode with DC adjustment is disabled. Some flash models may not be supported. Read Docs First!
W (518) flash HPM: High performance mode of this flash model hasn't been supported.
I (518) MSPI Timing: Flash timing tuning index: 2
I (519) octal_psram: vendor id    : 0x0d (AP)
I (519) octal_psram: dev id       : 0x02 (generation 3)
I (519) octal_psram: density      : 0x03 (64 Mbit)
I (519) octal_psram: good-die     : 0x01 (Pass)
I (519) octal_psram: Latency      : 0x01 (Fixed)
I (520) octal_psram: VCC          : 0x01 (3V)
I (520) octal_psram: SRF          : 0x01 (Fast Refresh)
I (520) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (520) octal_psram: BurstLen     : 0x01 (32 Byte)
I (521) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (521) octal_psram: DriveStrength: 0x00 (1/1)
I (530) MSPI Timing: PSRAM timing tuning index: 2
I (530) esp_psram: Found 8MB PSRAM device
I (531) esp_psram: Speed: 120MHz
I (531) cpu_start: Multicore app
I (766) esp_psram: SPI SRAM memory test OK
I (775) cpu_start: Pro cpu start user code
I (775) cpu_start: cpu freq: 240000000 Hz
I (775) app_init: Application information:
I (775) app_init: Project name:     whip_demo
I (775) app_init: App version:      b5ef846-dirty
I (775) app_init: Compile time:     May 14 2025 20:46:01
I (775) app_init: ELF file SHA256:  2c95bd03c...
I (775) app_init: ESP-IDF:          v5.4.1-368-g8e473c6bed
I (776) efuse_init: Min chip rev:     v0.0
I (776) efuse_init: Max chip rev:     v0.99
I (776) efuse_init: Chip rev:         v0.2
I (776) heap_init: Initializing. RAM available for dynamic allocation:
I (776) heap_init: At 3FCBBCE8 len 0002DA28 (182 KiB): RAM
I (776) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (776) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (777) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (778) spi_flash: detected chip: gd
I (778) spi_flash: flash io: dio
W (778) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (778) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (779) sleep_gpio: Enable automatic switching of GPIO sleep configuration
[... 147 lines cut ...]
a=rtcp-fb:102 nack
a=rtcp-fb:102 nack pli
a=fmtp:102 level-asymmetry-allowed=1
a=rtpmap:106 H264/90000
a=rtcp-fb:106 goog-remb
a=rtcp-fb:106 transport-cc
a=rtcp-fb:106 ccm fir
a=rtcp-fb:106 nack
a=rtcp-fb:106 nack pli
a=fmtp:106 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=4d001f
a=ssrc:1 cname:webrtc-h264
a=sendrecv
a=mid:1
c=IN IP4 0.0.0.0
a=rtcp-mux
a=fingerprint:sha-256 3A:37:1F:6E:7D:EC:BE:3A:CE:20:54:B7:DD:42:24:6D:48:30:AB:FA:52:15:9A:2A:3A:59:59:05:95:4B:77:C6
a=setup:passive
a=ice-ufrag:Xd4P
a=ice-pwd:oKWwUlPd9ZwfQgpeN3WiStdY
a=candidate:0 1 UDP 2113929471 192.168.121.79 0 typ host

I (5869) wifi:<ba-add>idx:1 (ifx:0, 10:32:1d:f7:c1:a0), tid:0, ssn:0, winSize:64
Get remote SDP 404 page not found�?I
I (6559) HTTPS_CLIENT: HTTP POST Status = 404, content_length = 18
SDP not contain candidate continue

TempoTian, 2025-05-16:
可以打印下你推流的地址，这个看是return 404了，地址不对, 确认下有没有输入错误
我测试的时候用的地址是`start http://162.168.10.252:8889/stream/whip username:psw`
带basic认证的

OpenWor, 2025-05-16:
这是我的推流地址https://47.97.77.226:8889/whip/publish，账号与密码为admin:admin，在ESP32中未能成功推流，报错404。是否与前缀https有关呢？看到您使用的http，您可以通过https://47.97.77.226:8889/whip/查看到推流的视频。

TempoTian, 2025-05-21:
可以推流成功了，之前base64编码有个bug，https://github.com/espressif/esp-webrtc-solution/commit/2ac873762bc64f1cd56f18454ff58e2cd641b92c 这个change fix了

另外实际的推流网址是https://47.97.77.226:8889/whip/whip 上面的publish只是网页的地址并不是whip服务的地址

OpenWor, 2025-05-21:
感谢您的工作，目前demo已经可以正常运行。