espressif/esp-webrtc-solution #47: ESP_WEBRTC_EVENT_DISCONNECTED / ESP_WEBRTC_EVENT_CONNECT_FAILED never fired

shushenghong, 2025-05-10:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

1. use this lib as a client connect to a webrtc server 
2. shutdown the server after connect successfully
3. the client should fire ESP_WEBRTC_EVENT_DISCONNECTED / ESP_WEBRTC_EVENT_CONNECT_FAILED event, but it doesn't

### Actual behavior (suspected bug)

the client nothing happend, but continue print webrtc log

### Error logs or terminal output

```shell
I (757539) webrtc: Send A:749440 [101:22725] Recv A:94296285 [0:0]
I (757539) PEER_DEF: Send 101:24947 d_out:0 receive 4:0 din:0
I (757539) SCTP: Send: 1238 Ack: 1237 Cached: 0 Receive:0
I (757559) SCTP: Recv: -1284146317 ACK:-1284146317 Cached: 0

I (759559) webrtc: Send A:751440 [99:22275] Recv A:94296285 [0:0]
I (759559) PEER_DEF: Send 100:24700 d_out:0 receive 4:0 din:0
I (759559) SCTP: Send: 1238 Ack: 1237 Cached: 0 Receive:0
I (759559) SCTP: Recv: -1284146317 ACK:-1284146317 Cached: 0

I (761579) webrtc: Send A:753480 [100:22500] Recv A:94296285 [0:0]
I (761579) PEER_DEF: Send 102:25194 d_out:0 receive 4:0 din:0
I (761579) SCTP: Send: 1238 Ack: 1237 Cached: 0 Receive:0
I (761599) SCTP: Recv: -1284146317 ACK:-1284146317 Cached: 0

I (763599) webrtc: Send A:755500 [101:22725] Recv A:94296285 [0:0]
I (763599) PEER_DEF: Send 101:24947 d_out:0 receive 4:0 din:0
I (763599) SCTP: Send: 1238 Ack: 1237 Cached: 0 Receive:0
I (763619) SCTP: Recv: -1284146317 ACK:-1284146317 Cached: 0
```

### Steps to reproduce the behavior

as described

### Project release version

latest

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

13.6.7

### Shell

ZSH

TempoTian, 2025-05-13:
@shushenghong During test I find if close server directly it will pop up error log like:
Do you meet the same log, if so I will close connection internally to avoid continue send data.

[image]

shushenghong, 2025-05-17:
no, i don't meet this error log

shushenghong, 2025-05-28:
yes, the doorbell_local demo, if i close browser, esp board console will show the log and close the webrtc connection.

my situation is, the esp board as a webrtc client, like openai demo, connect an webrtc server, then shutdown the server, the esp board client will no detect disconnect.

shushenghong, 2025-06-07:
nice, waitting for new version

closed 2025-09-16 by wujiangang