espressif/esp-webrtc-solution #53: Clicking Photo or record video

iothub01, 2025-05-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

I want to know how to capture a photo and store it in spiffs or external memory and send it to server through MQTT or HTTP or etc.

### Use cases

we can use this as Intruder alert and send capture image to server/google drive.

iothub01, 2025-05-19:
Any update on this ?

TempoTian, 2025-05-20:
Yes, you can check the rec2play cli in https://github.com/espressif/esp-webrtc-solution/blob/main/solutions/videocall_demo/main/media_sys.c#L229
If capture only video, change the audio type to NONE then it capture only video, you can try following 2 test code.
```c
int test_capture_video(void)
{
    esp_capture_sink_cfg_t sink_cfg = {
        .video_info = { .codec = ESP_CAPTURE_CODEC_TYPE_MJPEG, .width = VIDEO_WIDTH, .height = VIDEO_HEIGHT, .fps = 20 },
    };
    // Create capture
    esp_capture_path_handle_t capture_path = NULL;
    esp_capture_setup_path(capture_sys.capture_handle, ESP_CAPTURE_PATH_PRIMARY, &sink_cfg, &capture_path);
    esp_capture_enable_path(capture_path, ESP_CAPTURE_RUN_TYPE_ALWAYS);
    uint32_t start_time = (uint32_t)(esp_timer_get_time() / 1000);
    esp_capture_start(capture_sys.capture_handle);
    while ((uint32_t)(esp_timer_get_time() / 1000) < start_time + 2000) {
        media_lib_thread_sleep(30);
        esp_capture_stream_frame_t frame = {
            .stream_type = ESP_CAPTURE_STREAM_TYPE_VIDEO,
        };
        frame.stream_type = ESP_CAPTURE_STREAM_TYPE_VIDEO;
        while (esp_capture_acquire_path_frame(capture_path, &frame, true) == ESP_CAPTURE_ERR_OK) {
            av_render_video_data_t video_data = {
                .data = frame.data,
                .size = frame.size,
                .pts = frame.pts,
            };
            av_render_add_video_data(player_sys.player, &video_data);
            esp_capture_release_path_frame(capture_path, &frame);
        }
    }
    esp_capture_stop(capture_sys.capture_handle);
    return 0;
}

int test_capture_video_once(void)
{
    esp_capture_sink_cfg_t sink_cfg = {
        .video_info = { .codec = ESP_CAPTURE_CODEC_TYPE_MJPEG, .width = VIDEO_WIDTH, .height = VIDEO_HEIGHT, .fps = 20 },
    };
    // Create capture
    esp_capture_path_handle_t capture_path = NULL;
    esp_capture_setup_path(capture_sys.capture_handle, ESP_CAPTURE_PATH_PRIMARY, &sink_cfg, &capture_path);
    esp_capture_enable_path(capture_path, ESP_CAPTURE_RUN_TYPE_ONCE);
    uint32_t start_time = (uint32_t)(esp_timer_get_time() / 1000);
    esp_capture_start(capture_sys.capture_handle);
    while (1) {
        media_lib_thread_sleep(1000);
		esp_capture_enable_path(capture_path, ESP_CAPTURE_RUN_TYPE_ONCE);
        esp_capture_stream_frame_t frame = {
            .stream_type = ESP_CAPTURE_STREAM_TYPE_VIDEO,
        };
        frame.stream_type = ESP_CAPTURE_STREAM_TYPE_VIDEO;
        if (esp_capture_acquire_path_frame(capture_path, &frame, false) == ESP_CAPTURE_ERR_OK) {
            av_render_video_data_t video_data = {
                .data = frame.data,
                .size = frame.size,
                .pts = frame.pts,
            };
            av_render_add_video_data(player_sys.player, &video_data);
            esp_capture_release_path_frame(capture_path, &frame);
        } else {
	    break;
	}
    }
    esp_capture_stop(capture_sys.capture_handle);
    return 0;
}

```

iothub01, 2025-05-23:
Thanks , it works .

iothub01, 2025-05-23:
if i run p2p webrtc first and then close connection and call this function again to capture image -
i got error -

I (60503) ESP_CAPTURE: Video source caps negotiate done: 1920x1080@25fps
I (60509) CAPTURE_SIMP: Enter audio encoder thread
I (60510) CAPTURE_SIMP: Enter video encoder thread
E (60515) CAPTURE_SIMP: Fail to acquire audio frame ret -3
I (60523) CAPTURE_SIMP: Audio encoder thread exit

and if i get capture image first and then run webrtc connection , i donot get video stream visible on other side

iothub01, 2025-05-28:
same issue , if i run get_image then during webrtc , video is not streaming and if i perform webrtc first and then cut call and then try to get image , it does not work

TempoTian, 2025-06-05:
@iothub01 You can use latest version. I reproduce this issue and fixed it in [44fdb0d](https://github.com/espressif/esp-webrtc-solution/commit/44fdb0d4826fd0553cdfe59e48acf4c1b53b0b32)

TempoTian, 2025-06-06:
If you use https://github.com/espressif/esp-webrtc-solution/issues/62#issuecomment-2943253482 can you still reproduce it?
Seems code not updated, update to latest version and have a try. I use the test code it is OK now.

TempoTian, 2025-06-11:
@iothub01 Does the attached code previous mentioned which I tested work or not? I can not see the diff with your newly attached code now.