espressif/esp-webrtc-solution #55: Patching script out of date?

atomirex, 2025-05-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

common/patch_idf.pl - should it be run or not? Either way it needs comments.

### Actual behavior (suspected bug)

I wasn't sure if you are supposed to run common/patch_idf.pl or not, but I did, and found it did things, except said (accurately) that there was no file in components/srtp (because there is no component srtp).

### Error logs or terminal output

```shell
esp-webrtc-solution-v1.0.0/solutions/common/../../components/srtp/libsrtp/crypto/cipher/cipher.c file not exists
```

### Steps to reproduce the behavior

1. Run perl common/patch_idf.pl

### Project release version

v1.0.0

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

Sequoia

### Shell

ZSH

TempoTian, 2025-05-23:
@atomirex When use new IDF either release/v5.4 or master, no need to do patch any more, libsrtp has been commited as a componenthttps://components.espressif.com/components/espressif/esp_libsrtp/versions/1.0.0  and is downloaded directly from component registry. You can build directly, if have any issue, please let me know.

atomirex, 2025-05-23:
Thanks, that's good to know.

closed 2025-05-23 by atomirex