espressif/esp-webrtc-solution #62: SD Card Recording  store

iothub01, 2025-06-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Is there any example or code to store video/image in sd card using sd card slot provided in dev kit

### Use cases

for security

iothub01, 2025-06-12:
i am using mqtt based signalling for Webrtc. and using your sd card code same to same. I have included a mqtt message on which i call record_video(5000) and with button click or server command , i perform webrtc. 
If i Run store video for first time and then start webrtc , it work but if i again run store video and then run webrtc , video is not getting transmitted in webrtc.

iothub01, 2025-06-12:
is there a way to disable this -
esp_capture_add_muxer_to_path(capture_path, &muxer_cfg);
    esp_capture_enable_muxer(capture_path, true);

TempoTian, 2025-06-19:
I have received your mail, I can not reproduce the issue use your code. You can use doorbell_local demo to verify this issue.
put your doo_sdcard.c to main folder and replace main.c with follow file.
[doorbell_local_main.zip](https://github.com/user-attachments/files/20814916/doorbell_local_main.zip)
You can use CLI to test
1. cli enter:
rec --> It begin to record image
2.  after finished do webrtc
3. Stop webrtc redo 1
Repeat the steps, not find issue as you mentioned. make sure capture is full stopped before webrtc run.