espressif/esp-webrtc-solution #68: The website of doorbell demo can't be accessed.

lw603633, 2025-06-29:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

only once

### Expected behavior

The website of WebRTC can't be accessed.

### Actual behavior (suspected bug)

The website of doorbell demo can't be accessed.

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

Open the website

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

other (details in Additional context)

TempoTian, 2025-07-02:
@lw603633 Thanks for your info, I have ask IT to help check this issue. If on urgent can use https://webrtc.espressif.cn/doorbell as substitution. And use `sel` cli to select it, It is a little slow, but work ok currently.

hulife, 2025-08-12:
> Fixed, you can try it now https://webrtc.espressif.com/doorbell

The website cannot be opened now.

TempoTian, 2025-08-13:
@hulife Thanks for your info, I have asked IT to help to check, once fixed, I will notify to you.

TempoTian, 2025-08-13:
@hulife Fixed, now you can try it.

TempoTian, 2025-08-13:
The default code use a fixed room, you can use random room to test. 
First board enter a random room
```
leave
join xxxxnnnndd
```
Then on browser enter same room, after board join success.

closed 2025-09-16 by wujiangang