espressif/esp-webrtc-solution #70: Xirsys signaling support

deldering, 2025-07-03:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Has support been considered for the signaling implementation via [Xirsys](https://docs.xirsys.com/?pg=api-signal)? 

### Use cases

Running the remote examples via the Xirsys servers

TempoTian, 2025-07-04:
I’ve reviewed the [Xirsys signaling introduction](https://docs.xirsys.com/?pg=api-signal), and it doesn’t look very complex. You can refer to the existing peer default implementation as a model, since the protocol is nearly the same. It uses HTTPS POST to obtain the ICE server and WebSocket address, and then uses WebSocket to exchange SDP and events—the flow is almost identical.

I can draft a prototype using a cursor, but it will require setting up servers and debugging. If you can help verify it and it works as expected, I can integrate it into the official signaling implementation.

TempoTian, 2025-07-04:
Cursor write a very draft version maybe some error but I check the overall flow is all right, you can modify based on it I think.

[xirsys_signaling.zip](https://github.com/user-attachments/files/21050551/xirsys_signaling.zip)

TempoTian, 2025-07-28:
It will fail directly if no port is set.

[image]

deldering, 2025-07-28:
Currently the `get_ice_servers_response` only stores the stun url or the first turn url. What steps should be taken to allow for storing all the urls?

deldering, 2025-07-29:
Hi, I have been able to set up a connection after working through some issues so that is great. One thing that isn't working yet is the data channel. I've set `enable_data_channel` to true but the `ESP_WEBRTC_EVENT_DATA_CHANNEL_CONNECTED` doesn't seem to trigger. It doesn't matter whether I have `manual_ch_create` turned on or off.

TempoTian, 2025-07-30:
Yes, I found the issue. In my code, I force to use client role for controlled. Do you use esp32s3 and use latest code? I can compile a fix for you to check whether fixed or not

deldering, 2025-07-30:
Yes I'm running the esp32s3. I'd love to get a potential fix!

TempoTian, 2025-07-31:
[libpeer_default.zip](https://github.com/user-attachments/files/21521627/libpeer_default.zip) try this lib and put into esp_peer/libs/esp32s3 folder. If verified OK I can commit it.

deldering, 2025-07-31:
Thanks for the file! I've tested it and I can now successfully use the data channel. I have one more question with regards to re-using the webrtc object after a call is finished. Currently we call `stop_webrtc` followed by `start_webrtc` to be able to start a new call when a previous one has ended. Is there some other way how we can properly reset the webrtc handle to allow for a new call to be made without having the also restart the connection to the signaling server?

TempoTian, 2025-08-01:
Error code 401 is generated by STUN server. I see you end up the connection when receive `hang_up`. You shoud call `enable_connection` again to build new connection or call `esp_peer_new_connection ` API directly.

[image]

401 is caused by may be you are missing username and password
🔍 What Causes STUN Error 401? typically happens when:
   - The STUN server requires authentication (username + message integrity using a password or shared secret).
   - The client sends a Binding Request without MESSAGE-INTEGRITY or with an invalid one.
   - The client does not include a proper USERNAME attribute.
   - The password (or key) used to generate the integrity is incorrect or missing.

deldering, 2025-08-04:
Thanks, it was indeed something related to Xirsys themselves. Unrelated question: How do I select the other microphone for the korvo board? The channel_mask says `1 | 2` for the aec mode but I only ever seem to get data from microphone one and never from microphone two. How does this also relate to the `channel` having a value of `4`?