espressif/esp-webrtc-solution #75: sometimes repeating: UDP: Failed to sendto: 12 Not enough space

shushenghong, 2025-08-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

often

### Expected behavior

webrtc with only sound, some times repeating this log, what's this meaning? and when this happened, the sound is discontinuous

### Actual behavior (suspected bug)

一直打印 
E (320448) UDP: Failed to sendto: 12 Not enough space
E (320448) UDP: Failed to sendto: 12 Not enough space

### Error logs or terminal output

```shell
E (320448) UDP: Failed to sendto: 12 Not enough space
E (320448) UDP: Failed to sendto: 12 Not enough space
```

### Steps to reproduce the behavior

i don't know

### Project release version

latest release

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

13.6.7 (22G720)

### Shell

ZSH

kakopappa, 2026-09-13:
We hit this on an ESP32 (AI-Thinker ESP32-CAM) with esp_peer v1.5.5 and can confirm the diagnosis above: it tracks Wi-Fi signal quality closely. At -85 dBm the DTLS handshake never completed; at -79 dBm the same board streamed for 90 s. Free heap was healthy throughout (~49 kB free, 28 kB largest block), so it is clearly the driver's TX buffer pool rather than the allocator.

Two code-level points that may be worth fixing regardless of link quality, plus one note for Arduino users.

### 1. `select()` overwrites `errno`, so the retry only runs once

In `udp_socket_sendto()` (`components/esp_peer/src/transport/udp.c`, ~line 221) `errno` is read on both sides of the `select()` call:

```c
if ((errno == ENOBUFS || errno == 12) && retry_count < 2) {
    ...
    if ((ret = select(fd + 1, NULL, &write_set, NULL, &tv)) < 0) { ... }
    goto RETRY;          /* errno now reflects select(), not sendto() */
}
```

After `goto RETRY`, the condition tests whatever `select()` left behind, so the intended "retry up to twice" is effectively one retry, and the final `ESP_LOGE` can print the wrong error. Capturing it immediately after `sendto()` fixes both:

```c
int send_errno = errno;
if ((send_errno == ENOBUFS || send_errno == 12) && retry_count < 2) { ... }
```

### 2. The `-200` return is silent

```c
// Special return value so that can retry
if (retry_count > 0) {
    udp_dec_user(udp_socket);
    return -200;
}
```

A caller treating this as "buffer full, retry later" cannot tell a transient stall from a permanent one. In our case `esp_peer_send_data()` returned `ESP_PEER_ERR_WOULD_BLOCK` continuously, the data-channel send cache never drained, and the DTLS handshake never completed — with no error emitted anywhere in the stack. The device reported a healthy session while the viewer showed a frozen frame. We could only find it by patching this function to log:

```c
if (retry_count > 0) {
    static uint32_t deferred = 0;
    if ((++deferred & 63) == 1) {
        ESP_LOGE(TAG, "sendto deferred (%u so far): errno %d (%s), %d bytes",
                 (unsigned)deferred, send_errno, strerror(send_errno), len);
    }
    udp_dec_user(udp_socket);
    return -200;
}
```

Rate-limited because it can fire dozens of times per second on a weak link.

### 3. Notes for Arduino ESP32 users

- **menuconfig is not available.** The Arduino core ships a prebuilt SDK, so `CONFIG_ESP_WIFI_DYNAMIC_TX_BUFFER_NUM` and the LWIP UDP settings cannot be raised from a sketch. The workaround suggested above does not apply there.
- **`ESP_LOGW` is compiled out.** The Arduino SDK builds with `CONFIG_LOG_MAXIMUM_LEVEL=1`, so the two existing `ESP_LOGW` calls in `udp.c` never reach an Arduino user's console. Any new diagnostic here needs `ESP_LOGE` to be visible.
- **Data-channel cache size matters on ESP32.** With Wi-Fi and a TLS socket already up, we had a single ~28 kB contiguous internal block left. `send_cache_size = 16 kB` + `recv_cache_size = 8 kB` consumed it, and the Wi-Fi driver could then no longer allocate TX buffers at all — `sendto()` failed with `ENOMEM` even for 88-byte STUN packets. Reducing them to 6 kB / 3 kB raised our observed heap floor from 31.6 kB to 49.7 kB and allowed the handshake to complete on a link where it previously could not.

TempoTian, 2026-09-14:
@kakopappa Thanks for you deep analysis.

For mentioned issue 1,2, I will review and committed in next version if needed.

For 3, the protocol itself does not consume too much mem, most mem consumed by MbedTLS and related network stack.
- If run on ram only board, need fine-tune memory size, for small data to be transfered can change it to quite small value.
- If want to send large data, recommend to use board which carried with PSRAM, most of the fifo and task stacks can allocated on PSRAM, Wifi and UDP related fifos can also put on PSRAM, then it can have better performance.

I will study how to achieve handshake success ratio on bad network also, during test network is not always poor, it recovered soon after `esp_peer_send_data` return WOULD_BLOCK, so I not pay special attention to it yet.