espressif/esp-webrtc-solution #76: custom message on speaker

iothub01, 2025-08-14:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

How can we play custom message on speaker like message sent from server to device and its plays that custom message 

### Use cases

video doorbell

TempoTian, 2025-08-15:
Custom message means text? If play text need tts support. If message is audio can play it as webrtc audio stream I think.

TempoTian, 2025-09-01:
I have not seen official support for it yet (https://github.com/espressif/esp-video-components/blob/master/esp_cam_sensor/README.md). You can select camera from the supported lists I think.

closed 2026-02-27 by wujiangang