espressif/esp-webrtc-solution #88: Is it works with the latest release of GPT-realtime with the image input?

ramkumarkoppu, 2025-09-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Is it works with the latest release of GPT-realtime with the image input?

keryc, 2025-09-10:
@radimkarnis I implemented the new gpt-realtime model in my own integration by reusing key components from esp-webrtc-solution in case you'd like to take a look: https://github.com/keryc/esp32-webrtc-openai 

@TempoTian yes, it works as you mention, but the API used in your demo is the [legacy](https://platform.openai.com/docs/api-reference/realtime_beta) one. It is more stable with the new API and requires minimal changes.

TempoTian, 2025-09-10:
@keryc That’s awesome to hear! 🎉 Your code brings complete visual support with chat, which is super exciting. Really appreciate the effort you’ve put into this!