espressif/esp-webrtc-solution #90: Turn not working

Alan5142, 2025-09-15:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

When using a TURN server, it should allow the communication between both peers.

### Actual behavior (suspected bug)

I have tested the `esp_peer` component through IDF component manager. My peer is a custom desktop console app (built with webrtc.rs), and overall works fine when both of them specify `peer_cfg.ice_trans_policy = ESP_PEER_ICE_TRANS_POLICY_ALL` (`RTCIceTransportPolicy::All` in the rust side). However, when I specify `ESP_PEER_ICE_TRANS_POLICY_RELAY` or when any of them is in a different network, it no longer works.

I am using [Cloudflare TURN service](https://developers.cloudflare.com/realtime/turn/)

I attach part of the logs after I received a `ESP_PEER_STATE_PAIRING` signal. After that, I receive a `ESP_PEER_STATE_CONNECT_FAILED` event. I received tons of those logs before receiving the CONNECT FAILED event.

### Error logs or terminal output

```shell
I (86007) AGENT: 0 Send binding request (cand:0) local:192.168.100.232:53554 remote:172.26.144.1:60995 id:6fe91e6c2322a8846df77827
I (86010) AGENT: 0 Send binding request (cand:0) local:192.168.100.232:53554 remote:192.168.177.1:51326 id:1e63da772a81f9c13740f7d1
I (86021) AGENT: 0 Send binding request (cand:0) local:xxx.xxx.xxx.xxx:53554 remote:yyy.yyy.yyy.yyy:25248 id:621f7dc2473fade00ecfd396
I (86032) AGENT: 0 Send binding request (cand:0) local:xxx.xxx.xxx.xxx:53554 remote:yyy.yyy.yyy.yyy:25148 id:45480065562666204bf4e964
I (86043) AGENT: 0 Send binding request (cand:0) local:xxx.xxx.xxx.xxx:53554 remote:yyy.yyy.yyy.yyy:15130 id:57f7686d7936f3746ee7eb07
I (86055) AGENT: 0 Send binding request (cand:0) local:xxx.xxx.xxx.xxx:53554 remote:yyy.yyy.yyy.yyy:15645 id:55ee4c0b44c1aa5d4f7df387
I (86066) AGENT: 0 Send binding request (cand:0) local:xxx.xxx.xxx.xxx:53554 remote:yyy.yyy.yyy.yyy:25248 id:16dc99605368c1de0aa98cb7
I (86078) AGENT: 0 Send binding request (cand:0) local:xxx.xxx.xxx.xxx:53554 remote:yyy.yyy.yyy.yyy:25148 id:314d8d5d5964648b6f641684
I (86089) AGENT: 0 Send binding request (cand:0) local:xxx.xxx.xxx.xxx:53554 remote:yyy.yyy.yyy.yyy:15130 id:09f4ba361af7c2e06b1623bb
I (86101) AGENT: 0 Send binding request (cand:0) local:xxx.xxx.xxx.xxx:53554 remote:yyy.yyy.yyy.yyy:15645 id:127e9dbf19cb4497662fbe85
I (86112) AGENT: 0 Send binding request (cand:0) local:104.30.145.172:51354 remote:104.30.150.84:22689 id:0945b15a232c184841d5cd1c
I (86123) AGENT: 0 send indication bind request
```

### Steps to reproduce the behavior

1. Create a minimal setup of ESP_PEER with an external signal server.
2. Connect either the ESP32 or the peer to another network
3. Try to connect both of them

### Project release version

v1.2.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

### Additional context

I am using a ESP32 S3

Alan5142, 2025-09-15:
> I test use TURN can work if set `ESP_PEER_ICE_TRANS_POLICY_ALL`, but failed if set `ESP_PEER_ICE_TRANS_POLICY_RELAY `. Is it same phenomenon as your part? Seems some regression with `ESP_PEER_ICE_TRANS_POLICY_RELAY`.

If both are in the same network, yes, it fails when any of the peers use that option. If both are in different networks (one of them being mobile network), it fails as they try to use the TURN server.

TempoTian, 2025-09-16:
I have fixed the issue that I can reproduce local use doorbell_demo when set policy to relay only. The lib is as below, you can put it into managed__components/espressif__esp_peer/lib/esp32s3 to test doorbell_demo also. but from your log, I see no stun binding request is responsed. You can try to use wireshark to fetch network packets, I can help you to analysis it. 

[libpeer_default.zip](https://github.com/user-attachments/files/22354343/libpeer_default.zip)

TempoTian, 2025-09-16:
Can you attach the matched log and remove sensitive info?

Alan5142, 2025-09-16:
Yes, those are the full logs since the device connected to the signaling server (custom websocket server):

<details>

```I (3157) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3162) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (3209) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (4234) esp_netif_handlers: sta ip: 192.168.137.140, mask: 255.255.255.0, gw: 192.168.137.1
I (4235) test: got ip:192.168.137.140
I (4236) test: IPv4 Address: 192.168.137.140
I (4246) test: connected to ap
W (4249) websocket_client: `reconnect_timeout_ms` is not set, or it is less than or equal to zero, using default time out 10000 (milliseconds)
W (4262) websocket_client: `network_timeout_ms` is not set, or it is less than or equal to zero, using default time out 10000 (milliseconds)
I (4274) test: Other event id:6
I (4277) test: WEBSOCKET_EVENT_BEFORE_CONNECT
I (4283) websocket_client: Started
I (4852) test: WEBSOCKET_EVENT_CONNECTED
I (34983) wifi:<ba-add>idx:0 (ifx:0, b2:dc:ef:91:09:d8), tid:0, ssn:12, winSize:64
I (46022) test: Received WS data event with op_code: 1
I (46023) test: Received WS data event fin: 1
I (46023) test: Handling CONNECTED event, sending REQICE
I (46637) test: Received WS data event with op_code: 1
I (46637) test: Received WS data event fin: 1
I (46638) test: ICE Server: URL=stun:stun.cloudflare.com:3478, User=, Credential=
I (46647) test: ICE Server: URL=stun:stun.cloudflare.com:53, User=, Credential=
I (46659) test: ICE Server: URL=turn:turn.cloudflare.com:3478?transport=udp, User=<TURN SERVER USER>, Credential=<TURN SERVER PASSWORD>
I (46683) test: ICE Server: URL=turn:turn.cloudflare.com:3478?transport=tcp, User=<TURN SERVER USER>, Credential=<TURN SERVER PASSWORD>
I (46707) test: ICE Server: URL=turns:turn.cloudflare.com:5349?transport=tcp, User=<TURN SERVER USER>, Credential=<TURN SERVER PASSWORD>
I (46731) test: ICE Server: URL=turn:turn.cloudflare.com:53?transport=udp, User=<TURN SERVER USER>, Credential=<TURN SERVER PASSWORD>
I (46755) test: ICE Server: URL=turn:turn.cloudflare.com:80?transport=tcp, User=<TURN SERVER USER>, Credential=<TURN SERVER PASSWORD>
I (46780) test: ICE Server: URL=turns:turn.cloudflare.com:443?transport=tcp, User=<TURN SERVER USER>, Credential=<TURN SERVER PASSWORD>
I (46804) DTLS: Init SRTP OK
W (47889) PEER_DEF: Create cache for RTP packets fail, resent logic is disabled
I (47890) test: Peer state changed: 2
I (47890) AGENT: Start agent as Controlling
I (47893) PEER_DEF: Start DTLS role as 1
I (47912) AGENT: Send STUN binding request
I (47920) AGENT: 0 Get candidate success user: psw:
I (47925) AGENT: Send STUN binding request
I (47992) AGENT: Send STUN binding request
I (47993) AGENT: Send allocate now
Got error code 401
I (48005) AGENT: Send allocate now
I (48007) AGENT: Send allocate now
I (48011) AGENT: 0 Get candidate success user:<TURN SERVER USER> psw:<TURN SERVER PASSWORD>
W (48017) AGENT: Skip TCP stun server turn:turn.cloudflare.com:3478?transport=tcp
W (48025) AGENT: Skip TCP stun server turns:turn.cloudflare.com:5349?transport=tcp
I (48032) AGENT: Skip TURN server turn:turn.cloudflare.com:53?transport=udp for only support one
W (48040) AGENT: Skip TCP stun server turn:turn.cloudflare.com:80?transport=tcp
W (48047) AGENT: Skip TCP stun server turns:turn.cloudflare.com:443?transport=tcp
I (48055) test: Peer message received: type=1, size=521
I (48060) test: Sending SDP: v=0
o=- 7801614086411001801 2 IN IP4 0.0.0.0
s=-
t=0 0
a=msid-semantic: esp-webrtc
a=group:BUNDLE 0
m=audio 9 UDP/TLS/RTP/SAVPF 8
c=IN IP4 0.0.0.0
a=mid:0
a=sendrecv
a=rtcp-mux
a=rtpmap:8 PCMA/8000
a=ssrc:4 cname:EspSSpbA0
a=inactive
a=fingerprint:sha-256 94:25:9F:B8:4D:4F:6E:DE:D8:BD:93:6A:98:45:63:70:6A:F7:E1:6D:B2:E7:9C:E6:0E:C5:F9:7C:11:77:DB:7E
a=setup:passive
a=ice-ufrag:WwUl
a=ice-pwd:Pd9ZwfQgpeN3WiStdYMx5Ru0
a=candidate:0 1 UDP 10151935 104.30.145.172 39655 typ relay raddr 0.0.0.0 rport 0
[... 19 lines cut ...]
E (58164) AGENT: 0 Remote candidate over limited 10
E (58168) AGENT: 0 Remote candidate over limited 10
E (58173) AGENT: 0 Remote candidate over limited 10
E (58178) AGENT: 0 Remote candidate over limited 10
E (58182) AGENT: 0 Remote candidate over limited 10
0 Sorted pair 0 type: 4 local:104.30.145.172:39655 Remote:104.30.145.173:9824
I (58194) test: Peer state changed: 3
I (58202) AGENT: 0 Send binding request (cand:0) local:104.30.145.172:39655 remote:104.30.145.173:9824 id:4c598d93643bd7252b449456
I (58209) AGENT: 0 send indication bind request

I (58667) AGENT: 0 Send binding request (cand:0) local:104.30.145.172:39655 remote:104.30.145.173:9824 id:6db505597d157d8d71dfed43
I (58668) AGENT: 0 send indication bind request

I (59127) AGENT: 0 Send binding request (cand:0) local:104.30.145.172:39655 remote:104.30.145.173:9824 id:666d97166fd595896931589d
I (59128) AGENT: 0 send indication bind request

I (59587) AGENT: 0 Send binding request (cand:0) local:104.30.145.172:39655 remote:104.30.145.173:9824 id:1dee132f3b3a772600f32298
I (59588) AGENT: 0 send indication bind request

I (60047) AGENT: 0 Send binding request (cand:0) local:104.30.145.172:39655 remote:104.30.145.173:9824 id:1ae53a307c418dd42170b934
I (60048) AGENT: 0 send indication bind request

I (60356) test: Peer state changed: 7```

</details>

TempoTian, 2025-09-16:
You can use wireshark to check the following stun binding details, I see esp32s3 send stun binding but peer not responsed, also not receive peer stun binding request also. Maybe peer not ready for connection?

Alan5142, 2025-09-16:
The first picture is from the ESP32 S3 client.

[image]

The second picture is from the PC client

[image]

If needed, this is my esp_peer configuration:
```c
esp_peer_default_cfg_t extra_peer_cfg = {};
extra_peer_cfg.data_ch_cfg.recv_cache_size = 1536;
extra_peer_cfg.data_ch_cfg.send_cache_size = 1536;
extra_peer_cfg.agent_recv_timeout = 10000;

esp_peer_cfg_t peer_cfg = {};
peer_cfg.server_lists = ice_servers.data();
peer_cfg.server_num = static_cast<uint8_t>(ice_servers.size());
peer_cfg.role = ESP_PEER_ROLE_CONTROLLING;
peer_cfg.ice_trans_policy = ESP_PEER_ICE_TRANS_POLICY_RELAY;
peer_cfg.audio_info.codec = ESP_PEER_AUDIO_CODEC_NONE;
peer_cfg.video_info.codec = ESP_PEER_VIDEO_CODEC_NONE;
peer_cfg.audio_dir = ESP_PEER_MEDIA_DIR_NONE;
peer_cfg.video_dir = ESP_PEER_MEDIA_DIR_NONE;
peer_cfg.enable_data_channel = true;
peer_cfg.ctx = this;
peer_cfg.on_state = state_callback;
peer_cfg.on_msg = msg_callback;
peer_cfg.on_data = data_callback;
peer_cfg.extra_cfg = &extra_peer_cfg;
peer_cfg.extra_size = sizeof(extra_peer_cfg);
peer_cfg.no_auto_reconnect = false;
peer_cfg.manual_ch_create = false;```

TempoTian, 2025-09-16:
When set with all policy, I see it use local candidate. And peer binding is sent, also responsed for binding request. I see one issue is that why server return 41076 for both client?

TempoTian, 2025-09-16:
I think maybe webrtc.rs has bug, I see that its turn relay address is 41076, and esp32s3 relay address is 42542. When send indication and create permission it should use peer port instead of its own port 41076.

TempoTian, 2025-09-16:
PS: For simple test you can also use 2 esp32s3 board to test, one configured as controlling one as controlled, it can easy verify.

Alan5142, 2025-09-16:
Hello again!

Since I only have a pair of ESP32 C5 laying around, I have decided to test the doorbell local demo but specifying the TURN server.
Below are my modifications to the files:
webrtc_test.html:
```javascript
const configuration = {
                iceServers: [
                    {
                        urls: [
                            'stun:stun.l.google.com:19302',
                            'stun:stun1.l.google.com:19302'
                        ]
                    },
                    {
                        urls: 'turn:turn.cloudflare.com:3478?transport=udp',
                        username: 'XXXX',
                        credential: 'YYYY'
                    }
                ],
                iceCandidatePoolSize: 10,
                bundlePolicy: 'max-bundle',
                rtcpMuxPolicy: 'require',
                sdpSemantics: 'unified-plan',
                iceTransportPolicy: 'relay' // Force use of TURN server
            };
```

And webrtc_http_server.c
```c
esp_peer_signaling_ice_info_t ice_info = {
        .is_initiator = true,
        .server_info.stun_url = "stun:stun.l.google.com:19302",
    };
    // Notify for rule
    sig_cfg.on_ice_info(&ice_info, cfg->ctx);
    esp_peer_signaling_ice_info_t turn_info = {
        .is_initiator = true,
        .server_info.stun_url = "turn:turn.cloudflare.com:3478?transport=udp",
        .server_info.user = "XXX",
        .server_info.psw = "YYY",
    };
    sig_cfg.on_ice_info(&turn_info, cfg->ctx);
```

It produces the same result as my test . I have a question, is there a fixed size for the user and psw fields?, cloudflare returns a user and a credential with a length of 64 characters, so I don't know if this might generate an issue

```
I (2793) Webrtc_Test: Use browser to enter https://192.168.137.161/webrtc/test for test
I (2799) AV_RENDER: Adec got msg:Close
I (2800) AV_RENDER: Thread Adec exited
I (2800) AV_RENDER: Close done
I (10069) esp_https_server: performing session handshake
I (10564) wifi:<ba-add>idx:0 (ifx:0, b2:dc:ef:91:09:d8), tid:0, ssn:12, winSize:64
I (10878) esp_https_server: performing session handshake
Get post request
Request length 42
Get post {"type":"customized","data":"ACCEPT_CALL"}
I (25277) webrtc: Received customized data
I (25277) DOOR_BELL: Receive command ACCEPT_CALL
I (25290) DTLS: Init SRTP OK
I (26416) webrtc: peer_connection_task started
I (26416) webrtc: PeerConnectionState: 2
I (26418) AGENT: Start agent as Controlling
I (26419) PEER_DEF: Start DTLS role as 1
I (26450) AGENT: Send STUN binding request
I (26451) AGENT: Send allocate now
Got error code 401
I (26460) AGENT: Send allocate now
I (26462) AGENT: Send allocate now
[... 102 lines cut ...]
I (29949) AGENT: 0 Send binding request (cand:0) local:104.30.145.172:50034 remote:104.30.145.172:59722 id:666d97166fd595896931589d
I (29950) AGENT: 0 send indication bind request

Get post request
Request length 66
Get post {"type":"candidate","candidate":"","sdpMid":"0","sdpMLineIndex":0}
I (30447) AGENT: 0 Send binding request (cand:0) local:104.30.145.172:50034 remote:104.30.145.172:59722 id:1dee132f3b3a772600f32298
I (30449) AGENT: 0 send indication bind request

I (30944) AGENT: 0 Send binding request (cand:0) local:104.30.145.172:50034 remote:104.30.145.172:59722 id:1ae53a307c418dd42170b934
I (30946) AGENT: 0 send indication bind request

I (31119) webrtc: PeerConnectionState: 7
Get post request
Request length 40
Get post {"type":"customized","data":"DENY_CALL"}
I (31949) webrtc: Received customized data
I (31950) DOOR_BELL: Receive command DENY_CALL
I (31961) AV_RENDER: Adec got msg:Close
I (31961) AV_RENDER: Thread Adec exited
I (31962) AV_RENDER: Close done
I (31973) webrtc: PeerConnectionState: 1
I (31973) AV_RENDER: Close done
I (31974) AGENT: agent_tDisconnected done
```

Alan5142, 2025-09-16:
Hi!

Finally I tested with Edge and it worked without problems, it finally connected. I feel confident that I can close this now.

Also, I have a few more questions but I'm not sure if this is the place to ask them: is IPv6 support planned?, and, is ESP32 C5 support planned?

Thank you!

TempoTian, 2025-09-17:
Yes, I can add C5 support. I have analysis this issue it is caused by turn server reply with padding bytes which handle incorrectly. I will provide a lib for you to verify. 
Send Part：

[image]
Receive part:

[image]

Alan5142, 2025-09-17:
Sure, I'll do the test tomorrow, thank you so much for the support!

Alan5142, 2025-09-26:
Sorry for the late response, I'm currently testing the new lib

Alan5142, 2025-09-26:
I get some issues with the TURN servers on Microsoft Edge (not tested with Chrome). It seems that it connects successfully though.

[image]

[image]

[image]

And then I get lots of errors:

```
E (33420) DTLS: mbedtls_ssl_read error: -27648
E (33421) DTLS: mbedtls_ssl_read error: -27648
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
E (33562) DTLS: mbedtls_ssl_read error: -27648
E (33563) DTLS: mbedtls_ssl_read error: -27648
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
I (33678) webrtc: Send A:9080 [198:15840] Recv A:284217806 [0:0]
I (33678) PEER_DEF: Send 235:49200 d_out:16384 receive 199:24760 din:0
I (33679) SCTP: Send: 1298 Ack: 1233 Cached: 75524 Receive:0
D 7
I (33690) SCTP: Recv: 3918331996 ACK:3918331996 Cached: 0 SackCount:21

D 7
D 7
E (33728) DTLS: mbedtls_ssl_read error: -27648
E (33729) DTLS: mbedtls_ssl_read error: -27648
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
Send string Send to channel1 count 9
D 7
E (33868) DTLS: mbedtls_ssl_read error: -27648
E (33869) DTLS: mbedtls_ssl_read error: -27648
D 7
E (33881) DTLS: mbedtls_ssl_read error: -27648
E (33882) DTLS: mbedtls_ssl_read error: -27648
D 7
D 7
D 7
D 7
D 7
D 7
D 7
[... 905 lines cut ...]
D 7
D 7
D 7
D 7
D 7
D 7
E (46639) DTLS: mbedtls_ssl_read error: -27648
E (46640) DTLS: mbedtls_ssl_read error: -27648
D 7
D 7
D 7
D 7
D 7
D 7
D 7
D 7
E (46797) DTLS: mbedtls_ssl_read error: -27648
E (46799) DTLS: mbedtls_ssl_read error: -27648
D 7
D 7
D 7
D 7
Get post request
Request length 40
```