espressif/esp-webrtc-solution #95: open ai demo panic when esp_srmodel_init

shushenghong, 2025-09-20:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

run normally

### Actual behavior (suspected bug)

panic

### Error logs or terminal output

```shell
I (5263) ES7210: Enable ES7210_INPUT_MIC2
I (5293) ES7210: Enable ES7210_INPUT_MIC3
I (5293) ES7210: Enable ES7210_INPUT_MIC4
I (5303) ES7210: Enable TDM mode
I (5313) ES7210: Unmuted
I (5313) Adev_Codec: Open codec device OK
I (5313) MODEL_LOADER: The storage free size is 22016 KB
I (5313) MODEL_LOADER: The partition size is 5168 KB

assert failed: esp_cache_freeze_caches_disable_interrupts esp_cache_utils.c:96 (s_task_stack_is_sane_when_cache_frozen())

Backtrace: 0x4037a02d:0x3c328a40 0x40385631:0x3c328a60 0x4038e275:0x3c328a80 0x403839c4:0x3c328ba0 0x4037cb5f:0x3c328bc0 0x4037cb7f:0x3c328be0 0x4211689e:0x3c328c10 0x42116b7e:0x3c328c60 0x42115f7d:0x3c328ca0 0x420cc28a:0x3c328cc0 0x420cc389:0x3c328cf0 0x4201b7d1:0x3c328d10 0x4201b8c5:0x3c328d30 0x4201aef2:0x3c328d60 0x420c78ab:0x3c328d90 0x420ca9e5:0x3c328dc0 0x420cad80:0x3c328e10 0x40386195:0x3c328e40
--- 0x4037a02d: panic_abort at /Users/ss/Documents/tools/esp/v5.5/esp-idf/components/esp_system/panic.c:469
--- 0x40385631: esp_system_abort at /Users/ss/Documents/tools/esp/v5.5/esp-idf/components/esp_system/port/esp_system_chip.c:87
--- 0x4038e275: __assert_func at /Users/ss/Documents/tools/esp/v5.5/esp-idf/components/newlib/src/assert.c:80
--- 0x403839c4: esp_cache_freeze_caches_disable_interrupts at /Users/ss/Documents/tools/esp/v5.5/esp-idf/components/esp_mm/esp_cache_utils.c:96
--- 0x4037cb5f: s_stop_cache at /Users/ss/Documents/tools/esp/v5.5/esp-idf/components/esp_mm/esp_mmu_map.c:381
--- 0x4037cb7f: s_do_mapping at /Users/ss/Documents/tools/esp/v5.5/esp-idf/components/esp_mm/esp_mmu_map.c:441
--- 0x4211689e: esp_mmu_map at /Users/ss/Documents/tools/esp/v5.5/esp-idf/components/esp_mm/esp_mmu_map.c:594
--- 0x42116b7e: spi_flash_mmap at /Users/ss/Documents/tools/esp/v5.5/esp-idf/components/spi_flash/flash_mmap.c:88
--- 0x42115f7d: esp_partition_mmap at /Users/ss/Documents/tools/esp/v5.5/esp-idf/components/esp_partition/partition_target.c:172
--- 0x420cc28a: srmodel_mmap_init at /Users/ss/Documents/esp-webrtc/solutions/openai_demo/managed_components/espressif__esp-sr/src/model_path.c:330
--- 0x420cc389: esp_srmodel_init at /Users/ss/Documents/esp-webrtc/solutions/openai_demo/managed_components/espressif__esp-sr/src/model_path.c:526
--- 0x4201b7d1: open_afe at /Users/ss/Documents/esp-webrtc/solutions/openai_demo/managed_components/espressif__esp_capture/impl/capture_audio_src/capture_audio_aec_src.c:54
--- 0x4201b8c5: audio_aec_src_start at /Users/ss/Documents/esp-webrtc/solutions/openai_demo/managed_components/espressif__esp_capture/impl/capture_audio_src/capture_audio_aec_src.c:163
--- 0x4201aef2: audio_src_el_open at /Users/ss/Documents/esp-webrtc/solutions/openai_demo/managed_components/espressif__esp_capture/impl/capture_gmf_path/src/elements/gmf_audio_src.c:166
--- 0x420c78ab: esp_gmf_element_process_open at /Users/ss/Documents/esp-webrtc/solutions/openai_demo/managed_components/espressif__gmf_core/src/esp_gmf_element.c:266
--- 0x420ca9e5: process_func at /Users/ss/Documents/esp-webrtc/solutions/openai_demo/managed_components/espressif__gmf_core/src/esp_gmf_task.c:184
--- 0x420cad80: esp_gmf_thread_fun at /Users/ss/Documents/esp-webrtc/solutions/openai_demo/managed_components/espressif__gmf_core/src/esp_gmf_task.c:298
--- 0x40386195: vPortTaskWrapper at /Users/ss/Documents/tools/esp/v5.5/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

### Steps to reproduce the behavior

partiion.csv
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 3M,
model,  data, spiffs,         , 5168K
```

### Project release version

main branch

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

MacOS

### Operating system version

13.6.7 (22G720)

### Shell

ZSH

shushenghong, 2025-09-22:
thank you , i will try later, i see `strcmp(name, "aud_src") == 0`, is this should be `AUD_SRC`?

TempoTian, 2025-09-24:
1. 多声道的支持可以看https://github.com/espressif/esp-webrtc-solution/issues/93#issuecomment-3301349324
2.  你说的是BSS吗，BSS是在[SE](https://github.com/espressif/esp-sr/blob/master/include/esp32s3/esp_afe_config.h#L100)里面打开的
3.  `esp_codec_dev_set_in_gain` 是作用于所有channel的，`esp_codec_dev_set_in_channel_gain` 可以设置单个channel，适用于单独改变mic 采集音量不改变ref音量的场景。可以优先用esp_codec_dev_set_in_channel_gain提高mic的音量看看

- Read fail的问题之前倒是没遇到过，是开了特定效果才有的吗？我可以对齐配置本地看下

closed 2026-02-27 by wujiangang