espressif/esp-webrtc-solution #97: 通过webrtc连接https://webrtc.espressif.com/，报错ESP_PEER_ERR_NO_MEM

ClearrSky, 2025-09-26:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

我想通过在本地的ESP32开启一个webrtc连接，连接到https://webrtc.espressif.com/中的房间。
我希望的效果是：我的ESP32设备可以正确推送音视频流到房间中，而我可以通过电脑访问该网址的对应房间看到对应的音视频画面。但是在我开启连接的时候遇到了一些问题
```c
int webrtcHelper::start_webrtc()
{
    uint32_t rand_num = esp_random();  
    // 生成 9 位数 (100000000 ~ 999999999)
    uint32_t nine_digit = 100000000 + (rand_num % 900000000);
    char signal_url_buffer[64];  // 预留足够大缓冲区
    sprintf(signal_url_buffer, "https://webrtc.espressif.com/join/%lu", nine_digit);
    ESP_LOGI(TAG, "room id is %lu", nine_digit);

    if (webrtc) {
        esp_webrtc_close(webrtc);
        webrtc = NULL;
    }

    esp_peer_default_cfg_t peer_default_cfg = {
        .agent_recv_timeout = 100,   // Enlarge this value if network is poor
        .data_ch_cfg = {
            .send_cache_size = 1536, // Should big than one MTU size
            .recv_cache_size = 1536, // Should big than one MTU size
        },
        .rtp_cfg = {
            .audio_recv_jitter = {
                .cache_size = 1024,
            },
            .send_pool_size = 1024,
            .send_queue_num = 10,
        },
        // .data_ch_cfg = {
        //     .send_cache_size = WEBRTC_DATA_CH_SEND_CACHE_SIZE,
        //     .recv_cache_size = WEBRTC_DATA_CH_RECV_CACHE_SIZE,
        // }
    };

    esp_webrtc_cfg_t cfg = {
        .signaling_impl = esp_signaling_get_apprtc_impl(),
        .signaling_cfg = {
            .signal_url = signal_url_buffer,
            .ctx = &webrtc,
        },
        .peer_impl = esp_peer_get_default_impl(),
        .peer_cfg = {
            .audio_info = {
                .codec = ESP_PEER_AUDIO_CODEC_G711A,
            },
            .video_info = {
                .codec = ESP_PEER_VIDEO_CODEC_MJPEG,
                .width = 320,
                .height = 192,
                .fps = 1,
            },
            .audio_dir = ESP_PEER_MEDIA_DIR_SEND_ONLY,
            .video_dir = ESP_PEER_MEDIA_DIR_SEND_ONLY,
            .enable_data_channel = DATA_CHANNEL_ENABLED,
            .video_over_data_channel = false, //  video transfer over data channel
            .no_auto_reconnect = false,
            
            .extra_cfg = &peer_default_cfg,
            .extra_size = sizeof(peer_default_cfg),
            .on_custom_data = nullptr,
        },
    };
    esp_webrtc_media_provider_t media_provider = {
        .capture = capture_handle_,
        .player = NULL,
    };

    int ret = esp_webrtc_open(&cfg, &webrtc);
    if (ret != 0) {
        ESP_LOGE(TAG, "Fail to open webrtc");
        return ret;
    }

    esp_webrtc_set_media_provider(webrtc, &media_provider);
    // Set event handler
    esp_webrtc_set_event_handler(webrtc, webrtc_event_handler, webrtc);

    // Default disable auto connect of peer connection
    esp_webrtc_enable_peer_connection(webrtc, false);

    ret = esp_webrtc_start(webrtc);
    if (ret != 0) {
        ESP_LOGE(TAG, "Fail to start webrtc");
    }

    ESP_LOGI(TAG,"Open webrtc success!");
    return ret;
```
这是我开启webrtc的代码，为了方便debug，我在等待大约5秒之后设置了esp_webrtc_enable_peer_connection(webrtc, true);

### Actual behavior (suspected bug)

实际上在进行连接时，也就是调用esp_webrtc_enable_peer_connection(webrtc, true);之后，发生了报错，返回值为-2，也就是ESP_PEER_ERR_NO_MEM，似乎是内存不够，我在调用该函数之前打印了我的设备内存，看起来还比较充足。请问是我的实现有问题？比如没有正确处理该信令服务器的信息，还是真的内存不足，如果内存不足的话，以我的设置大概需要多少内存？

### Error logs or terminal output

```shell
>> Free Internal RAM: 138859 bytes
>> Largest Free Internal Block: 63488 bytes
>> Free PSRAM: 8256708 bytes
I (8533) MODEL_LOADER: Successfully load srmodels
I (8543) Assets: Applying new assets to display
I (8543) webrtc_helper: room id is 162633973
I (9423) esp-x509-crt-bundle: Certificate validated
I (10613) APPRTC_SIG: result SUCCESS
Initials set to 1
I (10613) HTTPS_CLIENT: HTTP Status = 200, content_length = ld
I (11463) esp-x509-crt-bundle: Certificate validated
Got url:stun:webrtc.espressif.com:3478 user_name: 1758867390:ninefingers psw:KKcqfTm0BwfwcslTkxMYeEyGYMM=
I (12453) HTTPS_CLIENT: HTTP Status = 200, content_length = ld
I (12463) webrtc: Pending connection until user enable
I (12463) APPRTC_SIG: Registering signaling channel.
I (12463) APPRTC_SIG: Connecting to wss://webrtc.espressif.com:8089/ws...
I (12473) websocket_client: Started
I (12473) webrtc_helper: Open webrtc success!
I (12483) Application: STATE: activating
I (12483) Ota: Current version: 2.0.0
I (13103) esp-x509-crt-bundle: Certificate validated
I (14313) APPRTC_SIG: WEBSOCKET_EVENT_CONNECTED
I (14313) APPRTC_SIG: send to remote : {"cmd":"register","roomid":"162633973","clientid":"38724972"}
I (82533) webrtc: ICE info loaded
>> Free Internal RAM: 129619 bytes
>> Largest Free Internal Block: 63488 bytes
>> Free PSRAM: 8188316 bytes
I (82533) esp_peer: ops->open return -2
E (82543) webrtc: Fail to open peer ret -2
```

### Steps to reproduce the behavior

无

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 11

### Shell

PowerShell

TempoTian, 2025-09-26:
大概率是忘了注册os adapter了: https://github.com/espressif/esp-webrtc-solution/blob/main/solutions/doorbell_demo/main/main.c#L278
```
#include "media_lib_adapter.h"
media_lib_add_default_adapter();
```

TempoTian, 2025-10-22:
@ClearrSky 上面的log看应该是内存非法访问了，请提供用的dependency.lock 文件我来用相应的版本复制下看。
下面的问题试着更新下esp32_camera的版本看，我看 https://github.com/espressif/esp32-camera/pull/781 这个issue是相关的，不过我看capture拉的是最新的应该。

closed 2026-02-27 by wujiangang