espressif/esp-webrtc-solution #98: esp-idf v6 does have iscv32-esp-elf-objdump-xesppie  tool

robogo-pj, 2025-09-26:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

complete the build.

### Actual behavior (suspected bug)

build can;t find riscv32-esp-elf-objdump-xesppie program 
.
Have riscv32-esp-elf-objdump-xespv2p1
riscv32-esp-elf-objdump-xespv2p2

### Error logs or terminal output

```shell
I get this error: home/pj/.espressif/tools/riscv32-esp-elf/esp-15.2.0_20250920/riscv32-esp-elf/bin/riscv32-esp-elf-objdump-xesppie ["-h", "/home/pj/esp-webrtc-solution/solutions/doorbell_local/managed_components/espressif__esp_new_jpeg/lib/esp32p4/libesp_new_jpeg.a"] failed with error(No such file or directory (os error 2))
```

### Steps to reproduce the behavior

1 install master esp-idf
2 configue ssid, password. I also set the device name to -:

#if CONFIG_IDF_TARGET_ESP32P4
#define TEST_BOARD_NAME "ESP32_P4_EYE"
3 idf.py set-target esp32p4
4 idf.py build

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu 22.04

### Shell

Bash

### Additional context

on set-target get thes warning on set-target -- ESP-TEE is currently supported only on the esp32c6;esp32h2;esp32c5 SoCs
NOTICE: Skipping optional dependency: espressif/esp32-camera

also getting loads of warnings/notices in build output(leads me to believe esp_webrtc:solution needs earlier version of esp-idf >>>
idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/pj/esp-webrtc-solution/solutions/doorbell_local/build
Executing "ninja all"...
[7/1697] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
[... 197 lines cut ...]
Generated /home/pj/esp-webrtc-solution/solutions/doorbell_local/build/bootloader/bootloader.bin
[133/133] cd /home/pj/esp-webrtc-solution/solutions/doorbell_local/bu...rtc-solution/solutions/doorbell_local/build/bootloader/bootloader.bin
Bootloader binary size 0x5d30 bytes. 0x2d0 bytes (3%) free.
[1693/1697] Generating esp-idf/esp_system/ld/sections.ld
FAILED: [code=1] esp-idf/esp_system/ld/sections.ld /home/pj/esp-webrtc-solution/solutions/doorbell_local/build/esp-idf/esp_system/ld/sections.ld 
cd /home/pj/esp-webrtc-solution/solutions/doorbell_local/build && /home/pj/.espressif/python_env/idf6.0_py3.12_env/bin/python /home/pj/esp-idf/tools/ldgen/ldgen.py --config /home/pj/esp-webrtc-solution/solutions/doorbell_local/sdkconfig --fragments-list "/home/pj/esp-idf/components/riscv/linker.lf;/home/pj/esp-idf/components/esp_driver_gpio/linker.lf;/home/pj/esp-idf/components/esp_pm/linker.lf;/home/pj/esp-idf/components/esp_mm/linker.lf;/home/pj/esp-idf/components/spi_flash/linker.lf;/home/pj/esp-idf/components/esp_usb_cdc_rom_console/linker.lf;/home/pj/esp-idf/components/esp_system/linker.lf;/home/pj/esp-idf/components/esp_system/app.lf;/home/pj/esp-idf/components/esp_common/common.lf;/home/pj/esp-idf/components/esp_common/soc.lf;/home/pj/esp-idf/components/esp_rom/linker.lf;/home/pj/esp-idf/components/hal/linker.lf;/home/pj/esp-idf/components/log/linker.lf;/home/pj/esp-idf/components/heap/linker.lf;/home/pj/esp-idf/components/soc/linker.lf;/home/pj/esp-idf/components/esp_hw_support/linker.lf;/home/pj/esp-idf/components/esp_hw_support/dma/linker.lf;/home/pj/esp-idf/components/esp_hw_support/ldo/linker.lf;/home/pj/esp-idf/components/esp_hw_support/mspi_timing_tuning/linker.lf;/home/pj/esp-idf/components/freertos/linker_common.lf;/home/pj/esp-idf/components/freertos/linker.lf;/home/pj/esp-idf/components/esp_libc/src/esp_libc.lf;/home/pj/esp-idf/components/esp_libc/src/system_libs.lf;/home/pj/esp-idf/components/esp_libc/src/libc.lf;/home/pj/esp-idf/components/esp_libc/src/libm.lf;/home/pj/esp-idf/components/esp_driver_gptimer/linker.lf;/home/pj/esp-idf/components/esp_ringbuf/linker.lf;/home/pj/esp-idf/components/esp_driver_uart/linker.lf;/home/pj/esp-idf/components/app_trace/linker.lf;/home/pj/esp-idf/components/esp_event/linker.lf;/home/pj/esp-idf/components/esp_driver_usb_serial_jtag/linker.lf;/home/pj/esp-idf/components/vfs/linker.lf;/home/pj/esp-idf/components/lwip/linker.lf;/home/pj/esp-idf/components/esp_netif/linker.lf;/home/pj/esp-idf/components/esp_driver_spi/linker.lf;/home/pj/esp-idf/components/esp_gdbstub/linker.lf;/home/pj/esp-idf/components/esp_driver_pcnt/linker.lf;/home/pj/esp-idf/components/esp_driver_mcpwm/linker.lf;/home/pj/esp-idf/components/esp_driver_ana_cmpr/linker.lf;/home/pj/esp-idf/components/esp_driver_i2s/linker.lf;/home/pj/esp-idf/components/esp_driver_dac/linker.lf;/home/pj/esp-idf/components/esp_driver_bitscrambler/linker.lf;/home/pj/esp-idf/components/esp_driver_rmt/linker.lf;/home/pj/esp-idf/components/esp_driver_sdm/linker.lf;/home/pj/esp-idf/components/esp_driver_i2c/linker.lf;/home/pj/esp-idf/components/esp_driver_ledc/linker.lf;/home/pj/esp-idf/components/esp_driver_parlio/linker.lf;/home/pj/esp-idf/components/esp_driver_twai/linker.lf;/home/pj/esp-idf/components/driver/i2c/linker.lf;/home/pj/esp-idf/components/driver/twai/linker.lf;/home/pj/esp-idf/components/esp_adc/linker.lf;/home/pj/esp-idf/components/esp_driver_isp/linker.lf;/home/pj/esp-idf/components/esp_driver_i3c/linker.lf;/home/pj/esp-idf/components/esp_psram/linker.lf;/home/pj/esp-idf/components/esp_eth/linker.lf;/home/pj/esp-idf/components/esp_lcd/linker.lf;/home/pj/esp-idf/components/ieee802154/linker.lf;/home/pj/esp-idf/components/openthread/linker.lf;/home/pj/esp-webrtc-solution/solutions/doorbell_local/managed_components/espressif__esp_cam_sensor/linker.lf;/home/pj/esp-webrtc-solution/solutions/doorbell_local/managed_components/espressif__esp_ipa/linker.lf" --input /home/pj/esp-webrtc-solution/solutions/doorbell_local/build/esp-idf/esp_system/ld/sections.ld.in --output /home/pj/esp-webrtc-solution/solutions/doorbell_local/build/esp-idf/esp_system/ld/sections.ld --kconfig /home/pj/esp-idf/Kconfig --env-file /home/pj/esp-webrtc-solution/solutions/doorbell_local/build/config.env --libraries-file /home/pj/esp-webrtc-solution/solutions/doorbell_local/build/ldgen_libraries --objdump /home/pj/.espressif/tools/riscv32-esp-elf/esp-15.2.0_20250920/riscv32-esp-elf/bin/riscv32-esp-elf-objdump
/home/pj/.espressif/tools/riscv32-esp-elf/esp-15.2.0_20250920/riscv32-esp-elf/bin/riscv32-esp-elf-objdump-xesppie ["-h", "/home/pj/esp-webrtc-solution/solutions/doorbell_local/managed_components/espressif__esp_new_jpeg/lib/esp32p4/libesp_new_jpeg.a"] failed with error(No such file or directory (os error 2))

thread 'main' panicked at main.rs:313:5:
internal error: entered unreachable code
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Traceback (most recent call last):
  File "/home/pj/esp-idf/tools/ldgen/ldgen.py", line 176, in <module>
    main()
  File "/home/pj/esp-idf/tools/ldgen/ldgen.py", line 131, in main
    dump = StringIO(subprocess.check_output([objdump, '-h', library], env=new_env).decode())
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pj/anaconda3/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pj/anaconda3/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/pj/.espressif/tools/riscv32-esp-elf/esp-15.2.0_20250920/riscv32-esp-elf/bin/riscv32-esp-elf-objdump', '-h', '/home/pj/esp-webrtc-solution/solutions/doorbell_local/managed_components/espressif__esp_new_jpeg/lib/esp32p4/libesp_new_jpeg.a']' returned non-zero exit status 101.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/pj/esp-webrtc-solution/solutions/doorbell_local/build/log/idf_py_stderr_output_74456 and /home/pj/esp-webrtc-solution/solutions/doorbell_local/build/log/idf_py_stdout_output_74456

robogo-pj, 2025-09-27:
t reinstalled older esp-idf version 5.5.1. Got the build to complete and flashed. esp32-p4-eye with esp-webrtc-solution.

Still probably wrong version as getting lots warnings of declared variables not used.

Connect with mincom at 38400 baud. no response when type help <enter>. hit on off button and got this:
Welcome to minicom 2.8

OPTIONS: I18n 
Port /dev/ttyACM0, 21:09:38

Press CTRL-A Z for help on special keys
                                                                                
I (595) esp_image: segment 5: paddr=0030c300 vaddr=4ff17f00 size=0464ch ( 17996) load
I (605) boot: Loaded app from partition at offset 0x10000                       
I (606) boot: Disabling RNG early entropy source...                             
I (618) hex_psram: vendor id    : 0x0d (AP)                                     
I (618) hex_psram: Latency      : 0x01 (Fixed)
I (618) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (619) hex_psram: dev id       : 0x03 (generation 4)
I (624) hex_psram: density      : 0x07 (256 Mbit)
I (628) hex_psram: good-die     : 0x06 (Pass)
I (632) hex_psram: SRF          : 0x02 (Slow Refresh)
I (637) hex_psram: BurstType    : 0x00 ( Wrap)
I (641) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (646) hex_psram: BitMode      : 0x01 (X16 Mode)
I (650) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (655) hex_psram: DriveStrength: 0x00 (1/1)
I (659) MSPI DQS: tuning success, best phase id is 0
I (837) MSPI DQS: tuning success, best delayline id is 17
I (837) esp_psram: Found 32MB PSRAM device
I (838) esp_psram: Speed: 200MHz
I (838) hex_psram: psram CS IO is dedicated
I (842) cpu_start: Multicore app
I (1794) esp_psram: SPI SRAM memory test OK
I (1803) cpu_start: Pro cpu start user code
I (1803) cpu_start: cpu freq: 360000000 Hz
I (1804) app_init: Application information:
I (1804) app_init: Project name:     doorbell_demo
I (1808) app_init: App version:      v1.0.0-40-gf74713a-dirty
I (1814) app_init: Compile time:     Sep 26 2025 20:55:45
I (1819) app_init: ELF file SHA256:  a3cdc295b...
I (1823) app_init: ESP-IDF:          v5.5.1
I (1827) efuse_init: Min chip rev:     v0.1
I (1831) efuse_init: Max chip rev:     v1.99 
I (1835) efuse_init: Chip rev:         v1.3
I (1839) heap_init: Initializing. RAM available for dynamic allocation:
I (1845) heap_init: At 4FF20860 len 0001A760 (105 KiB): RAM
I (1851) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (1856) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (1861) heap_init: At 50108080 len 00007F80 (31 KiB): RTCRAM
I (1867) heap_init: At 30100068 len 00001F98 (7 KiB): TCM
I (1872) esp_psram: Adding pool of 32768K of PSRAM memory to heap allocator
I (1879) spi_flash: detected chip: generic
I (1882) spi_flash: flash io: qio
W (1885) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary .
I (1898) host_init: ESP Hosted : Host chip_ip[18]
I (1902) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
sdio_mempool_create free:33690236 min-free:33690236 lfb-def:33030144 lfb-8bit:33030144

I (1919) H_API: ** add_esp_wifi_remote_channels **
I (1923) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x40015c42] Rx[0x40025856]
I (1930) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x40015b8a] Rx[0x40025856]
I (1938) main_task: Started on CPU0
I (1941) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1948) main_task: Calling app_main()
I (1952) Board: Init board.
E (1955) CODEC_INIT: No codec device found
E (1959) MEDIA_SYS: Memory allocate fail on 136
E (1963) MEDIA_SYS: Fail to create audio render

[... 54 lines cut ...]
4ff25aa0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff25ac0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4ff25ae0: 0x00000000 0x0000000c 0x4ff26efc 0x4ff20874 0x4ff25ae0 0x00001400 0xa5a5a5a5 0xa5a5a5a5
4ff25b00: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25b20: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25b40: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25b60: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25b80: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25ba0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25bc0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25be0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25c00: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25c20: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25c40: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25c60: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25c80: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25ca0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25cc0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25ce0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
4ff25d00: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5
CTRL-A Z for help | 38400 8N1 | NOR | Minicom 2.8 | VT102 | Offline | ttyACM0                                                  

Tried Putty in serial RAW and telnet modes and got unable to open connection on /dev/ttyACM0. no such file or directory.

Attached file has the build output with warnings:

TempoTian, 2025-09-28:
Recommend to use IDF release/v5.5 or release/v5.4 to test, v6.0 is stil under develop, some module is still NG, I can only make it build pass and not test on it yet.

TempoTian, 2025-09-28:
esp32p4-eys and dev kit has lot of differ, so same configuration can not reused.
To support esp32p4 need following changes:

1. For camera need change:
```
CONFIG_CAMERA_OV2710=y
```
2. For SDIO connection

[image]
Change SDIO setting use menuconfig according the actual pin

[image]

3. For audio support you can use factory code bsp to get codec_dev handle to use microphone
https://github.com/espressif/esp-dev-kits/tree/master/examples/esp32-p4-eye/examples/common_components/esp32_p4_eye
` bsp_audio_pdm_microphone_init` is used to get the handle

robogo-pj, 2025-09-28:
> Recommend to use IDF release/v5.5 or release/v5.4 to test, v6.0 is stil under develop, some module is still NG, I can only make it build pass and not test on it yet.

tried v6, 5.5.1 & 5.4.2.  5.5.1 was the only version that complied and download code tp the device. But diidn't work

robogo-pj, 2025-09-29:
there's a file called sdkconfig.defaults.esp32p4 in the doorbell_local folder that contains:
# Use camera SC2336
CONFIG_CAMERA_SC2336=y
CONFIG_CAMERA_SC2336_MIPI_RAW10_1920x1080_25FPS_2_LANE=y

do I replace it with:
CONFIG_CAMERA_OV2710=y
CONFIG_CAMERA_OV2710_MIPI_RAW10_1920x1080_25FPS_2_LANE=y
smeone said to lower resolution but the values they said were none standard. I will be using yolo11 segementation and.or Masl R-CNN to find classify and segment weeds and disesCONFIG_CAMERA_OV2710=y
CONFIG_CAMERA_OV2710_MIPI_RAW10_1920x1080_25FPS_2_LANE=yes in turfgrass.

TempoTian, 2025-09-30:
You can change sdio settings use `idf.py menuconfig` it will pop ui a simple UI. Currently SDIO is specail used for wifi communication.
goto component config --> esp_host config --> Host SDIO config, change like:
```
CLK GOIO NUM --> 28
```
I will prepare a runable example for you once I have spare time.

robogo-pj, 2025-09-30:
still not configured right. The system was resetting with:

**E (338298) transport: Not able to connect with ESP-Hosted slave device          
I (338298) transport: Reset slave using GPIO[54]                                
I (338299) os_wrapper_esp: GPIO [54] configured**  

both os_wrapper_esp and CONFIG_ESP_GPIO_SLAVE_RESET_SLAVE are set to 54. Is this a conflict. 

[esp32p4eye.log](https://github.com/user-attachments/files/22627163/esp32p4eye.log)

closed 2025-10-01 by robogo-pj